FreeCAD/FreeCAD #6243: Crash when editing patterns in sketch workbench

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4713

* **Reporter:** orionrobots
* **Date submitted:** 7/25/2021
* **FreeCAD version:** 0.19
* **Category:** Bug
* **Status:** new
* **Tags:** #tobeclosed

# Original report text

I have been trying to make a complex pattern - the holes in a breadboard. I have a face and sketching on the face, I create first a construction line 4.5mm from the side of the board. 
I then made a 1.5mm square.
I created a pattern of 2 x 5 of these, at 2.54mm separation. I used the symmetry tool to place the 2 columns on either side of the construction line.

Making sure the angle was good and that it made sense, I then attempted to create a pattern of 5x1 f these 2x5 blocks. I had made a number of attempts to get the parameters correct - 4 or 5 attempts to adjust and then undo loops to get the style I wanted. However, at the last undo, FreeCAD crashed.

# Additional information

Backtrace/report:

See backtrace in attached log2.txt

# Steps to reproduce

* Make a pattern in sketch mode - with multiple columns.
* Make a further pattern of these patterns.
* Adjust these - lengths, and angles.
* Undo a retry with different row/column counts. 
* Attempt to recreate the patterns.

# FreeCAD Info

```
OS: macOS 10.16
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24291 (Git)
Build type: Release
Branch: (HEAD detached at 0.19.2)
Hash: 7b5e18a0759de778b74d3a5c17eba9cb815035ac
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: C/Default (C)

```

# Other bug information

* **Priority:** normal
* **Severity:** crash
* **Category:** Bug
* **OS: MacOS 11.4 (20F71)**
* **Platform:** Intel
* **Updated:** 12/15/2021

# Discussion from Mantis ticket

### Comment by orionrobots 2021-07-25 18:54
Please can this be moved to sketcher? It should be there.

---

### Comment by Kunda1 2021-07-27 23:56
oriobrobots Thanks for the report. In general we ask users to first post to the forum. Because this for example may be fixed in the 0.20 dev version. For now, do you mind installing the weekly-build and testing to see if the crash still happens? https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

---

### Comment by Kunda1 2021-08-07 11:32
orionrobots bump

---

### Comment by orionrobots 2021-08-07 14:31
Build: OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.25311 (Git)
Build type: Release
Branch: master
Hash: c5524bb58a7126fa2242d6137eb82ee9d54e82d4
Python version: 3.9.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.2
Locale: C/Default (C)

Steps taken:
* Part Design
* New part
* Sketch - XY Plane.
* Create rect - consider it the outer rect for this sketch
* Use the symmetry constraint - horiz/vertical.
* Shift-D constrain both (let's say 54mm by 84mm for example). Should be fully constrained.
* Close sketch and Pad this 10mm.
* Select top surface, then tasks, create sketch.
* Create external geom references to the four edges.
* Create a further rect - make it 2mm on the vertical left. Make other edge equal.
* Select the 4 edges of this shape.
* Create array -> 5 x 2 - drag 90 degrees. It won't be spot on, so double click the angle - set to -90. Select the distance, and set it to 2.54mm.
* This is where I perform an undo step - since 2mm would be too large. Undo until there's no array - just the rect.
* Set the rect edge to 1.5mm.
* Reseelect, create 5 by 2 array. Reset -90 and 2.54mm.
* Select the existing array. Now we create a further array of these. we want a 5 x 1 array, still at -90 ish.
* Set the distance for the second array to 6.54mm apart. This won't give quite the right result. Ctrl -z and redo as 6 * 2.54mm.
* Set the angle to exactly -90. FreecCAD will start feeling slow at this point.
* Find the two middle rectangles. on the left-most edge, select the two points, and the horizontal centre line - perform a symmetry constraint. This will take some time to resolve. And didn't behave as expected (that rect had moved, with the whole reference group moving, but not by the same amount).
* ctrl-z to undo. Try a top and bottom point - and perform symmetry with them. Again - this will take some time to resolve. And still be wrong - ctrl-z and wait.
* Try top and bottom of middle set... result is still wrong, and it took ages to resolve.

Good news - I don't seem to be making it crash - but it is slow and not creating the symmetry I expect. 
* I finally get it symmetrical by selecting the middle rectangle of the second column in the middle array.
* Create a dimension - 2.54mm horizontal to the one of the left edge points.
* Select the whole array (Freecad will stall here) - mirror around the vertical centre line. Long time - but no crash. The view is pretty cluttered - but it's still alive.
* To enforce the mirror - select the rightmost top point of the left array, and the mirror point on the right array, then the vertical centre line. Add a symmetry constraint. Takes a while - but succeeds.
* The sketch seems under constrained - so use the under-constrained to see where the problem is, and on this basis, make a symmetry constraint for the points of the middle box of the right array to match the symmetry constraint on the left array - this constraint didn't get mirrored.

Should be fully constrained. Cut this into the original shape 5mm.

It' looks like I can't reproduce the crash. We can close this.

---

### Comment by jnxd 2021-08-10 07:26
orionrobots, is there any reason you want to make a pattern in sketcher rather than trying to make a pattern of the pocket made from just one rectangle? That might be able to make your life a little easier. We should move this discussion to the forum.

---

### Comment by Kunda1 2021-08-20 14:35
orionrobots please open a new thread per jnxd comment in ~15791 and x-post to here

---

### Comment by orionrobots 2021-08-21 21:24
Ok - thread opened at https://forum.freecadweb.org/viewtopic.php?f=3&t=61418.

---

### Comment by yorik 2021-12-15 09:42
Put the backtrace of the OP in an attached txt file to workaround export issues

luzpaz, 2022-05-03:
This is really a chronic undo/redo problem in Sketcher that we need to address

luzpaz, 2022-05-16:
@NickUfer Always post your full [About](https://wiki.freecad.org/About) info when reporting or offering feedback.

luzpaz, 2022-07-17:
@NickUfer ping

maxwxyz, 2024-03-30:
Is this still relevant after the latest improvements with the array tool and symmetry tool? Otherwise I'll close this.

closed 2024-05-18 by maxwxyz