FreeCAD/FreeCAD #6244: With stylesheets, Scroll bar overlaps arrows in Properties tab with stylesheets, never disappears

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4716

* **Reporter:** dxp.dev
* **Date submitted:** 7/29/2021
* **FreeCAD version:** 0.2
* **Category:** Bug
* **Status:** new
* **Tags:** 

# Original report text

Context : With any stylesheet (not with "no-stylesheet" theme)

Use-case : The properties of the data can be edited (such as length with up and down arrows, pocket type with a drop down menu by clicking on the down arrow, etc.).

Problem: The scroll bar overlaps those arrows, and when clicking on them, the scroll bar has priority and instead of modifying the properties, it just make the data window scroll.

See attached [video](https://mon-partage.fr/f/X8GfurTh/)

Proposed solution: The editing arrows shouldn't overlap with the scroll bar, or the scroll bar should disappear after  1/2second as in the "no-stylesheet" theme

# Additional information

[FreeCAD forum topic](https://forum.freecadweb.org/viewtopic.php?f=34&t=60639)

# Steps to reproduce

On FC 0.20 (git)

* Select a stylesheet (Behave Dark, all the Dark-*, Darker-*, Light-* or proDark)
* Open a file, apply anything to a model (pocket, pad, etc.)
* In properties tab, clic on the drop-down arrows or up/down arrow at the right of the property you want to edit
* Instead of clicking those arrows, scroll bar takes precedence and move the view (see attached video)

# FreeCAD Info

```
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.25288 (Git)
Build type: Release
Branch: master
Hash: bbb557aab6129efd9486d6df19f06c2976fdb7e5
Python version: 3.9.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.2
Locale: C/Default (C)
```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: MacOS  Big Sur 11.5**
* **Platform:** MacBook Pro mid-2015
* **Updated:** 8/11/2021

# Discussion from Mantis ticket

### Comment by chennes 2021-08-09 14:49
I can't reproduce this on Windows, so this may be a Mac-only Qt bug.

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.25442 (Git)
Build type: Debug
Branch: master
Hash: d635f03a60a7ae1e6f9d390b9cbd7bdb91692996
Python version: 3.8.6+
Qt version: 5.15.1
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

---

### Comment by dxp.dev 2021-08-09 16:56
On the forum, user Syres couldn't reproduce on Windows either:

[quote=Syres]On Windows as soon as the user squashes the Property panel to cause the scroll bar to appear the up/down arrows and combo box drop down arrow all shift to the left so they don't overlap (as would be expected), tested with Behave Dark stylesheet using:

OS: Windows 7 Version 6.1 (Build 7601: SP 1)
Word size of FreeCAD: 64-bit
Version: 0.20.25306 (Git)
Build type: Release
Branch: master
Hash: 5f5b3d9b67136edda1f6738f1a83154990f7839d
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United Kingdom (en_GB)[/quote]

---

### Comment by chennes 2021-08-11 17:13
I can't reproduce on:

OS: macOS 10.15
Word size of FreeCAD: 64-bit
Version: 0.20.25450 (Git)
Build type: Release
Branch: master
Hash: ed8e63270827afddd2f3e59cea2384af66e0f0d5
Python version: 3.9.1
Qt version: 5.15.2
Coin version: 4.0.0
OCC version: 7.5.0
Locale: English/United States (en_US)

Maybe fixed in recent Qt versions?

luzpaz, 2022-04-01:
I bumped the [forum thread](https://forum.freecadweb.org/viewtopic.php?p=585310#p585310)

luzpaz, 2022-05-08:
Per dxp.dev in the [forum thread](https://forum.freecadweb.org/viewtopic.php?p=593909#p593909) 
This is an upstream macOS issue. The workaround:
> MacOS System Preferences :arrow_right: General :arrow_right: Show scroll bars
set to `Always`

luzpaz, 2022-05-08:
CC @looooo @ipatch @PLChris 
just giving you a heads up about this known issue

closed 2022-05-08 by luzpaz

ipatch, 2022-05-08:
> CC @looooo @ipatch @PLChris just giving you a heads up about this known issue for macOS

yes, this i something i've noticed as well.