FreeCAD/FreeCAD #6245: Add ability for load ToolBar SVG icons from Python string, like as XPM

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4717

* **Reporter:** Evgeniy
* **Date submitted:** 7/31/2021
* **FreeCAD version:** 0.2
* **Category:** Feature
* **Status:** new
* **Tags:** python, SVG, toolbar, UI, workbench

# Original report text

Discussion [from forum](https://forum.freecadweb.org/viewtopic.php?f=22&t=16136&start=10)

My view of how it should look like:

```# default icon
Icon = """
/* SVG */
static const char *test_icon[]={
"<svg xmlns='http://www.w3.org/2000/svg' height='64' width='64'>",
"<rect height='64' width='64' fill='#204a87' />",
"</svg>"};
"""

# replace fill color
Icon.replace("fill='#204a87'","fill='#00FF00'")

# and icon with changed colors to toolbar

... Command creation code ...
	
def GetResources(self):
	return {"MenuText": "My Command",
		"Accel": "Ctrl+M",
		"ToolTip": "ToolTip command",
		"Pixmap"  : Icon}```

# FreeCAD Info

```
OS: Ubuntu 20.04.2 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.20.25281 (Git)
Build type: Release
Branch: master
Python version: 3.8.10
Qt version: 5.12.8
Coin version: 4.0.0
OCC version: 7.5.2

```

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 7/31/2021

luzpaz, 2022-04-03:
@chennes can you weigh in on this feature request? Is there any capability in the recent 'Addon Manager adding macro to toolbar' effort that is similar to this?

luzpaz, 2023-09-09:
@Syres916 maybe you have a moment to take a look at this ? (https://forum.freecadweb.org/viewtopic.php?f=22&t=16136)

kadet1090, 2026-09-13:
Workbenches should never rely on hardcoded icons. One reason being this won't ever be themeable resource, but in general hardcoding icons is simply a bad idea. While we cannot prevent it from happening, it's not something we should actively seek. Closing as not-planned.

closed 2026-09-13 by kadet1090