FreeCAD/FreeCAD #6247: Trying to use Gmsh python library from FreeCAD fails on gmsh.initialize with AttributeError: dlsym(RTLD_DEFAULT, gmshInitialize)

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4721

* **Reporter:** keithsloan52
* **Date submitted:** 8/8/2021
* **FreeCAD version:** 0.19
* **Category:** Bug
* **Status:** new
* **Tags:** 

# Original report text

If pip install gmsh python library under native python everything works,
All be it my standard python3 is 3.9.5
If I install under FreeCAD with
pip install gmsh -t<FreeCAD python Path>
Then from the python console I can successfully import gmsh
But if I try gmsh.initialize() I get
gmsh.initialize()
19:13:10    File "/Applications/FreeCAD 0.19.1.app/Contents/Resources/lib/python3.8/site-packages/gmsh.py", line 232, in initialize
19:13:10      lib.gmshInitialize(
19:13:10    File "/Applications/FreeCAD 0.19.1.app/Contents/Resources/lib/python3.8/ctypes/__init__.py", line 386, in __getattr__
19:13:10      func = self.__getitem__(name)
19:13:10    File "/Applications/FreeCAD 0.19.1.app/Contents/Resources/lib/python3.8/ctypes/__init__.py", line 391, in __getitem__
19:13:10      func = self._FuncPtr((name_or_ordinal, self))
19:13:10  AttributeError: dlsym(RTLD_DEFAULT, gmshInitialize): symbol not found

Have tried raising as an issue at gmsh github without success

# Steps to reproduce

pip3 install gmsh -t<Path to valid FreeCAD library>
from FreeCAD python Console
    import gmsh
    gmsh.initialize()

# FreeCAD Info

```
OS: macOS 10.15
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24276 (Git)
Build type: Release
Branch: (HEAD detached at 0.19.1)
Hash: a88db11e0a908f6e38f92bfc5187b13ebe470438
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: C/Default (C)

```

# Other bug information

* **Priority:** normal
* **Severity:** major
* **Category:** Bug
* **Updated:** 8/8/2021

luzpaz, 2022-03-11:
CC @KeithSloan since you're the OP

KeithSloan, 2022-03-11:
The problem is that the Gmsh python library needs to be able to find a matching Gmsh shared library.
Things work with RealThunder linkstage3 builds but not FreeCAD 0.19 and 0.20 builds.

KeithSloan, 2022-04-12:
I originally had problems with pip install [https://gitlab.onelab.info/gmsh/gmsh/-/issues/1449](url) which got closed and I was eventually able to find the library at least. As it then works with Native Gmsh and Python the  AttributeError: dlsym(RTLD_DEFAULT, gmshInitialize) is not a Gmsh problem as it only happens with FreeCAD.

One of the reasons I am trying to use the python Gmsh library is to produce a Minimum CAD Mesh see https://gitlab.onelab.info/gmsh/gmsh/-/issues/1726#note_16314.

In order to get this working and ALSO not have the RTD_DEAULT problem was to 
[](https://gitlab.onelab.info/geuzaine)Christophe Geuzaine
[Christophe Geuzaine ](https://gitlab.onelab.info/geuzaine)[@geuzaine](https://gitlab.onelab.info/geuzaine) · [2 weeks ago]
Did you figure it out?
Keith Sloan
[Keith Sloan ](https://gitlab.onelab.info/keithsloan)[@keithsloan](https://gitlab.onelab.info/keithsloan) · [2 weeks ago](https://gitlab.onelab.info/gmsh/gmsh/-/issues/1726#note_16734)
Author
I have stuff working by installing SDK and copying the following to /Applications/FreeCAD_19.2.app/Contents/Resources/lib

gmsh.jl		libgmsh.4.9.4.dylib
gmsh.py		libgmsh.4.9.dylib

Have not worked out how I could achieve this via pip

luzpaz, 2022-04-12:
@KeithSloan can you re-edit the format of your previous comment. looks like you copy/pasted but it didn't translate well

luzpaz, 2022-06-05:
@KeithSloan what is the error when you try this in v0.20dev ?

KeithSloan, 2022-06-06:
When I try with latest beta I get a new error
with code that has
````
    global doc
    print(doc)

````
    File "/Users/keithsloan/Library/Preferences/FreeCAD/Mod/GDML/./freecad/gdml/importGDML.py", line 1768, in processVol
        print(doc)
    <class 'NameError'>: name 'doc' is not defined

Is this due to python 3.9?

[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.28936 (Git)
Build type: Release
Branch: master
Hash: b02705d79f
Python 3.9.13, Qt 5.12.9, Coin 4.0.0, OCC 7.5.3
Locale: C/Default (C)
Installed mods: 
  * MeshRemodel 1.8919.0
  * ImportNURBS
  * dxf_library
  * InventorLoader 1.3.0
  * workfeature
  * Defeaturing
  * Macro-Vision
  * GDML 1.8.0
  * Manipulator 1.4.3
  * Design456
  * SWPart
  * OpenSCAD_Alt_Import
  * Jupyter
  * 3D_Printing_Tools
  * toSketch
  * Curves 0.3.0
[/code]

closed 2024-12-29 by github-actions[bot]