FreeCAD/FreeCAD #6252: Children and/or parent to be highlighted when hovering over something in the tree/3D viewport

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4730

* **Reporter:** Pauvres_honteux
* **Date submitted:** 8/20/2021
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

Forum topic: 
[Children and/or parent to be highlighted when hoovering over something in the tree/3D viewport](https://forum.freecadweb.org/viewtopic.php?f=34&t=57545)

This is about finding a features parent and/or children in a simple way.

# Additional information

N/A

# Steps to reproduce

N/A

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 8/20/2021

NXTler, 2023-08-26:
This seems very useful to me, especially for complex objects. I remember other programs like Unreal Engine (not related to CAD) and Solid Edge having the parent highlighted in a dimmer color.

luzpaz, 2023-09-23:
Adding relationship to #4133

luzpaz, 2023-09-23:
@PaddleStroke is this something that may be useful for Assembly WB ?

Cobras62, 2024-05-23:
Hello,
I'm bringing this discussion up again: is this a request for an equivalent to the "partial highlight" function in Freecad Realthunder? If so, I agree, we need to be able to implement this function in main Freecad.
If we want to attract new users to Freecad and above all keep them there, this is one of the functions that I think is important because of the ease of use it provides. Thanks to it, it's very easy, when you click on elements in the construction tree, to see which elements are linked by highlighting them in the 3D view.

luzpaz, 2024-08-07:
Any coding proposals ?

foreachidea, 2026-03-06:
happy i found this, was about to submit new issue. i would redefine slightly:

- **ux/ui: inconsistent mouse hover over object in model panel tree effect**

mouse hovering over an object in model panel tree highlights the object in the inventor view. While it works on primitives and primitives nested in Parts, it doesn’t seem to highlight primitives nested in Groups. Hovering over a Group containing primitives does not highlight all nested primitives. be great if it would :)