FreeCAD/FreeCAD #6253: Crash on start within RDP session

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4734

* **Reporter:** Night_Gryphon
* **Date submitted:** 8/25/2021
* **FreeCAD version:** 0.2
* **Category:** Bug
* **Status:** new
* **Tags:** crash, Win

# Original report text

FreeCad crash on win 8.1 while starting within RDP session. The blank main window appears immediately followed by two crash windows.
If i start FreeCad from local console and than login with RDP it continue to work.

Tried with build 0.20.24894 and 0.20.25429 (latest). Same result with 0.19.24267

The crash log show multiple "Access Violation" errors.

Forum
https://forum.freecadweb.org/viewtopic.php?f=3&t=61571

# Additional information

Log: Id: 6504: SymInit: Symbol-SearchPath: '.;C:\DIST\FreeCAD_weekly-builds-24894-Win-Conda_vc14.x-x86_64\bin;C:\DIST\FreeCAD_weekly-builds-24894-Win-Conda_vc14.x-x86_64\bin;C:\WINDOWS;C:\WINDOWS\system32;SRV*C:\websymbols*http://msdl.microsoft.com/download/symbols;', symOptions: 530, UserName: '*****'
Log: Id: 6504: OS-Version: 6.2.9200 () 0x300-0x1
Log: Id: 6504: C:\DIST\FreeCAD_weekly-builds-24894-Win-Conda_vc14.x-x86_64\bin\freecad.exe:freecad.exe (00007FF789770000), size: 163840 (result: 0), SymType: '-nosymbols-', PDB: 'C:\DIST\FreeCAD_weekly-builds-24894-Win-Conda_vc14.x-x86_64\bin\freecad.exe', fileVersion: 0.20.0.0
Log: Id: 6504: C:\WINDOWS\SYSTEM32\ntdll.dll:ntdll.dll (00007FF91CD50000), size: 1757184 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\SYSTEM32\ntdll.dll', fileVersion: 6.3.9600.19678
Log: Id: 6504: C:\WINDOWS\system32\KERNEL32.DLL:KERNEL32.DLL (00007FF91C570000), size: 1306624 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\system32\KERNEL32.DLL', fileVersion: 6.3.9600.19719
Log: Id: 6504: C:\WINDOWS\system32\KERNELBASE.dll:KERNELBASE.dll (00007FF91A1B0000), size: 1134592 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\system32\KERNELBASE.dll', fileVersion: 6.3.9600.19724
Log: Id: 6504: C:\DIST\FreeCAD_weekly-builds-24894-Win-Conda_vc14.x-x86_64\bin\FreeCADGui.dll:FreeCADGui.dll (00007FF8FB5E0000), size: 19902464 (result: 0), SymType: '-exported-', PDB: 'C:\DIST\FreeCAD_weekly-builds-24894-Win-Conda_vc14.x-x86_64\bin\FreeCADGui.dll'
Log: Id: 6504: C:\DIST\FreeCAD_weekly-builds-24894-Win-Conda_vc14.x-x86_64\bin\FreeCADApp.dll:FreeCADApp.dll (00007FF8FD6F0000), size: 7098368 (result: 0), SymType: '-exported-', PDB: 'C:\DIST\FreeCAD_weekly-builds-24894-Win-Conda_vc14.x-x86_64\bin\FreeCADApp.dll'
Log: Init: Showing main window
Log: Main window restored
Log: Id: 6504: C:\DIST\FreeCAD_weekly-builds-24894-Win-Conda_vc14.x-x86_64\bin\FreeCADBase.dll:FreeCADBase.dll (00007FF9074D0000), size: 2256896 (result: 0), SymType: '-exported-', PDB: 'C:\DIST\FreeCAD_weekly-builds-24894-Win-Conda_vc14.x-x86_64\bin\FreeCADBase.dll'
Err: Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
Log: The event type 14 was sent to QtWebEngineCore::RenderWidgetHostViewQtDelegateWidget
Object tree:
	QtWebEngineCore::RenderWidgetHostViewQtDelegateWidget is child of
	WebGui::WebView is child of
	WebGui::BrowserView is child of
	QMdiSubWindow is child of
	QWidget is child of
	QMdiArea is child of
	Gui::MainWindowErr: Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
Log: The event type 17 was sent to QtWebEngineCore::RenderWidgetHostViewQtDelegateWidget
Object tree:
	QtWebEngineCore::RenderWidgetHostViewQtDelegateWidget is child of
	WebGui::WebView is child of
	WebGui::BrowserView is child of
	QMdiSubWindow is child of
	QWidget is child of
	QMdiArea is child of
	Gui::MainWindowLog: Show main window
Log: Toolbars restored
Log: 3Dconnexion device not attached.
Log: Init: Entering event loop
Err: Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
Log: The event type 1 was sent to QtWebEngineCore::RenderWidgetHostViewQtDelegateWidget
Object tree:
	QtWebEngineCore::RenderWidgetHostViewQtDelegateWidget is child of
	WebGui::WebView is child of
	WebGui::BrowserView is child of
	QMdiSubWindow is child of
	QWidget is child of
	QMdiArea is child of
	Gui::MainWindowLog: Id: 6504: C:\DIST\FreeCAD_weekly-builds-24894-Win-Conda_vc14.x-x86_64\bin\MSVCP140.dll:MSVCP140.dll (00007FF9055E0000), size: 598016 (result: 0), SymType: '-exported-', PDB: 'C:\DIST\FreeCAD_weekly-builds-24894-Win-Conda_vc14.x-x86_64\bin\MSVCP140.dll', fileVersion: 14.28.29325.2
Log: Id: 6504: C:\DIST\FreeCAD_weekly-builds-24894-Win-Conda_vc14.x-x86_64\bin\VCRUNTIME140.dll:VCRUNTIME140.dll (00007FF908D60000), size: 102400 (result: 0), SymType: '-exported-', PDB: 'C:\DIST\FreeCAD_weekly-builds-24894-Win-Conda_vc14.x-x86_64\bin\VCRUNTIME140.dll', fileVersion: 14.28.29325.2
Log: Id: 6504: C:\WINDOWS\SYSTEM32\api-ms-win-crt-runtime-l1-1-0.dll:api-ms-win-crt-runtime-l1-1-0.dll (00007FF9031C0000), size: 16384 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\SYSTEM32\api-ms-win-crt-runtime-l1-1-0.dll', fileVersion: 10.0.14393.2247
Log: Id: 6504: C:\WINDOWS\SYSTEM32\api-ms-win-crt-environment-l1-1-0.dll:api-ms-win-crt-environment-l1-1-0.dll (00007FF9129D0000), size: 12288 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\SYSTEM32\api-ms-win-crt-environment-l1-1-0.dll', fileVersion: 10.0.14393.2247
Log: Id: 6504: C:\WINDOWS\SYSTEM32\api-ms-win-crt-heap-l1-1-0.dll:api-ms-win-crt-heap-l1-1-0.dll (00007FF9031D0000), size: 12288 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\SYSTEM32\api-ms-win-crt-heap-l1-1-0.dll', fileVersion: 10.0.14393.2247
Log: Id: 6504: C:\WINDOWS\SYSTEM32\api-ms-win-crt-math-l1-1-0.dll:api-ms-win-crt-math-l1-1-0.dll (00007FF912240000), size: 20480 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\SYSTEM32\api-ms-win-crt-math-l1-1-0.dll', fileVersion: 10.0.14393.2247
Log: Id: 6504: C:\WINDOWS\SYSTEM32\api-ms-win-crt-stdio-l1-1-0.dll:api-ms-win-crt-stdio-l1-1-0.dll (00007FF9031E0000), size: 16384 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\SYSTEM32\api-ms-win-crt-stdio-l1-1-0.dll', fileVersion: 10.0.14393.2247
Err: Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
Log: The event type 14 was sent to QtWebEngineCore::RenderWidgetHostViewQtDelegateWidget
Object tree:
	QtWebEngineCore::RenderWidgetHostViewQtDelegateWidget is child of
	WebGui::WebView is child of
	WebGui::BrowserView is child of
	QMdiSubWindow is child of
	QWidget is child of
	QMdiArea is child of
	Gui::MainWindowLog: Id: 6504: C:\WINDOWS\SYSTEM32\api-ms-win-crt-locale-l1-1-0.dll:api-ms-win-crt-locale-l1-1-0.dll (00007FF910270000), size: 12288 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\SYSTEM32\api-ms-win-crt-locale-l1-1-0.dll', fileVersion: 10.0.14393.2247
Log: Init: Processing command line files
Err: Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
Err: Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
Log: Id: 6504: C:\DIST\FreeCAD_weekly-builds-24894-Win-Conda_vc14.x-x86_64\bin\Qt5Widgets_conda.dll:Qt5Widgets_conda.dll (00007FF8FD190000), size: 5591040 (result: 0), SymType: '-exported-', PDB: 'C:\DIST\FreeCAD_weekly-builds-24894-Win-Conda_vc14.x-x86_64\bin\Qt5Widgets_conda.dll', fileVersion: 5.12.9.0
Err: Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
Log: The event type 77 was sent to Gui::MainWindow
Object tree:
	Gui::MainWindowErr: Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
Log: The event type 77 was sent to Gui::MainWindow
Object tree:
	Gui::MainWindowErr: Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
Log: The event type 77 was sent to Gui::MainWindow
Object tree:
	Gui::MainWindowErr: Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
Log: The event type 77 was sent to Gui::MainWindow
Object tree:
	Gui::MainWindowErr: Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
Log: The event type 77 was sent to Gui::MainWindow
...

# Steps to reproduce

Login to windows 8.1 machine via RDP
Start freecad.exe

# FreeCAD Info

```
OS: Windows 8.1 (6.3)
Word size of FreeCAD: 64-bit
Version: 0.20.24894 (Git)
Build type: Release
Branch: master
Hash: a378b6b66f8a5992fc1452567ff59bdb613abc80
Python version: 3.8.10
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.2
Locale: Russian/Russia (ru_RU)

```

# Other bug information

* **Priority:** normal
* **Severity:** crash
* **Category:** Bug
* **Updated:** 9/8/2021

# Discussion from Mantis ticket

### Comment by Night_Gryphon 2021-09-08 09:00
Using MESA3D solved the problem

https://forum.freecadweb.org/viewtopic.php?p=414094#p414094

Please add MESA3D library to FreeCAD distribution and some checks to load Mesa3D in case system OpenGL lib is non functional...

luzpaz, 2022-05-08:
> Please add MESA3D library to FreeCAD distribution and some checks to load Mesa3D in case system OpenGL lib is non functional...

should we add this as a dependency in cmake? 

CC @wwmayer @looooo

wwmayer, 2022-05-09:
> and some checks to load Mesa3D in case system OpenGL lib is non functional

How should this be checked? You can only see that the system OpenGL doesn't work if FreeCAD crashes. But nevertheless, the culprit is the system OpenGL and according to the latest forum post from apeltauer upgrading the driver should fix the problem.

The Mesa3D might be an alternative for the QtWebEngineCore (i.e. when the StartPag is shown) but since it was said that its performance is quite poor I don't consider it a good alternative for the 3D view.

IMO all what we could do is adding the MESA3D DLL to a Windows installer (but rename the file to avoid to load it by accident) and it would be up to the user to rename the file again if he has trouble with OpenGL.

luzpaz, 2022-05-09:
So it would be up to the user to manually rename MESA3D DLL ?

wwmayer, 2022-05-11:
Yes, but only if the user trouble with OpenGL

luzpaz, 2022-05-11:
Maybe we can create a comment for this in the cmake output?

adrianinsaval, 2022-05-27:
according to werner:

> IMO all what we could do is adding the MESA3D DLL to a Windows installer (but rename the file to avoid to load it by accident) and it would be up to the user to rename the file again if he has trouble with OpenGL.

see https://forum.freecadweb.org/viewtopic.php?p=414094#p414094 to get it, we should then add an entry in the Known OS specific problems post explaining this.

donovaly, 2022-05-27:
I would go the other way round:
- add it to the  Known OS specific problems
- link there the download link, for example the proposed: https://fdossena.com/?p=mesa/index.frag

This way we don't deliver a big DLL to all users while > 99% don't need it.

adrianinsaval, 2022-05-27:
Considering the user would have to find instructions online anyways I think that makes sense so once this stuff is documented I think this issue can be closed.
@PLChris

donovaly, 2022-05-27:
OK, I added it now to the know limitations:
https://wiki.freecadweb.org/Release_notes_0.20#Remote_Desktop_under_Windows

Can anyone please proof-read it?

luzpaz, 2022-05-27:
LGTM

donovaly, 2022-05-27:
> LGTM

"Let's get the money", "Literally Garbage To Me" or "Let's get this merged"? ;-)

I wonder that the library has to be renamed. I have no clue what the suffix "sw" means. And when one replaced this file, why is then Mesa 3D taken, not OpenGL? Especially this is what I don't understand and therefore thinks my description is wrong.

luzpaz, 2022-05-27:
Looks Good To Me :laughing:

donovaly, 2022-06-03:
> > LGTM
> 
> "Let's get the money", "Literally Garbage To Me" or "Let's get this merged"? ;-)
> 
> I wonder that the library has to be renamed. I have no clue what the suffix "sw" means. And when one replaced this file, why is then Mesa 3D taken, not OpenGL? Especially this is what I don't understand and therefore think my description is wrong.

@wwmayer ca you please proof-read? I am pretty sure my description is incorrect and I don't know the necessary details. I assume you do.

wwmayer, 2022-06-12:
> @wwmayer ca you please proof-read?

Looks good to me.

> I wonder that the library has to be renamed. I have no clue what the suffix "sw" means. And when one replaced this file, why is then Mesa 3D taken, not OpenGL? Especially this is what I don't understand and therefore think my description is wrong.

The DLL is loaded by Qt, for more details see: https://doc.qt.io/qt-5/windows-requirements.html#graphics-drivers

donovaly, 2022-06-12:
Thanks for reviewing

adrianinsaval, 2023-05-11:
is any action still required here or should we close this?

mosfet80, 2023-09-15:
try this https://github.com/FreeCAD/FreeCAD/pull/10684

wwmayer, 2024-12-02:
> With FreeCAD_1.0.0-conda, which is the new stable, it's not possible to use it anymore in virtual machines and remode desktop sessions.

Does the installation contain the file opengl32sw.dll? If you rename it to opengl32.dll does it fix the problem?

> This system is running OpenGL 1.1.

In the past when I was using a freshly installed Windows then the OpenGL version always was 1.1 and I had to install the latest graphic driver to make it working.

Renaming the opengl32sw.dll is a workaround to enable OpenGL with software rendering.

nagyrobi, 2024-12-04:
> Does the installation contain the file opengl32sw.dll? If you rename it to opengl32.dll does it fix the problem?

With the original opengl32sw.dll distributed with FreeCAD renamed to opengl32.dll the error messages change slightly, when trying to open the Preferences window:
```
15:05:02  GDI::createContext: wglCreateContext failed. (Operation successfully completed.)
15:05:02  Unable to create a GL Context.
15:05:14  GDI::createContext: wglCreateContext failed. (Operation successfully completed.)
15:05:14  Unable to create a GL Context.
15:05:15  Base exception thrown for 'Gui::Dialog::DlgSettings3DViewImp'
15:05:15  <Exception> Access violation
15:05:18  GDI::createContext: wglCreateContext failed. (Operation successfully completed.)
15:05:18  Unable to create a GL Context.
15:05:18  GDI::createContext: wglCreateContext failed. (Operation successfully completed.)
15:05:18  Unable to create a GL Context.
15:05:18  (qt.qpa.backingstore) composeAndFlush: QOpenGLContext creation failed
15:05:18  GDI::createContext: wglCreateContext failed. (Operation successfully completed.)
15:05:18  Unable to create a GL Context.
15:05:18  (qt.qpa.backingstore) composeAndFlush: makeCurrent() failed
15:05:21  GDI::createContext: wglCreateContext failed. (Operation successfully completed.)
15:05:21  Unable to create a GL Context.
15:05:21  (qt.qpa.backingstore) composeAndFlush: makeCurrent() failed
15:05:21  GDI::createContext: wglCreateContext failed. (Operation successfully completed.)
15:05:21  Unable to create a GL Context.
15:05:21  (qt.qpa.backingstore) composeAndFlush: makeCurrent() failed
15:05:21  GDI::createContext: wglCreateContext failed. (Operation successfully completed.)
15:05:21  Unable to create a GL Context.
15:05:21  (qt.qpa.backingstore) composeAndFlush: makeCurrent() failed
15:05:21  GDI::createContext: wglCreateContext failed. (Operation successfully completed.)
15:05:21  Unable to create a GL Context.
15:05:21  (qt.qpa.backingstore) composeAndFlush: makeCurrent() failed
```

nagyrobi, 2024-12-04:
> In the past when I was using a freshly installed Windows then the OpenGL version always was 1.1 and I had to install the latest graphic driver to make it working.

This is a copy of Windows 11 running in a QEMU VM. It's currently configured to use [SPICE](https://www.spice-space.org/) virtual graphics card (with appropriate drivers installed), but the system itself is accessed through Microsoft Remote Desktop client.

I use for example PrusaSplicer on the same machine, which renders 3D views without any issues in the same environment, just like the previous FreeCAD version did.

nagyrobi, 2025-01-02:
A way to trick is to pass through the GPU from the host, which in my case is `CoffeeLake-S GT2 [UHD Graphics 630]`. Rebooting the VM will result in Windows's Device Manager to detect it as a video card, let it install the appropriate drivers (Windows dit it by itself). Connecting through RDP and starting FreeCAD after that will function properly.

This, however comes with the compromise of not being able anymore to live migrate the VM between the hosts, as it's tied directly to the hardware GPU...