FreeCAD/FreeCAD #6254: Integration of 3D Parametric Curve macro

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4735

* **Reporter:** NewJoker
* **Date submitted:** 8/26/2021
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** macros

# Original report text

I think that it would be nice to have the 3D Parametric Curve macro implemented as a built-in function in FreeCAD. That's because this can be considered as one of the core functionalities of CAD software.

Forum thread: https://forum.freecadweb.org/viewtopic.php?f=8&t=61453

# FreeCAD Info

```
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24276 (Git)
Build type: Release
Branch: releases/FreeCAD-0-19
Hash: a88db11e0a908f6e38f92bfc5187b13ebe470438
Python version: 3.8.6+
Qt version: 5.15.1
Coin version: 4.0.1
OCC version: 7.5.0
Locale: Polish/Poland (pl_PL)
```

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 8/26/2021

luzpaz, 2022-05-25:
Related to #6254 and ultimately with #6255

luzpaz, 2022-08-27:
@tomate44 ...along with this ticket as well ? TIA :blush:

tomate44, 2022-08-28:
This one would be more suitable to a dev that feel comfortable with Gui / Qt.

luzpaz, 2022-08-28:
@tomate44 is this appropriate for Surface WB ?

tomate44, 2022-08-28:
It mainly depends on our interpretation of **Surface** workbench.
Some may think it should only group all the tools that are related to surfaces.
I think it should group the tools that are related to a **Surfacing workflow** = **bottom-up workflow** starting from points / curves then building surfaces, and possibly shells / solids.
However, I think this particular tool would better fit in the Part WB (maybe the primitives tools ?).