FreeCAD/FreeCAD #6255: Integration of Curves workbench

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4736

* **Reporter:** NewJoker
* **Date submitted:** 8/26/2021
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

Curves workbench is one of the most useful and important add-on modules for FreeCAD. It provides tools that can be considered as crucial for many designs, regardless of the branch of engineering. Thus, it would be great to have it integrated with FreeCAD.

Forum thread (mentioned): https://forum.freecadweb.org/viewtopic.php?f=8&t=61453&start=13

# FreeCAD Info

```
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24276 (Git)
Build type: Release
Branch: releases/FreeCAD-0-19
Hash: a88db11e0a908f6e38f92bfc5187b13ebe470438
Python version: 3.8.6+
Qt version: 5.15.1
Coin version: 4.0.1
OCC version: 7.5.0
Locale: Polish/Poland (pl_PL)
```

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 8/30/2021

# Discussion from Mantis ticket

### Comment by openBrain 2021-08-30 17:43
Would upvote this one. :)

luzpaz, 2022-03-21:
CC @tomate44

adrianinsaval, 2022-07-21:
should we consider this issue out of scope? see https://forum.freecadweb.org/viewtopic.php?p=611120#p611120

luzpaz, 2022-07-21:
Thanks. 

Per the thread:
> No, our goal is not to merge the whole Curves workbench.
We plan only to implement the BlendCurve and the BlendSurface tools

[image]

Yea, I guess it should be closed. Maybe open a ticket that called "Port Curve WB BlendCurve and BlendSurface tools to Surface WB" ?

CC @tomate44

FEA-eng, 2022-07-21:
> should we consider this issue out of scope? see https://forum.freecadweb.org/viewtopic.php?p=611120#p611120

But the discussion there is about the integration of just a single tool while the rest will remain in the Curves workbench and could be implemented in the future, as requested here.

adrianinsaval, 2022-07-21:
My point is that the curves workbench dev himself has said he won't be even attempting to merge it any time soon and it is not the first time he's said so, therefore this request is unrealistic and serves no purpose.

luzpaz, 2022-07-21:
> It's also not reasonable to expect other people to take over responsibility for those tools and put them on master.  

Is it though? Just playing devil's advocate here for a moment: a ticket like this although very generalized, does signal the reader or curious dev to the fact that this would be a valid FreeCADroadmap feature. At first glance most users don't know that CurvesWB is python based while SurfaceWB is cpp. And porting the latter to the former would be a highly regarded prize. So perhaps instead of closing this ticket, using it as a tracking ticket to show what efforts have been done and what still remain to reach the end goal

adrianinsaval, 2022-07-21:
I don't think python vs CPP is a roadblock, IIRC there are more than one workbench that uses both. I take it then the idea is that this issue is for future devs that might be adventurous enough to take and combine both codebases?

tomate44, 2022-07-21:
I don't plan to merge the whole Curves workbench into master. Many tools are really half-baked experiments. But I think several tools are pretty useful and popular (joinCurves, splitCurves, discretize and approximate, Sketch_on_Surface and some others) and would deserve integration.
I am not confident with C++, but I hope our attempt with BlendCurve / BlendSurface this summer will put me on the right track to go further.
We got the BlendCurve feature working today !

luzpaz, 2022-07-21:
> But I think several tools are pretty useful and popular (joinCurves, splitCurves, discretize and approximate, Sketch_on_Surface and some others) and would deserve integration.

@tomate44 so should  we open a general ticket to track said feature requests ? or open a ticket for each ?

luzpaz, 2023-01-09:
@tomate44 please advise. Should I close this ticket and instead we track feature tickets in Curves repo about porting above features to master?

PaddleStroke, 2023-12-21:
@tomate44 I think trying to move the functions that you have that are stable enough to core is a great idea. Hope you are still considering it !
Maybe you should start by one tool and move your way up. One step at the time.

pierreporte, 2024-08-28:
@maxwxyz I think that #12699 supersedes this one.

closed 2024-08-28 by maxwxyz