FreeCAD/FreeCAD #6256: Increasing Number of decimals doesn't incrase the number of decimal places shown

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4737

* **Reporter:** ifohancroft
* **Date submitted:** 8/30/2021
* **FreeCAD version:** 0.2
* **Category:** Bug
* **Status:** assigned
* **Tags:** 

# Original report text

If for a dimension constraint (pretty much anything requiring you to enter a number) I set a floating point number that has 5 decimal places, unless I open the dimension for editing, I only see 4 decimal places, despite Number of Decimals under Preferences > General > Units being set to 5 (Chrisb has it set to 6 and can see 6 decimal places).

If I increase Number of decimals, I still continue to see only 4 decimal places. The only thing that changes is that if I open the dimension for editing, there is now an extra 0 at the end.

[Forum Post](https://forum.freecadweb.org/viewtopic.php?p=528969)

# Steps to reproduce

1. Set a dimension constraint on an element in Sketcher and set it to a floating point number having more decimal places than what Number of decimals is set to
2. Check how many of the decimal places are being shown
3. Increase the value of Number of decimals
4. See that the number is showing the same and the only thing that changed is that you now have an extra 0 or more at the end of the number when you open the constraint for editing.

# FreeCAD Info

```
OS: Manjaro Linux (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 0.20.25288 (Git) AppImage
Build type: Release
Branch: master
Hash: bbb557aab6129efd9486d6df19f06c2976fdb7e5
Python version: 3.9.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.2
Locale: English/United States (en_US)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: Manjaro 5.10 Kernel**
* **Platform:** GNU/Linux
* **Updated:** 12/20/2021

# Discussion from Mantis ticket

### Comment by openBrain 2021-08-30 17:43
The current behavior is much probably correct regarding what the coder wants to implement.
Let's talk on if we want something different on the forum thread before continuing this ticket.

---

### Comment by uwestoehr 2021-12-20 20:05
This bug also applies when you set e.g. only 2 digits in the general preferences of FreeCAD. Here is a bug report about this: https://forum.freecadweb.org/viewtopic.php?f=10&t=64645#p554894

---

### Comment by uwestoehr 2021-12-20 20:11
Here is an idea what the bug could be: https://forum.freecadweb.org/viewtopic.php?f=10&t=64645&p=555225#p555225

luzpaz, 2022-05-03:
Per @donovaly in the forum:  
[image]

ifohancroft, 2022-05-03:
Like I said in the forum, for me, ideally, it would be if the option was to control the number of digits shown after the decimal point.

closed 2024-12-29 by github-actions[bot]