FreeCAD/FreeCAD #6257: Dogbone Dressup does not work in corners with an edge length of less than 1.9mm

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4740

* **Reporter:** kormorix
* **Date submitted:** 9/4/2021
* **FreeCAD version:** 0.2
* **Category:** Bug
* **Status:** new
* **Tags:** dogbone

# Original report text

***de:*** DogboneDressup funktioniert nicht in Ecken mit Kantenlänge kleiner 1,9mm, wenn die Ecke aus Richtung der kurzen Kante angefahren wird.

***en:*** DogboneDressup does not work in corners with an edge length of less than 1.9mm if the corner is approached from the direction of the short edge.

# Additional information

***de:***Bei der vorangegangenen Version 24276 habe ich das nicht beobachtet (muß aber nicht heißen, daß es dort nicht auftrat).

***en:*** I did not observe this in the previous version 24276 (but does not have to mean that it did not occur there).

# Steps to reproduce

***de:*** Rechteckiges Teil, Dicke z.B. 4mm, mit Zapfen konstruieren (siehe Anhang). 
In "path" einen Job anlegen, 2mm Schaftfräser soll Aussenkontur des Teils fräsen. Das erzeugte "Profile" mit der Eigenschaft "DogboneDressup" versehen. Bei Kantenlängen des Zapfens von weniger als 1,9mm erzeugt FreeCAD nur an der einen (in der Bearbeitungsrichtung ersten Ecke) von zwei Innenecken den Dogbone, wenn clockwise gefräst wird. Bei CCW gar kein Dogbone. 

Nachtrag: ist vom Fräser-Durchmesser abhängig! Das oben beschriebene gilt für Fräser-Durchmesser = 2mm!
Nachtrag: Abhängig direkt vom Fräserdurchmesser. In obigem Beispiel war der Fräserdurchmesser real 1,9mm (und nicht 2,0mm!).

Die Dogbone-Eigenschaften (Doppelklick auf DogboneDressup) funktionieren entsprechend auch nicht richtig.

***en:*** Construct a rectangular part, e.g. 4mm thick, with tenons (see appendix).
Create a job in "path", the 2mm end mill is to mill the outer contour of the part. Provide the created "Profile" with the property "DogboneDressup". If the tenon edge lengths are less than 1.9mm, FreeCAD only creates the dogbone at one (first corner in the machining direction) of two inside corners when milling clockwise. At CCW no dogbone at all.

Addendum: depends on the cutter diameter! The above described applies to cutter diameter = 2mm!
Addendum: Depending directly on the cutter diameter. In the example above, the cutter diameter was actually 1.9mm (and not 2.0mm!).

The dogbone properties (double click on DogboneDressup) do not work properly either. 

# FreeCAD Info

```
OS: macOS 10.15
Word size of FreeCAD: 64-bit
Version: 0.20.25566 (Git)
Build type: Release
Branch: master
Hash: 8c361dd3fc83840a911da950a845eb9f9342a0b6
Python version: 3.9.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.2
Locale: C/Default (C)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: MacOS Catalina**
* **Platform:** Ryzentosh
* **Updated:** 9/6/2021

# Discussion from Mantis ticket

### Comment by kormorix 2021-09-04 18:49
... hier noch die Anlagen (screenshots)

---

### Comment by kormorix 2021-09-04 20:28
[*]

---

### Comment by Kunda1 2021-09-06 10:14
Thank you, in the future please discuss the issue on the Path forum first (our guidelines for reporting bugs is to use the forum, that's because sometimes tickets aren't necessary due to fixes made on the spot. Also please translate bugs to english. Thanks again.)

---

### Comment by Kunda1 2021-09-06 10:17
CC sliptonic

sliptonic, 2022-04-06:
This is a limitation of the dogbone.  Converting to a feature request and tagging for 0.21.

sliptonic, 2023-08-12:
Cannot duplicate anymore.  I believe this was resolved by the dogbone refactor work last year.  closing

closed 2023-08-12 by sliptonic