FreeCAD/FreeCAD #6259: Section View Shows Hidden Stairs

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4746

* **Reporter:** robocat
* **Date submitted:** 9/21/2021
* **FreeCAD version:** 0.2
* **Category:** Bug
* **Status:** assigned
* **Tags:** Arch, BIM, section, TechDraw

# Original report text

Upon making a simple level containing a floor with a staircase going through it in BIM, a problem is found whereby if the staircase is of structure "One stringer" or "Two stringers", any section view of that staircase will be faulty. (Please see example files).

By putting the section into a TechDraw ArchView the result is that the entire staircase is shown on top of other geometry regardless of if it is obscured by the floor or even if it is above the horizontal section plane.

For the prior forum discussion see: https://forum.freecadweb.org/viewtopic.php?f=23&t=62227

The suspected affected parts of FreeCAD are: "both Stairs and ArchSection/TechDraw-ArchView has some bug" - paullee

The summary of the discussion: "So, to summarize, "Structure" - "One Stringer" / "Two Stringer" are broken somehow, and no solid is resulted and thus the strange error. Making Riser / Tread Thickness > 0 make Stair return solid, thus also 'fix the problem'." - paullee

The current workaround is to set "Structure" to "Massive" or change "Riser Thickness" from 0 to 1.

# Steps to reproduce

1. In the BIM workbench create a slab with a hole in it.
2. Create a wall next to the hole in the floor.
3. Add a staircase through the hole.
4. Set the staircase structure property to "One stringer" or "Two stringers".
5. Move all these components into a level.
6. Section plane the level. This should be horizontally by default. This should also pass through the staircase. If it does not make the staircase longer by increasing Height and Length properties.
7.Swap to the TechDraw workbench.
8. Insert default page.
9. Click on the section in the tree and then click Insert Arch Workbench Object
10. Change the "Scale" of the ArchView if needed (0.02) and change the "Line Width" and "Cut Line Width" if needed (0.2).

# FreeCAD Info

```
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.25645 (Git)
Build type: Release
Branch: master
Hash: 37d9757399b4c2ec30318eb88d7cd7c508246345
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United Kingdom (en_GB)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: Windows 10 2009**
* **Platform:** Windows
* **Updated:** 12/13/2021

# Discussion from Mantis ticket

### Comment by wandererfan 2021-12-12 18:54
This can't be fixed in TechDraw.  Arch created the view using SVG and TechDraw just displays whatever it is given.

---

### Comment by paullee 2021-12-13 00:20
Can change the Project from TechDraw to Arch ?

---

### Comment by chennes 2021-12-13 02:47
I've changed the project this is categorized under from TechDraw to Arch.

---

### Comment by paullee 2021-12-13 17:22
Thanks !

luzpaz, 2022-04-11:
@paullee0 is this issue still presenting in latest v0.20dev ?

luzpaz, 2022-09-13:
@paullee0 ping

luzpaz, 2023-02-26:
@Roy-043 since you worked on the #8514 would you mind taking a look at this and weigh-in ?

paullee0, 2023-02-27:
👏👏👏

PR #8688 (2023-02-28): [Arch] Arch_SectionPlane improve handling of non-solids
Fixes #6259.

Note that all Draft objects, except annotations, are processed as 'normal' objects. The produced results now more closely match what Draft_Shape2DView would create from a section plane.

Test file I have used (based on the file from the issue):
[staircase_test_roy.zip](https://github.com/FreeCAD/FreeCAD/files/10841246/staircase_test_roy.zip)

- [x]  Your Pull Request meets the requirements outlined in section 5 of [CONTRIBUTING.md](https://github.com/FreeCAD/FreeCAD/blob/master/CONTRIBUTING.md) for a Valid PR

---
closed 2023-02-28 by yorikvanhavre