FreeCAD/FreeCAD #6261: Make "Missing coincidences" and "Open vertices" available in Sketcher

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4748

* **Reporter:** chrisb
* **Date submitted:** 9/21/2021
* **FreeCAD version:** 0.2
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

The functions "Missing coincidences" and "Open vertices" from the Validate Sketch tool would be very convenient if they were available directly in Sketcher while editing a sketch.
[Forum discussion](https://forum.freecadweb.org/viewtopic.php?p=534845#p534889)

# FreeCAD Info

```
OS: macOS High Sierra (10.13)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24276 (Git)
Build type: Release
Branch: (HEAD detached at 0.19.1)
Hash: a88db11e0a908f6e38f92bfc5187b13ebe470438
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: C/Default (C)

```

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 9/21/2021

adrianinsaval, 2022-02-09:
forum thread is mostly about a different feature, perhaps it's better to split whatever actually pertains to this feature request or start a new discussion?

luzpaz, 2022-02-09:
This is a feature so I'm in no rush to bother abdullah right now about it. I think the sketcher folks can sort this out when it comes time to do a new release.

closed 2024-03-30 by maxwxyz