FreeCAD/FreeCAD #6266: recompute undos removeSplitter

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4754

* **Reporter:** basnijholt
* **Date submitted:** 9/27/2021
* **FreeCAD version:** 0.19
* **Category:** Bug
* **Status:** feedback
* **Tags:** #pending-forum

# Original report text

When calling `move`, the object moves, then I call `obj.Shape.removeSplitter()` and `doc.recompute` and the shifted box is unmoved.

# Steps to reproduce

Execute the following in the Python console:

```from freecad import app as FreeCAD

import Draft

doc = FreeCAD.newDocument("testDoc")
box1 = doc.addObject("Part::Box", "Box1")
doc.recompute()
box2 = Draft.move(box1, FreeCAD.Vector(1, 0, 0), copy=True)
box2.Shape = box2.Shape.removeSplitter()

doc.recompute()```

# FreeCAD Info

```
OS: macOS 10.16
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24291 (Git)
Build type: Release
Branch: (HEAD detached at 0.19.2)
Hash: 7b5e18a0759de778b74d3a5c17eba9cb815035ac
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: C/Default (C)

```

# Other bug information

* **Priority:** normal
* **Severity:** major
* **Category:** Bug
* **Updated:** 1/24/2022

# Discussion from Mantis ticket

### Comment by openBrain 2021-10-13 13:52
basnijholt : the problem is that you're not authorized to override a Shape property. You should use a new object.
I tested and there seems to me that there are 2 problem though : 1) the rendered shape is temporarily overridden (before change is rejected) + 2) trying to assign a new shape resets the Placement property -- this is what you see --

Could you please open a new thread in our forum according [our reporting guidelines](https://forum.freecadweb.org/viewtopic.php?f=3&t=5236) and post the link here ? Thx

---

### Comment by Kunda1 2022-01-24 20:16
Forum thread: https://forum.freecadweb.org/viewtopic.php?f=8&t=65635

adrianinsaval, 2022-02-08:
title should be changed, removeSplitter is not related to the issue. It might not be an issue actually, altought it is confusing for users this is expected behavior according to @wwmayer, should this be closed?
cc @0penBrain 

also there's a weird formatting bug in the imported bug report

0penBrain, 2022-02-08:
IMO unless we want to enforce documenting limitations explained by wwmayer in the forum post, we can close this bug.

adrianinsaval, 2022-06-25:
from wmayer comments in the forum I think this issue can be closed

closed 2022-06-25 by chennes