FreeCAD/FreeCAD #6268: Crash with AdditivePipe

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4758

* **Reporter:** hcbang
* **Date submitted:** 10/3/2021
* **FreeCAD version:** 0.2
* **Category:** Bug
* **Status:** new
* **Tags:** upstream

# Original report text

FC crashes when trying additivePipe with CornerTransition set to RightCorner

# Additional information

whas discussed in the help forum with the same title,
the same error under windows10

# Steps to reproduce

get the added FC File
Click on Sketch005, which is the closed curve to be extruded.
Click on AdditivePipe
Change CornerTransition to RightCorner
Click on Buttom Object
Click on one Part of Sketch002, which is the line for the pipe direction
Et voila, FC crashes on my computer

# FreeCAD Info

```
OS: Ubuntu 21.04 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.25541 (Git) AppImage
Build type: Release
Branch: master
Hash: ca043dcb6d20d70b8e24d75e810683214eecc5d4
Python version: 3.9.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.2
Locale: German/Germany (de_DE)

```

# Other bug information

* **Priority:** normal
* **Severity:** crash
* **Category:** Bug
* **Updated:** 10/13/2021

# Discussion from Mantis ticket

### Comment by hcbang 2021-10-03 12:40
I get no error, when the sketch to be extruded is on the endpoint or startingpoint? of the path sketch

---

### Comment by openBrain 2021-10-05 07:40
As we may expect, it crashes deep in OCC :
```Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x41040) [0x7fbaac35c040]
#1  0x7fba6ff50c0c in BRepFill_PipeShell::BuildHistory(BRepFill_Sweep const&) from /lib/x86_64-linux-gnu/libTKBool.so.7+0x49c
#2  0x7fba6ff46f5a in BRepFill_PipeShell::Build() from /lib/x86_64-linux-gnu/libTKBool.so.7+0x92a
#3  0x7fba6efd3ae1 in BRepOffsetAPI_MakePipeShell::Build() from /lib/x86_64-linux-gnu/libTKOffset.so.7+0x11
#4  0x7fba6fb566fd in BRepBuilderAPI_MakeShape::Shape() from /lib/x86_64-linux-gnu/libTKTopAlgo.so.7+0x3d
#5  0x7fba15d2b4c8 in PartDesign::Pipe::execute() from /home/OB/dev/FCBuild/Mod/PartDesign/_PartDesign.so+0xc4e
#6  0x7fbaaedd5dd5 in App::DocumentObject::recompute() from /home/OB/dev/FCBuild/lib/libFreeCADApp.so+0xeb
#7  0x7fba75f56f51 in Part::Feature::recompute() from /home/OB/dev/FCBuild/Mod/Part/Part.so+0x1f
#8  0x7fbaaecd8c4d in App::Document::_recomputeFeature(App::DocumentObject*) from /home/OB/dev/FCBuild/lib/libFreeCADApp.so+0x1d1
#9  0x7fbaaecd994b in App::Document::recomputeFeature(App::DocumentObject*, bool) from /home/OB/dev/FCBuild/lib/libFreeCADApp.so+0x10b
#10  0x7fba1572c0a0 in PartDesignGui::TaskFeatureParameters::recomputeFeature() from /home/OB/dev/FCBuild/Mod/PartDesign/PartDesignGui.so+0x82
#11  0x7fba157c19fa in PartDesignGui::TaskPipeParameters::onSelectionChanged(Gui::SelectionChanges const&) from /home/OB/dev/FCBuild/Mod/PartDesign/PartDesignGui.so+0x558```

luzpaz, 2022-04-03:
@0penBrain is this OCC issue a known issue or do we need to open an upstream ticket ?

0penBrain, 2022-04-03:
@luzpaz I have no clue about that. 😞

0penBrain, 2022-04-03:
@luzpaz I did a search and that's only ticket/post/... that include `BRepFill_PipeShell::BuildHistory` so most probably we can open an upstream ticket.

luzpaz, 2022-04-03:
Thanks! Posted to [OCC Bugs in the Bugtracker](https://forum.freecadweb.org/viewtopic.php?p=585850#p585850)

chennes, 2022-04-03:
This does not crash when using:
```
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.28615 (Git)
Build type: Debug
Branch: master
Hash: 9a7b411659d4dfcbf6f2c72374d62c7bf83ea0ba
Python 3.8.6+, Qt 5.15.2, Coin 4.0.1, OCC 7.7.0.dev
Locale: English/United States (en_US)
```
(though it does not create the pipe, it produces an error). So it appears that at least the crash has already been fixed upstream.

grubertechnic, 2023-07-27:
@chennes : works flawless here

```
OS: Debian GNU/Linux 11 (bullseye) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33657 (Git)
Build type: Release
Branch: master
Hash: 2a1d4d4994bf3707f7a07aef07e17f5270b4854b
Python 3.9.2, Qt 5.15.2, Coin 4.0.0, Vtk 9.0.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * QuickMeasure 2022.10.28
  * CurvedShapes 1.0.4
  * Curves 0.6.12
```

No crash either, als that the result does not work is expected. The start shape is in a position, where it can be that shifted, that the result is a self intersecting shell. When it is position explicit correct, it works flawless too.

[image]

grubertechnic, 2023-07-29:
@Roy-043 : just to be clear, it is a candidate to close, it works no all as expected.

That the file does not work out of the box is an user error, when corrected it, it works also without any issue.

chennes, 2023-07-30:
I'm closing this as probably-fixed by upstream. If it recurs please re-open. Thanks!

closed 2023-07-30 by chennes