FreeCAD/FreeCAD #6270: CSG file from OpenSCAD does not load. Part.OCCError

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4761

* **Reporter:** ozzee
* **Date submitted:** 10/9/2021
* **FreeCAD version:** 0.2
* **Category:** Bug
* **Status:** confirmed
* **Tags:** #post-to-forum, OpenSCAD

# Original report text

Log shows:
00:02:29  End processing CSG file
00:02:33  Traceback (most recent call last):
  File "C:\Users\gianni\Downloads\FreeCAD_0.20.25645_Win-LPv12.5.4_vc17.x-x86-64\Mod\OpenSCAD\OpenSCADFeatures.py", line 406, in execute
    self.createGeometry(fp)
  File "C:\Users\gianni\Downloads\FreeCAD_0.20.25645_Win-LPv12.5.4_vc17.x-x86-64\Mod\OpenSCAD\OpenSCADFeatures.py", line 448, in createGeometry
    pipe_shell.build()
<class 'Part.OCCError'>: Approx_CurvlinFunc::GetUParameter

# Steps to reproduce

Import attached CSG file.

# Other bug information

* **Priority:** normal
* **Severity:** major
* **Category:** Bug
* **Updated:** 1/25/2022

# Discussion from Mantis ticket

### Comment by openBrain 2021-10-13 14:15
ozzee : could you please open a new thread in our forum ([per our reporting guidelines](https://forum.freecadweb.org/viewtopic.php?f=3&t=5236)) and add link here ? Thx

---

### Comment by chennes 2021-12-29 03:08
The actual bit of failing code is the part that connects the circular part of the funnel to the highly-elongated part:
```
// 'inner'
          linear_extrude(height=25.002, twist=0.0, scale=[0.3, 2.85], $fn=128) {
            polygon(points=[[0.0, 10.0], [-0.4906767432741801, 9.987954562051724], [-0.9801714032956065, 9.95184726672197], [-1.4673047445536165, 9.89176509964781], [-1.950903220161282, 9.807852804032304], [-2.4298017990326386, 9.700312531945439], [-2.9028467725446214, 9.56940335732209], [-3.3688985339221995, 9.415440651830208], [-3.826834323650897, 9.238795325112868], [-4.275550934302819, 9.039892931234434], [-4.713967368259977, 8.819212643483551], [-5.1410274419322155, 8.577286100002722], [-5.55570233019602, 8.314696123025453], [-5.956993044924333, 8.03207531480645], [-6.3439328416364535, 7.730104533627371], [-6.715589548470184, 7.40951125354959], [-7.071067811865475, 7.0710678118654755], [-7.409511253549589, 6.715589548470185], [-7.73010453362737, 6.343932841636455], [-8.032075314806448, 5.956993044924335], [-8.314696123025453, 5.555702330196022], [-8.57728610000272, 5.141027441932218], [-8.81921264348355, 4.713967368259978], [-9.039892931234432, 4.275550934302824], [-9.238795325112868, 3.826834323650899], [-9.415440651830208, 3.3688985339221986], [-9.569403357322088, 2.902846772544624], [-9.700312531945439, 2.429801799032641], [-9.807852804032304, 1.9509032201612861], [-9.89176509964781, 1.467304744553618], [-9.951847266721968, 0.9801714032956083], [-9.987954562051724, 0.49067674327417965], [-10.0, 1.2246467991473533e-15], [-9.987954562051724, -0.4906767432741772], [-9.95184726672197, -0.9801714032956059], [-9.89176509964781, -1.4673047445536158], [-9.807852804032304, -1.9509032201612837], [-9.700312531945439, -2.429801799032638], [-9.56940335732209, -2.902846772544621], [-9.41544065183021, -3.368898533922197], [-9.238795325112868, -3.8268343236508966], [-9.039892931234434, -4.275550934302818], [-8.819212643483551, -4.7139673682599765], [-8.577286100002722, -5.1410274419322155], [-8.314696123025454, -5.55570233019602], [-8.03207531480645, -5.9569930449243325], [-7.730104533627371, -6.343932841636453], [-7.409511253549591, -6.715589548470184], [-7.071067811865477, -7.071067811865475], [-6.715589548470186, -7.409511253549589], [-6.343932841636452, -7.730104533627372], [-5.956993044924332, -8.03207531480645], [-5.555702330196022, -8.314696123025453], [-5.141027441932218, -8.57728610000272], [-4.713967368259978, -8.81921264348355], [-4.275550934302824, -9.039892931234432], [-3.8268343236509033, -9.238795325112864], [-3.3688985339222075, -9.415440651830204], [-2.9028467725446245, -9.569403357322088], [-2.4298017990326413, -9.700312531945439], [-1.9509032201612866, -9.807852804032303], [-1.4673047445536145, -9.89176509964781], [-0.9801714032956045, -9.95184726672197], [-0.4906767432741803, -9.987954562051724], [-1.8369701987210296e-15, -10.0], [0.4906767432741766, -9.987954562051724], [0.9801714032956009, -9.95184726672197], [1.4673047445536107, -9.891765099647811], [1.950903220161283, -9.807852804032304], [2.4298017990326377, -9.700312531945439], [2.9028467725446205, -9.56940335732209], [3.3688985339222044, -9.415440651830206], [3.8268343236509, -9.238795325112866], [4.275550934302822, -9.039892931234434], [4.713967368259976, -8.819212643483551], [5.1410274419322155, -8.577286100002722], [5.555702330196018, -8.314696123025454], [5.956993044924328, -8.032075314806452], [6.343932841636449, -7.730104533627374], [6.7155895484701835, -7.409511253549591], [7.071067811865474, -7.071067811865477], [7.409511253549589, -6.715589548470186], [7.730104533627365, -6.343932841636459], [8.03207531480645, -5.9569930449243325], [8.314696123025453, -5.555702330196022], [8.57728610000272, -5.141027441932219], [8.819212643483548, -4.713967368259979], [9.039892931234432, -4.275550934302825], [9.238795325112864, -3.826834323650904], [9.415440651830208, -3.3688985339222], [9.569403357322088, -2.902846772544625], [9.700312531945439, -2.4298017990326417], [9.807852804032303, -1.9509032201612873], [9.89176509964781, -1.4673047445536238], [9.95184726672197, -0.980171403295605], [9.987954562051724, -0.4906767432741809], [10.0, -2.4492935982947065e-15], [9.987954562051724, 0.490676743274176], [9.95184726672197, 0.9801714032956002], [9.89176509964781, 1.467304744553619], [9.807852804032304, 1.9509032201612824], [9.70031253194544, 2.429801799032637], [9.56940335732209, 2.90284677254462], [9.41544065183021, 3.3688985339221955], [9.238795325112868, 3.8268343236508997], [9.039892931234434, 4.275550934302821], [8.819212643483551, 4.713967368259976], [8.577286100002722, 5.141027441932215], [8.314696123025456, 5.555702330196018], [8.032075314806452, 5.956993044924328], [7.730104533627369, 6.343932841636455], [7.409511253549592, 6.7155895484701835], [7.071067811865477, 7.071067811865474], [6.715589548470186, 7.409511253549588], [6.343932841636459, 7.730104533627365], [5.9569930449243325, 8.03207531480645], [5.555702330196023, 8.314696123025453], [5.141027441932219, 8.57728610000272], [4.71396736825998, 8.819212643483548], [4.275550934302826, 9.039892931234432], [3.8268343236509046, 9.238795325112864], [3.3688985339222004, 9.415440651830208], [2.9028467725446254, 9.569403357322088], [2.429801799032642, 9.700312531945439], [1.9509032201612877, 9.807852804032303], [1.4673047445536245, 9.89176509964781], [0.9801714032956057, 9.95184726672197], [0.49067674327418154, 9.987954562051724], [3.061616997868383e-15, 10.0]]);
          }
```

For some reason a slight twist (approximately 10 degrees) is being applied when this transform is occurring in FreeCAD. (Note to self: that polygon is really just a circle, radius=10 I think -- no idea why it's defined as a polygon here).

---

### Comment by Kunda1 2022-01-22 22:02
CC'ing keithsloan52 on this too

---

### Comment by keithsloan52 2022-01-25 21:37
Chris chennes have you been able to create a simpler test case that shows the same error, as I am not having much luck importing with the csg or scad file just seem to be spinning.

---

### Comment by chennes 2022-01-25 21:41
Did you try that snippet I posted just above? My recollection is that was enough to get the failure: it turned out to be pretty easy to reproduce, but I did not have a chance to debug it.

---

### Comment by keithsloan52 2022-01-25 21:59
Not seeing any problem with Cut and paste of linear_extrude.

chennes, 2022-02-12:
OK, good -- the OCC error is because the geometry becomes self-intersecting, the real problem is the incorrect twist being applied. I think I can track this one down...

KeithSloan, 2022-02-12:
Okay I added a couple of lines to OpenSCADFeatures.py to output the shapes as Brep files.
I attach here
[upper_face.brep.txt](https://github.com/FreeCAD/FreeCAD/files/8054611/upper_face.brep.txt)
[aux_spine.brep.txt](https://github.com/FreeCAD/FreeCAD/files/8054615/aux_spine.brep.txt)
[lower_face.brep.txt](https://github.com/FreeCAD/FreeCAD/files/8054616/lower_face.brep.txt)

The spine looks like just a line so I guess that might be okay

The top and bottom face look okay. 
[aux_spine.brep.txt](https://github.com/FreeCAD/FreeCAD/files/8054613/aux_spine.brep.txt)
e

chennes, 2022-02-12:
If I replace the polygon with the circle it is approximating, everything works fine:
```
          linear_extrude(height=25.002, twist=0.0, scale=[0.3, 2.85], $fn=128) {
             circle(r=10);
          }
```

KeithSloan, 2022-02-12:
I cannot download the original scad and csg from the tracker as it hangs the browser on my Mac.

Does the scad have a circle and the CSG the polygon?

The code does not know the polygon can be swapped for a circle and we need the function to work for any valid polygon ?

Or are you just saying in the process of debugging.

chennes, 2022-02-12:
The OpenSCAD also has a polygon -- it's just recognizably a circle, so to debug I tried switching. So OCC doesn't like that high-segment-count circle approximation.

KeithSloan, 2022-02-12:
Well with FreeCAD 0.19.2 it does not crash but the version of the importer does not support your updates for linear_extrude with scale option. Maybe an OCC regression? what happens if you remove the scale=[0.3, 2.85] 

OS: macOS 10.15
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24291 (Git)
Build type: Release
Branch: (HEAD detached at 0.19.2)
Hash: 7b5e18a0759de778b74d3a5c17eba9cb815035ac
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: C/Default (C)

chennes, 2022-02-12:
It works fine without the scale.

chennes, 2022-02-13:
>  Am I grasping at straws?

Well, probably, but that's kind of what we're reduced to here 🤷 I'll try to poke at it next week.

luzpaz, 2022-09-12:
soft bump

chennes, 2022-12-02:
Aha! Figured this one out. It's not obvious because of floating point imprecision, but this "polygon" actually has duplicated first and last points. OpenSCAD tolerates it because it's just a super-short line segment, but OCC doesn't know what to do with something whose length is shorter than `confusion`. Maybe we should auto-detect this problem and drop the final point?

luzpaz, 2023-02-26:
ping @KeithSloan for follow-up

KeithSloan, 2023-02-26:
Well with this and #6271 I think we should catch and put out a meaningful message.

I realise that OpenSCAD probably does not care but wonder if they would like to be advised.

1) Rational the end result if often a 3D printed object, so although OpenSCAD renders okay, might it impact on a Slicer.
2)My understanding and I may well be wrong but that there are on going efforts to speed up rendering etc. Think I saw a Gsoc idea for more GPU exploitation etc, are these issues likely to crop up for others?

Is it worth at least raising  with Paul Torten (t-paul)? Also as they heavily use OpenCSG Florian Kirsch and Hasso-Plattner-Institute Potsdam. mail@opencsg.org

chennes, 2023-02-26:
What do you propose should be done? IMO, it's just bad input: OpenSCAD/OpenCSG tolerates it, OCCT doesn't. I'm not sure what I'd be asking Paul to do.

KeithSloan, 2023-02-26:
I will email OpenCSG, say we have found some things that OpenCSG tolerates but we have problem with OCCT, are they interested in being informed.

closed 2023-02-27 by chennes