FreeCAD/FreeCAD #6271: CSG file from OpenScad does not render correctly

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4762

* **Reporter:** ozzee
* **Date submitted:** 10/9/2021
* **FreeCAD version:** 0.2
* **Category:** Bug
* **Status:** confirmed
* **Tags:** #post-to-forum

# Original report text

Attached CSG model file from OpenScad (see attached .scad file) appears to have a number of errors when rendering in FreeCAD.

1. The difference operations do not seem to be respected.
2. The multmatrix transformations do not seem to produce the same transforms in OpenSCAD (note many objects in the wrong places).

# Additional information

00:15:28  This function will be deprecated. Please use 'make_layer'.
00:15:28  ----------------
00:15:28  Layer
00:15:28  name: 0
00:15:28  line_color: (0.0, 0.0, 0.0)
00:15:28  line_width: 2.0
00:15:28  draw_style: Solid
00:15:28  transparency: 0
00:15:28  drawing 272 lines...
00:15:28  skipping texts...
00:15:28  skipping dimensions...
00:15:28  skipping points...
00:15:28  skipping leaders...
00:15:28  skipping hatches...
00:15:28  skipping *blocks...
00:15:28  successfully imported C:\Users\gianni\AppData\Local\Temp\fc-12712-528727-000001.dxf
00:15:28  End processing CSG file
00:15:28  Traceback (most recent call last):
  File "C:\Users\gianni\Downloads\FreeCAD_0.20.25645_Win-LPv12.5.4_vc17.x-x86-64\Mod\OpenSCAD\OpenSCADFeatures.py", line 406, in execute
    self.createGeometry(fp)
  File "C:\Users\gianni\Downloads\FreeCAD_0.20.25645_Win-LPv12.5.4_vc17.x-x86-64\Mod\OpenSCAD\OpenSCADFeatures.py", line 448, in createGeometry
    pipe_shell.build()
<class 'Part.OCCError'>: Approx_CurvlinFunc::GetUParameter
00:15:31  <Exception> Input shape is null
00:15:31  <Exception> Input shape is null
00:15:46  Traceback (most recent call last):
  File "C:\Users\gianni\Downloads\FreeCAD_0.20.25645_Win-LPv12.5.4_vc17.x-x86-64\Mod\OpenSCAD\OpenSCADFeatures.py", line 406, in execute
    self.createGeometry(fp)
  File "C:\Users\gianni\Downloads\FreeCAD_0.20.25645_Win-LPv12.5.4_vc17.x-x86-64\Mod\OpenSCAD\OpenSCADFeatures.py", line 448, in createGeometry
    pipe_shell.build()
<class 'Part.OCCError'>: Approx_CurvlinFunc::GetUParameter
00:15:46  <Exception> Input shape is null
00:15:46  <Exception> Input shape is null
00:16:50  Traceback (most recent call last):
  File "C:\Users\gianni\Downloads\FreeCAD_0.20.25645_Win-LPv12.5.4_vc17.x-x86-64\Mod\OpenSCAD\OpenSCADFeatures.py", line 406, in execute
    self.createGeometry(fp)
  File "C:\Users\gianni\Downloads\FreeCAD_0.20.25645_Win-LPv12.5.4_vc17.x-x86-64\Mod\OpenSCAD\OpenSCADFeatures.py", line 448, in createGeometry
    pipe_shell.build()
<class 'Part.OCCError'>: Approx_CurvlinFunc::GetUParameter
00:16:50  <Exception> Input shape is null
00:16:50  <Exception> Input shape is null
00:16:53  Traceback (most recent call last):
  File "C:\Users\gianni\Downloads\FreeCAD_0.20.25645_Win-LPv12.5.4_vc17.x-x86-64\Mod\OpenSCAD\OpenSCADFeatures.py", line 406, in execute
    self.createGeometry(fp)
  File "C:\Users\gianni\Downloads\FreeCAD_0.20.25645_Win-LPv12.5.4_vc17.x-x86-64\Mod\OpenSCAD\OpenSCADFeatures.py", line 448, in createGeometry
    pipe_shell.build()
<class 'Part.OCCError'>: Approx_CurvlinFunc::GetUParameter
00:16:53  <Exception> Input shape is null
00:16:53  <Exception> Input shape is null
00:16:56  Traceback (most recent call last):
  File "C:\Users\gianni\Downloads\FreeCAD_0.20.25645_Win-LPv12.5.4_vc17.x-x86-64\Mod\OpenSCAD\OpenSCADFeatures.py", line 406, in execute
    self.createGeometry(fp)
  File "C:\Users\gianni\Downloads\FreeCAD_0.20.25645_Win-LPv12.5.4_vc17.x-x86-64\Mod\OpenSCAD\OpenSCADFeatures.py", line 448, in createGeometry
    pipe_shell.build()
<class 'Part.OCCError'>: Approx_CurvlinFunc::GetUParameter
00:16:56  <Exception> Input shape is null
00:16:56  <Exception> Input shape is null
00:24:41  Traceback (most recent call last):
  File "C:\Users\gianni\Downloads\FreeCAD_0.20.25645_Win-LPv12.5.4_vc17.x-x86-64\Mod\OpenSCAD\OpenSCADFeatures.py", line 406, in execute
    self.createGeometry(fp)
  File "C:\Users\gianni\Downloads\FreeCAD_0.20.25645_Win-LPv12.5.4_vc17.x-x86-64\Mod\OpenSCAD\OpenSCADFeatures.py", line 448, in createGeometry
    pipe_shell.build()
<class 'Part.OCCError'>: Approx_CurvlinFunc::GetUParameter
00:24:41  <Exception> Input shape is null
00:24:41  <Exception> Input shape is null
00:24:43  Traceback (most recent call last):
  File "C:\Users\gianni\Downloads\FreeCAD_0.20.25645_Win-LPv12.5.4_vc17.x-x86-64\Mod\OpenSCAD\OpenSCADFeatures.py", line 406, in execute
    self.createGeometry(fp)
  File "C:\Users\gianni\Downloads\FreeCAD_0.20.25645_Win-LPv12.5.4_vc17.x-x86-64\Mod\OpenSCAD\OpenSCADFeatures.py", line 448, in createGeometry
    pipe_shell.build()
<class 'Part.OCCError'>: Approx_CurvlinFunc::GetUParameter
00:24:43  <Exception> Input shape is null
00:24:43  <Exception> Input shape is null

# Steps to reproduce

Import the attached CSG file.

# Other bug information

* **Priority:** normal
* **Severity:** major
* **Category:** Bug
* **Updated:** 12/29/2021

# Discussion from Mantis ticket

### Comment by openBrain 2021-10-13 14:17
ozzee : could you please open a new thread in our forum ([per our reporting guidelines](https://forum.freecadweb.org/viewtopic.php?f=3&t=5236)) and add link here ? Thx

---

### Comment by chennes 2021-12-29 02:51
In both the SCAD and CSG files there are many empty unions -- it's not clear what their intended purpose is, but FreeCAD is not currently treating them the same way that OpenSCAD does, which leads to a divergence between the two results. I will try to figure out what OpenSCAD does with those elements.

luzpaz, 2022-04-21:
Posted to the tracking forum thread: [Problem OpenSCAD scad & csg files - PLEASE Add them to this thread - Thanks](https://forum.freecadweb.org/viewtopic.php?p=590079#p590079)

luzpaz, 2022-06-06:
@chennes @KeithSloan care to notate this ticket with results from your testing ?

KeithSloan, 2022-06-07:
With the alternate importer I am seeing
````
08:39:20  Verbose = False
08:39:20  Processing : /Users/keithsloan/Downloads/Test-files-for-6271/anchorcad_RaspberryPi4Case_bottom_example.scad
08:39:20  Calling OpenSCAD
08:39:20  Using Alternate OpenSCAD Importer
08:39:20  0
08:39:20  0
08:39:20  <Exception> Input shape is null
08:39:20  Recompute failed : union
08:39:20  4
08:39:20  17
08:39:20  0
08:39:20  0
08:39:20  0
08:39:20  0
08:39:20  0
08:39:20  0
08:39:20  0
08:39:20  0
08:39:20  textmsg : text ( text="-648BD", size = 5, spacing = "1", font = "", direction = "ltr", language = "en", script = "Latn", halign = "left", valign = "bottom", $fn = 0, $fa = 12, $fs = 2);
08:39:20  Fontconfig error: Cannot load default config file
Fontconfig warning: ignoring UTF-8: not a valid region tag
Geometries in cache: 1
Geometry cache size in bytes: 4816
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Total rendering time: 0:00:00.028
   Top level object is a 2D object:
   Contours:       13

08:39:20  opening /var/folders/cj/z259fmwd41dgzl8mq8nppwd40000gn/T/fc-21211-756058-000002.dxf...
08:39:20  drawing 272 lines...
08:39:20  Joining geometry...
08:39:20  272 edges to join
08:39:26  skipping texts...
08:39:26  skipping dimensions...
08:39:26  skipping points...
08:39:26  skipping leaders...
08:39:26  skipping hatches...
08:39:26  skipping *blocks...
08:39:26  done processing
08:39:26  successfully imported /var/folders/cj/z259fmwd41dgzl8mq8nppwd40000gn/T/fc-21211-756058-000002.dxf
08:39:26  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/Users/keithsloan/Library/Preferences/FreeCAD/Mod/OpenSCAD_Alt_Import/freecad/OpenSCAD_Alt_Import/importCSG.py", line 96, in open
    processcsg(tmpfile)
  File "/Users/keithsloan/Library/Preferences/FreeCAD/Mod/OpenSCAD_Alt_Import/freecad/OpenSCAD_Alt_Import/importCSG.py", line 157, in processcsg
    result = parser.parse(f.read())
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.9/site-packages/ply/yacc.py", line 333, in parse
    return self.parseopt_notrack(input, lexer, debug, tracking, tokenfunc)
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.9/site-packages/ply/yacc.py", line 1120, in parseopt_notrack
    p.callable(pslice)
  File "/Users/keithsloan/Library/Preferences/FreeCAD/Mod/OpenSCAD_Alt_Import/freecad/OpenSCAD_Alt_Import/importCSG.py", line 1326, in p_text_action
    p[0] = [processTextCmd(t)]
  File "/Users/keithsloan/Library/Preferences/FreeCAD/Mod/OpenSCAD_Alt_Import/freecad/OpenSCAD_Alt_Import/importCSG.py", line 959, in processTextCmd
    face = importDXFface(tmpfilename,None,None)
  File "/Users/keithsloan/Library/Preferences/FreeCAD/Mod/OpenSCAD_Alt_Import/OpenSCAD2Dgeom.py", line 497, in importDXFface
    faces = edgestofaces(edges)
  File "/Users/keithsloan/Library/Preferences/FreeCAD/Mod/OpenSCAD_Alt_Import/OpenSCAD2Dgeom.py", line 410, in edgestofaces
    return Overlappingfaces(facel).makeshape()
  File "/Users/keithsloan/Library/Preferences/FreeCAD/Mod/OpenSCAD_Alt_Import/OpenSCAD2Dgeom.py", line 229, in makeshape
    return fusefaces(faces)
  File "/Users/keithsloan/Library/Preferences/FreeCAD/Mod/OpenSCAD_Alt_Import/OpenSCAD2Dgeom.py", line 366, in fusefaces
    return reduce(lambda p1,p2: p1.fuse(p2),faces)
<class 'TypeError'>: reduce() of empty sequence with no initial value

Looking at the union I see

![Image 07-06-2022 at 08 44](https://user-images.githubusercontent.com/2291247/172325688-a9fba19b-ea83-4981-9041-8e8c29aed22d.jpg)

Which is a union of 18 items, I suspect that don't overlap.

I also note there is an error with the text handling.

luzpaz, 2022-09-12:
Does 88ee6bb47f14 impact this ticket at all ?

KeithSloan, 2022-09-20:
"Does https://github.com/FreeCAD/FreeCAD/commit/88ee6bb47f1427e0b9424d3f4f7313866d28449d impact this ticket at all ?"

Might well do, maybe Chris @chennes could test to see if things are different with the latest version. I don't have ready access to latest.

chennes, 2022-12-02:
I have this narrowed down to a minimum failing example. The following CSG is rendered differently by OpenSCAD and FreeCAD, and actually I don't understand how it works in OpenSCAD...
```
linear_extrude(height = 9, center = false, convexity = 1, scale = [1, 1], $fn = 0, $fa = 12, $fs = 2) {
	polygon(points = [[0, 0], [-1.005, 0], [-1.005, 1.28383], [-0.5025, 0.781331], [0, 0.781331], [0.5025, 0.781331], [1.005, 1.28383], [1.005, 0], [1.005, -1.28383], [0.5025, -0.781331], [0, -0.781331], [-0.5025, -0.781331], [-1.005, -1.28383], [-1.005, 0]], paths = undef, convexity = 1);
}
```
Notice that the first point is `[0,0]` -- that doesn't make any sense to me, though, given that the shape as extruded in OpenSCAD doesn't have (or need) a vertex there. Somehow OpenSCAD is just eliminating the degenerate point, whereas FreeCAD tries to create the geometry as specified (of course), which fails. Can we reach out to anyone in the OpenSCAD community and find out what algorithm is being applied here to eliminate the presence of that vertex?

chennes, 2022-12-02:
OK, I got some help on the OpenSCAD IRC channel, and it turns out that even in OpenCAD you are ending up with a zero-width slit in the geometry. It's just that in this particular instance you get lucky, and CGAL doesn't care about it and can do the linear_extrude anyway. I'm not sure how feasible it is for us to detect and handle this kind of degeneracy.

luzpaz, 2023-02-26:
> I'm not sure how feasible it is for us to detect and handle this kind of degeneracy.  

Can we rename the ticket to reflect this edge-case ? or is a "Wont Fix"

chennes, 2023-02-26:
IMO it is wontfix--this is really a bug in the input file that OpenSCAD is accidentally handling the way the designer wanted it to.

luzpaz, 2023-02-26:
@KeithSloan volunteered to take action on this in: https://github.com/FreeCAD/FreeCAD/issues/6270#issuecomment-1445384517

closed 2023-02-27 by chennes

luzpaz, 2023-02-28:
@KeithSloan Thanks for facilitating the potential to *open up some avenues of cooperation in the longer term* with other FLOSS projects!