FreeCAD/FreeCAD #6272: [MSVC][Build][Permissive-][std:c++latest] FreeCAD failed to build on MSVC

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4765

* **Reporter:** QuellaZhang
* **Date submitted:** 10/15/2021
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** acknowledged
* **Tags:** 

# Original report text

Hi All,

The MSVC team recently added FreeCAD as part of RWC testing to detect compiler regressions. And we found some errors when building FreeCAD with '/permissive-' and '/std:c++latest' option. Can you help to fix these issues? Thanks.

I noticed that some of these errors are come from Boost library, and these issue have been fixed on Boost latest version. After I fixed Boost errors, more errors were exposed by the project itself.

# Steps to reproduce

1. open VS2019 x64 tools command
2. git clone https://github.com/FreeCAD/FreeCAD.git F:\gitP\FreeCAD\FreeCAD
3. download https://github.com/apeltauer/FreeCAD/releases/tag/LibPack_12.5.2 F:\gitP\FreeCAD\tools\FreeCADLibs_12.5.2_x64_VC17
4. mkdir F:\gitP\FreeCAD\FreeCAD\build_amd64 && cd F:\gitP\FreeCAD\FreeCAD\build_amd64
5. set _CL_= /permissive- or set _CL_= /std:c++latest
6. set PATH=F:\gitP\FreeCAD\tools\FreeCADLibs_12.5.2_x64_VC17\bin;%PATH%
7. set PATH=F:\gitP\FreeCAD\tools\FreeCADLibs_12.5.2_x64_VC17\bin\Scripts;%PATH%
8. cmake -G "Visual Studio 16 2019" -A x64 -DCMAKE_SYSTEM_VERSION=10.0.18362.0 -DFREECAD_LIBPACK_DIR=F:\gitP\FreeCAD\tools\FreeCADLibs_12.5.2_x64_VC17 -DBUILD_FEM_NETGEN=OFF -DFREECAD_RELEASE_PDB=OFF -DFREECAD_RELEASE_SEH=OFF -DCMAKE_BUILD_TYPE=Release .. 
9. msbuild /m /p:Platform=x64 /p:Configuration=Release FreeCAD.sln /t:Rebuild

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 12/22/2021

# Discussion from Mantis ticket

### Comment by chennes 2021-11-01 16:50
Thank you for the report -- we'll look into these issues.

---

### Comment by chennes 2021-11-04 02:42
https://github.com/FreeCAD/FreeCAD/commit/833868bd1394862f6fab079fa74685f09675129a fixes one of the issues.

---

### Comment by chennes 2021-11-06 15:57
https://github.com/FreeCAD/FreeCAD/commit/6d2aa6acc9f58d8f8cf0d4744d9a0ed86ac18ed5 fixes another.

---

### Comment by uwestoehr 2021-12-20 10:02
https://github.com/FreeCAD/FreeCAD/commit/6ad156b6b77 fixes another one

---

### Comment by uwestoehr 2021-12-22 00:40
and another one: https://github.com/FreeCAD/FreeCAD/commit/5fba9cec

luzpaz, 2022-04-15:
CC @chennes @donovaly what the status of this ticket?

donovaly, 2022-04-15:
I think only @wwmayer has the overview here. However, not pressing thus postponing.

xtemp09, 2023-10-10:
I've just compiled FreeCAD using gcc 12.3 with C++20 option on Ubuntu 22.04. If I set `CMAKE_CXX_STANDARD` to 20, it does not affect `BUILD_ENABLE_CXX_STD`. I don't know what the latter option is for.

Perhaps, MSVC can compile FreeCAD with `/std:c++latest` as well.

chennes, 2023-10-10:
I'm actively working on this one because Qt6 requires `/permissive-` with MSVC. I've got it working, actually, I was just polishing some other things before submitting the PR. This should be fixed within the week.

luzpaz, 2023-11-08:
@chennes does this ticket get fixed by #11014 ?

chennes, 2023-11-08:
At least the /permissive- part: I'm not testing with c++latest yet.

chennes, 2024-08-27:
To get `/std:c++latest` working the following need to be addressed:

1. Salome SMESH tries to detect whether it can use `::set_unexpected` (from pre-c++17) but assumes that if _MSC_VER is defined then it can. With the /std:c++latest flag, it is wrong. Utls_ExceptHandlers.hxx line 41 must be updated.
2. Same as above, but w.r.t. `::set_terminate`, on line 51.
3. Same as the previous two, but in SMESH_ExceptHandlers.hxx
4. In src/App/Application.cpp, we use `::set_enexpected" if _MSC_VER is detected, which is not a good enough detection now.
5. src/Gui/Macro.cpp: `import` is now a reserved word, and cannot be used as a variable name on line 83.

All told this should be pretty straightforward.

chennes, 2025-02-16:
These are ongoing low-priority issues, but also a pretty simple set of changes if a new dev wants to get involved. I've added a Good First Issue tag.

luzpaz, 2026-09-29:
@chennes is this still valid ?

chennes, 2026-10-02:
No, we can compile with std:c++latest now (in fact, we do so).

closed 2026-10-02 by chennes