FreeCAD/FreeCAD #6274: Source code installation failed

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4768

* **Reporter:** jack9603301
* **Date submitted:** 10/18/2021
* **FreeCAD version:** 0.2
* **Category:** Bug
* **Status:** new
* **Tags:** 

# Original report text

see: https://forum.freecadweb.org/viewtopic.php?f=4&t=63002&p=540842#p540842

This appears to be caused by `StartGui.so` and `Start.so` not having the permissions it should have, when executing the following command:

```
cd /usr/lib/x86_64-linux-gnu/
find -type f -name "*.so" | xargs -I {} sudo chmod a+x {}
```

It is normal for

# Steps to reproduce

To compile and install, run the following command:

```
cmake .. -DCMAKE_INSTALL_PREFIX=/usr -DPYTHON_EXECUTABLE=/usr/bin/python3 -DFREECAD_USE_PCL=ON -DFREECAD_USE_PYBIND11=ON -DCMAKE_BUILD_TYPE=Release
make
sudo make install
```

# FreeCAD Info

```

OS: Ubuntu 21.04 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.
Build type: Release
Branch: unknown
Hash: 0926a4148bcff11249fd4f56bc30256102ffe105
Python version: 3.9.5
Qt version: 5.15.2
Coin version: 4.0.0
OCC version: 7.5.2
Locale: Chinese/China (zh_CN)

```

# Other bug information

* **Priority:** normal
* **Severity:** trivial
* **Category:** Bug
* **OS: Kubuntu 21.04**
* **Platform:** Linux
* **Updated:** 10/19/2021

# Discussion from Mantis ticket

### Comment by jack9603301 2021-10-18 19:16
Well, I should add that it doesn't seem to be caused by a permissions issue, but rather by my accidental execution in the following path. Strange..
```
cd /usr/lib/x86_64-linux-gnu/
FreeCAD
```

---

### Comment by jack9603301 2021-10-18 19:22
The problem seems to be related to the fact that the associated dynamic libraries are not python3 working paths (such as /usr/mod)...

```
FreeCAD  -M /usr/lib/x86_64-linux-gnu
```

---

### Comment by jack9603301 2021-10-19 10:22
Run the following command to start:
```
FreeCAD  -M /usr/lib/x86_64-linux-gnu
```

I think this is a mistake, and if those dynamic library were in the same directory as python mods, then everything would be fine. 
Perhaps the people concerned just need to modify the Cmake installation configuration to fix this error.
This problem has been tested.
Although the cause of the error was misjudged once, it was indeed a flaw in the installation process 
This is not a serious error because it can be resolved by -M, but it is a defect. 
Some people like to compile and install (maybe to track and participate in freecad development in the future, maybe to track the latest changes in development branches, participate in the community, or simply want to compile themselves to meet 100% compatible ABI, but in any case, they want to compile and install)
As the most commonly used manual installation method of Linux, especially the operation and maintenance personnel, the use of compilation and installation should not be hindered by defects (tracing the exact directory of each dynamic library often wastes redundant time)

ps: Can someone help me move this bug report to the right project? It should not belong to part, but to freecad

yorikvanhavre, 2022-02-11:
`FreeCAD  -M /usr/lib/x86_64-linux-gnu` indeed should not be needed.
On most distributions, the FreeCAD built-in python modules (libXXX.so files) are indeed installed in /usr/lib/x86_64-linux-gnu so that should work... I'm wondering what is missing here?

waebbl, 2022-04-11:
On Gentoo we have the same issue, if built with `CMAKE_INSTALL_PREFIX=/usr`[1]. I made attempts for the Gentoo Linux package to relocate the Mods into `/usr/lib/freecad` when built with `CMAKE_INSTALL_PREFIX=/usr`, by patching all the CMake DESTINATION options in the Mods install targets[2,3]. But there's actual C++ and/or Python code, which expects the Mod directory just below prefix and thus looks for `/usr/Mod`.

IMO the point here is that FreeCAD's build system isn't compatible with `CMAKE_INSTALL_PREFIX=/usr`. It expects `CMAKE_INSTALL_PREFIX=/usr/lib/freecad` and seems to work only with such prefix.

See also[4] for a forum discussion on this.

[1] https://bugs.gentoo.org/831086
[2] https://github.com/waebbl/waebbl-gentoo/issues/354
[3] https://github.com/waebbl/waebbl-gentoo/issues/355
[4] https://forum.freecadweb.org/viewtopic.php?f=4&t=30596

wwmayer, 2022-04-18:
With 6bc65bdd99 it's now possible to launch FreeCAD without the -M option.

wwmayer, 2022-04-20:
> can this ticket be closed?

The actual bug is closed but the points @waebbl brought up should still be discussed. This can also be done in a separate forum thread.

luzpaz, 2022-04-20:
@wwmayer we could also do this
[image]

luzpaz, 2022-05-03:
@waebbl do you mind opening a new ticket so we can track this issue? :pray:

waebbl, 2022-05-03:
Done. Feel free to change the title if desired.

luzpaz, 2022-05-03:
Thanks! Closing ticket

closed 2022-05-03 by luzpaz