FreeCAD/FreeCAD #6275: Changing TC with same tool number but different spindle speed doesn't produce correct gcode

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4769

* **Reporter:** sliptonic
* **Date submitted:** 10/18/2021
* **FreeCAD version:** 0.2
* **Category:** Bug
* **Status:** assigned
* **Tags:** 

# Original report text

The user may desire to run the same tool at different feeds and speeds within a job.  This is achieved by creating two tool controllers and using the same tool number for both.  During post processing, the tool number of the new tool is compared to the old one and if they are the same, the tool change code is suppressed.  This is correct behavior.  However, if the spindle speed changes, an M3 (or M4) with the new spindle speed should be emitted and it is not.

# Additional information

https://forum.freecadweb.org/viewtopic.php?f=15&t=62762&start=10

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 10/18/2021

Russ4262, 2022-04-07:
0.20 pre-release analysis with Revision 28638:
 
This is a candidate in my opinion for correction in 0.20.  I can believe the re-use of a tool controller with different speeds occurring with some frequency in production lines.

J-Dunn, 2022-04-10:
Nice. This may also relate to LASER cutters / pyrography where spindle "speed" is used as the LASER power setting. Changing power setting from one op to another would seem a likely requirement.

PR #6726 (2022-04-18): [Path] Fixes #6275  Output for multiple TCs with same ToolNumber
Make PathPost emit tool controllers even if tool number is same.

- [ ]  Your pull request is confined strictly to a single module. That is, all the files changed by your pull request are either in `App`, `Base`, `Gui` or one of the `Mod` subfolders. If you need to make changes in several locations, make several pull requests and wait for the first one to be merged before submitting the next ones
- [ ]  In case your pull request does more than just fixing small bugs, make sure you discussed your ideas with other developers on the [... cut]
commit 27efdf9c (2022-04-18): Fixes #6275

Make PathPost emit tool controllers even if tool number is same.
commit 65013e85 (2022-04-18): Merge pull request #6726 from sliptonic/bug/2TCspinspeed

[Path] Fixes #6275  Output for multiple TCs with same ToolNumber
closed 2022-04-18 by sliptonic