FreeCAD/FreeCAD #6276: Import Stp file Failed And FreeCAD Crashed

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4771

* **Reporter:** DevWang
* **Date submitted:** 10/20/2021
* **FreeCAD version:** 0.2
* **Category:** Bug
* **Status:** feedback
* **Tags:** 

# Original report text

The Stp file Which export by NX 12.0, Import the stp file to FreeCAD in Debug Model , first asset error, then FreeCAD Crashed.

# Steps to reproduce

Step1: Import the Stp file in to FreeCAD; 

# FreeCAD Info

```
OS: Windows 10 Version 1909
Word size of FreeCAD: 64-bit
Version: 0.20.25836 (Git)
Build type: Debug
Branch: master
Hash: e94d7c32b53863b33462864405dfc508a60eae68
Python version: 3.8.6+
Qt version: 5.15.1
Coin version: 4.0.1
OCC version: 7.5.0
Locale: Chinese/China (zh_CN)
```

# Other bug information

* **Priority:** high
* **Severity:** crash
* **Category:** Bug
* **OS: Win10 **
* **Platform:** Windows
* **Updated:** 10/29/2021

# Discussion from Mantis ticket

### Comment by jnxd 2021-10-29 18:40
Not reproducible with the following build, but there appear to be artifacts:

OS: Ubuntu 20.04.3 LTS (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 0.20.26203 (Git)
Build type: Unknown
Branch: fem-constraint-qol
Hash: 3a5c5576ea269b8bae14325db844e4d9201a7fa4
Python version: 3.8.10
Qt version: 5.12.8
Coin version: 4.0.0
OCC version: 7.5.2
Locale: English/United States (en_US)

---

### Comment by jnxd 2021-10-29 18:43
DevWang I see your version is around 500 commits behind master. Please verify with the latest build, and if the bug is still there, please make a forum post as described here: https://wiki.freecadweb.org/Tracker.

luzpaz, 2022-04-11:
No crash, but i see some artifacts as well (but this is not uncommon AFAICT)
[image]

```
OS: Ubuntu Core 20 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.28671 (Git)
Build type: Release
Branch: master
Hash: aea110ba15897661a59f04ec52c701a00daea430
Python 3.8.10, Qt 5.15.3, Coin 4.0.0, OCC 7.6.1
Locale: English/United States (en_US)
Installed mods: 
  * ToolbarStyle
  * Assembly4 0.11.10
  * stickframe
  * sheetmetal 0.2.49
  * pinger
  * Assembly3 0.11.3
  * Behave-Dark-Colors 0.0.1
  * Curves 0.4.2
  * Ship
  * BIM 2021.12.0
  * A2plus 0.4.56
```

donovaly, 2022-04-12:
> @donovaly can you reproduce?

I see the same as you, no crash but artifacts

luzpaz, 2022-04-12:
@donovaly closing ticket.

closed 2022-04-12 by luzpaz