FreeCAD/FreeCAD #6277: Red screen after any 3D View interaction

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4772

* **Reporter:** Sylva1n
* **Date submitted:** 10/24/2021
* **FreeCAD version:** 0.2
* **Category:** Bug
* **Status:** new
* **Tags:** 

# Original report text

Any action that refreshes 3D View (move Camera, select Object... ) leads to a freeze or red screen. Freecad can be closed without crash.

Note that I use a daily Freecad running inside a ParallelsDesktop Ubuntu 20.04 virtual machine (since about one year). 

# Steps to reproduce

Any action leading to refresh 3D View.

# FreeCAD Info

```
OS: Ubuntu 20.04.3 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.20.
Build type: Release
Branch: unknown
Hash: 123906034cf0671a8838f3b44f858d6ad4989ea8
Python version: 3.8.10
Qt version: 5.12.8
Coin version: 4.0.0
OCC version: 7.5.2
Locale: English/United States (en_US)

```

# Other bug information

* **Priority:** high
* **Severity:** major
* **Category:** Bug
* **OS: Ubuntu 20.04**
* **Platform:** x86
* **Updated:** 10/24/2021

luzpaz, 2022-04-15:
@PLChris @donovaly have you seen any similar bug reports on the forum ?

donovaly, 2022-04-15:
> have you seen any similar bug reports on the forum ?

No.
@Sylva1n , does the problem still exist? If so, please open a forum thread that others can hep. If the bug can be reproduced, please reopen this bug.

closed 2022-04-15 by donovaly