FreeCAD/FreeCAD #6278: Duplicate object results in <Exception> Object can only be in a single GeoFeatureGroup

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4775

* **Reporter:** kisolre
* **Date submitted:** 10/31/2021
* **FreeCAD version:** 0.2
* **Category:** Bug
* **Status:** new
* **Tags:** 

# Original report text

When duplicating a PD body object sometimes features from the newly created body are mapped to the duplicated body.

# Additional information

[Forum thread](https://forum.freecadweb.org/viewtopic.php?t=44534)

# Steps to reproduce

Start FreeCAD.
Create new Body.
Draw something in a sketch, pad sketch.
Select Body.
Invoke Duplicate selected object, confirm dialog.
Error is thrown:
```
16:29:00  Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'RuntimeError'>: Object can only be in a single GeoFeatureGroup```

and sketch from the second body is mapped to a plane in the first body.

# FreeCAD Info

```
OS: Windows 8.1 Version 6.3 (Build 9600)
Word size of FreeCAD: 64-bit
Version: 0.20.26255 (Git)
Build type: Release
Branch: master
Hash: 2a9fd08e7f74630c1ad25ec06d984ce06d609d13
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: Bulgarian/Bulgaria (bg_BG)

```

# Other bug information

* **Priority:** normal
* **Severity:** major
* **Category:** Bug
* **OS: Windows **
* **Platform:** PC
* **Updated:** 10/31/2021

kisolre, 2022-06-05:
Still shows the same error on PD Body duplication.
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.29035 (Git)
Build type: Release
Branch: master
Hash: 7e638a624e0fad12691a4ac1dd82b57d2dd1146a
Python 3.8.6+, Qt 5.15.2, Coin 4.0.1, OCC 7.5.3
Locale: Bulgarian/Bulgaria (bg_BG)
Installed mods: 
  * 3DfindIT 1.2.0
  * A2plus 0.4.56
  * AirPlaneDesign 0.4.0
  * AnimationFreeCAD 1.0.0
  * ArchTextures
  * Assembly3 0.11.3
  * Assembly4 0.11.12
  * BIM 2021.12.0
  * BOLTSFC
  * CfdOF 1.16.4
  * CubeMenu
  * CurvedShapes 1.0.3
  * Curves 0.3.0
  * CycloidGearBox
  * Defeaturing
  * Design456
  * DesignSPHysics 0.6.1-2201-31-01
  * dodo
  * DynamicData 2.46.0
  * ExplodedAssembly
  * fasteners 0.3.41
  * FCGear 1.0.0
  * FCPDWorkbench 0.2.0
  * FEM_FrontISTR 0.1.0
  * frame
  * freecad.xray 2022.4.17
  * FreeCAD_mcm_addon
  * Help 1.0.0-alpha
  * InvGears 0.1.2
  * kicadStepUpMod 10.12.4
  * lattice2 1.0.0
  * Launcher
  * MakerWorkbench
  * Manipulator 1.4.3
  * Mechatronic
  * MeshRemodel 1.8919.0
  * MnesarcoUtils 0.1.0
  * Nikra-DAP
  * nurbs
  * OpticsWorkbench 1.0.8
  * ose-piping
  * parts_library
  * PlacementTools
  * Plot 2022.4.17
  * Pyramids-and-Polyhedrons
  * Reinforcement
  * Render 2022.1.0
  * Rocket 2.4.1
  * SearchBar 1.0.1
  * SelectorToolbar
  * sheetmetal 0.2.49
  * Ship 2022.4.11
  * Silk
  * SlopedPlanesMacro
  * StemfieWB
  * symbols_library
  * ThreadProfile 1.81.0
  * toSketch
  * Woodworking 1.0.0
  * workfeature
  * workfeature-macro
[/code]

luzpaz, 2023-08-13:
Added images and example file (from forum thread) in to this ticket

kisolre, 2023-08-15:
Still happens with the release 0.21.0

PaddleStroke, 2023-09-20:
I don't know this part of the code so I wouldn't be very helpful.

However what I can say is that this 'duplicate object' command is wrong on a UI standpoint. We should be able to do simple copy paste of object.
By the way if you do copy paste you don't have the same error message as the duplicate command.

I don't understand the point of this duplicate command. And copy paste works (even though the copy paste is needlessly complex with the dependency window). It should probably be removed imo.

kisolre, 2024-07-17:
With the pending release I think this needs some attention. I still get it with: 

OS: Windows 7 build 7601
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38166 (Git)
Build type: Release
Branch: main
Hash: 4bb6d191b4686919a2f247bcfce6c186703a3d86
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Bulgarian/Bulgaria (bg_BG)
Installed mods: 
  * A2plus 0.4.65
  * Assembly3 0.12.2
  * Assembly4 0.50.12
  * DynamicData 2.62.0
  * fasteners 0.5.24
  * freecad.gears 1.2.0
  * Launcher
  * MeshRemodel 1.9.23
  * Pyramids-and-Polyhedrons
  * QuickMeasure 2022.10.28
  * ThreadProfile 1.90.0
  * toSketch 1.0.1
  * workfeature

closed 2024-10-21 by yorikvanhavre