FreeCAD/FreeCAD #6279: Tag dressup throws error if applied after boundary dressup.

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4778

* **Reporter:** sliptonic
* **Date submitted:** 11/3/2021
* **FreeCAD version:** 0.2
* **Category:** Bug
* **Status:** new
* **Tags:** 

# Original report text

https://forum.freecadweb.org/viewtopic.php?f=15&t=63433

# FreeCAD Info

```
Issue reported on machine config below but reproduced on other platforms.

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24291 (Git)
Build type: Release
Branch: releases/FreeCAD-0-19
Hash: 7b5e18a0759de778b74d3a5c17eba9cb815035ac
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 11/3/2021

Russ4262, 2022-03-17:
@sliptonic Can we close this issue now, considering PR #6566 ?  

LuzPaz is helping me to address issues and auto-closing them with the use of appropriate keywords in commit and PR descriptions. I didn't achieve that with #6566.

luzpaz, 2022-03-19:
Closing :wink:

closed 2022-03-19 by luzpaz