FreeCAD/FreeCAD #6280: Rounding error when drawing in XZ plane

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4780

* **Reporter:** joelmo
* **Date submitted:** 11/8/2021
* **FreeCAD version:** 0.2
* **Category:** Bug
* **Status:** new
* **Tags:** 

# Original report text

When I draw in XZ (front) plane I get that some points are placed on the Y axis at around ~0.3 nm. I'm guessing this is a rounding error. This does not happen when I draw in XY. When drawing in XZ, the Z field in the UI is the Y axis.

# Steps to reproduce

Open draft. Create a line and move it around while watching X, Y, Z fields.

# FreeCAD Info

```
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.26155 (Git)
Build type: Release
Branch: master
Hash: 0926a4148bcff11249fd4f56bc30256102ffe105
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United Kingdom (en_GB)
```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 1/12/2022

# Discussion from Mantis ticket

### Comment by marioalexis 2022-01-12 14:58
With https://github.com/FreeCAD/FreeCAD/pull/5307/commits/1cd01248e0fb42baace06ae7ff2475e6dd60c68d the precision on the projected points was increased to ~1e-14, so this rounding error should no longer be seen.

luzpaz, 2022-04-12:
Can anyone still reproduce this?

luzpaz, 2022-04-15:
@marioalexis84 ping

luzpaz, 2022-04-29:
CC 
@marioalexis84 
@Roy-043

Roy-043, 2022-04-30:
I cannot reproduce with the version mentioned below. So this has indeed been fixed by @marioalexis84 with the already mentioned commit.

Note that even with the new code rounding errors will occur. But they are much smaller. You may for example see -0.00mm for the delta Z coordinate. These rounding errors cannot be avoided when calculating with floating point numbers:
https://docs.python.org/3/tutorial/floatingpoint.html

This issue can be closed.

```
OS: Windows 8.1 (6.3)
Word size of FreeCAD: 64-bit
Version: 0.20.28765 (Git)
Build type: Release
Python 3.8.13, Qt 5.12.9, Coin 4.0.0, OCC 7.5.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods: 
```

luzpaz, 2022-04-30:
Closing ticket. Fixed in 1cd01248e0fb42baace06ae7ff2475e6dd60c68d by @marioalexis84

closed 2022-04-30 by luzpaz