FreeCAD/FreeCAD #6281: Issue with Linear Pattern in the Part Design workbench when replicating an Additive Pipe based on an elliptical path.

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4781

* **Reporter:** vectronic
* **Date submitted:** 11/8/2021
* **FreeCAD version:** 0.2
* **Category:** Bug
* **Status:** new
* **Tags:** Part Design

# Original report text

I am trying to use Linear Pattern to duplicate a flange on a pin. The flange is created using an additive pipe which uses a path sketch involving an ellipse.

This doesn't work.

I have queried in the forum to see if this was:

- a known bug (doesn't seem to be)
- more experienced users can get it to work (they can't)
- more experienced users thought it might be a bug (they do)

Forum link: https://forum.freecadweb.org/viewtopic.php?f=3&t=63409

I have taken my sample from the forum, improved it based on their feedback and simplified it a lot more. I have also included a comparison of using a circle (instead of an ellipse) for the additive pipe path and this seems to work fine with the linear pattern feature.

In the linear-pattern-attempt-1.FCStd you will see:

* attempt-1-01.png : Linear pattern of a pin and a flange based on a circular path is rendered.
* attempt-1-02.png : Another flange based on an elliptical path is added to the first pin.
* attempt-1-03.png : When another linear pattern for the elliptical flange is added, the result is not as expected.

In the linear-pattern-attempt-2.FCStd you will see:

* attempt-2-01.png : A pin with a circular flange and elliptical flange is created first.
* attempt-2-02.png : If I try to create a linear pattern involving all of these at once (the pin, the circular flange and the elliptical flange) the linear pattern dialog displays an unexpected error.
* attempt-2-03.png : The final result is not as expected.

# Additional information

Screenshots: https://f001.backblazeb2.com/file/vectronic-public/freecad/screenshots.zip

Sample models: https://f001.backblazeb2.com/file/vectronic-public/freecad/models.zip

# Steps to reproduce

Open the linear-pattern-attempt-1.FCStd file and observe that:

- the linear pattern using the feature of an additive pipe based on a circle path sketch IS rendered.
- the linear pattern using the feature of an additive pipe based on an ellipse path sketch IS NOT rendered.

OR

Open the linear-pattern-attempt-2.FCStd file and observe that the linear pattern including both additive pipes is rendered incorrectly.sketch is not rendered.

# FreeCAD Info

```
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.Unknown
Build type: Release
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.3
Locale: C/Default (C)

```

# Other bug information

* **Priority:** normal
* **Severity:** major
* **Category:** Bug
* **OS: MacOS 11.6**
* **Platform:** Intel
* **Updated:** 11/8/2021

luzpaz, 2022-04-29:
Renamed forum thread

DrD4ffy, 2024-01-14:
Hi,

I am not sure if it is directly related to this issue, because it does not involve an additive pipe, but I think only the devs can say if the additive pipe or the linear pattern is the root cause.

It is definitely another example of "shapes created by linear pattern" are not correct. I came across this issue while following some Fusion 360 tutorials using FreeCAD. FreeCAD is not Fusion360, thats perfectly OK. But I could reduce the issue to a very simple example file, where I think you can expect the linear pattern to work.

I modelled a short pipe and wanted to create some sort of lamellas on the outside. So I created one at the bottom and used linear pattern to reproduce it ~30 times along the pipe. It looks good on the outside, but if you look on the inside you see some strange shapes:

[image]

I attatched the FreeCAD file:
[LinearPatternIssue.zip](https://github.com/FreeCAD/FreeCAD/files/13931818/LinearPatternIssue.zip)

I found it first in the latest dev version but also in the current release.

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * Curves 0.6.21
  * fasteners 0.5.0
  * IconThemes
  * OpenDark 2023.12.17
  * OpenDark.backup1704310892.017628 2023.10.30 (Disabled)
  * PieMenu1291 1.2.7
  * Pyramids-and-Polyhedrons
  * QuickMeasure 2022.10.28
  * Render 2023.12.28
  * sheetmetal 0.3.15
  * ThreadProfile 1.89.0
```

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35570 (Git)
Build type: Release
Branch: main
Hash: 3ba0c3d7958dad891e125e065a41d95faac47c6e
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * Curves 0.6.21
  * fasteners 0.5.0
  * IconThemes
  * OpenDark 2023.12.17
  * OpenDark.backup1704310892.017628 2023.10.30 (Disabled)
  * PieMenu1291 1.2.7
  * Pyramids-and-Polyhedrons
  * QuickMeasure 2022.10.28
  * Render 2023.12.28
  * sheetmetal 0.3.15
  * ThreadProfile 1.89.0
```

DrD4ffy, 2024-01-17:
I talked about the linear pattern with a more experienced colleague who is using 0.19. Interestingly, he pointed out to me that the example I posted shows no errors in 0.19. Attached is a screenshot of the same file with 0.19.
'''
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24267 +148 (Git)
Build type: Release
Branch: Branch_0.19.4
Hash: 476ecf091941bead59b14e44afa6064d5a66afa3
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: German/Germany (de_DE)
'''
[image]

luzpaz, 2024-12-11:
@DrD4ffy would you mind testing this against FreeCAD 1.0 ?

DrD4ffy, 2025-01-16:
@luzpaz Hi, sory for the late response but I was in holyday for several weeks with not much spare time before and after. No I had time to test.
### 1. Test
I took the original file posted here last year, recomputed the complete model and the issue was still there.
### 2. Test
I created a new model, reproducing the original steps (Additive cylinder, sketch hole on top, pocket, sketch ring at the bottom, pad, linear pattern) and the result was as expected (see [LinearPatternIssue2.zip](https://github.com/user-attachments/files/18436160/LinearPatternIssue2.zip))
### 3. Test
I tried to repair the original file by undoing and then redoing the steps one by one
- undo linear pattern -> still broken
- undo linear pattern, pad of the ring -> still broken
- undo linear pattern, pad of the ring, sketch for the ring -> it works
So, apparently the sketch of the ring is the place where the model breaks. To the user both files (LinearPatternIssue and LinearPatternIssue2) look identical but Sketch001 as it was done in 0.21/0.22dev results in a broken pattern, while Sketch001 as it was don in the current releas 1.0 results in a clean pattern without artifacts.

I hope this helps.

luzpaz, 2025-01-16:
@DrD4ffy Thanks! can you please post the full About info of the 1.0 version you tested on ?

DrD4ffy, 2025-01-16:
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * IconThemes
  * OpenTheme 2024.9.1

maxwxyz, 2025-04-16:
@pieterhijma is this an OCC issue?