FreeCAD/FreeCAD #6282: freecadcmd exportStep breaking model

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4782

* **Reporter:** knittely
* **Date submitted:** 11/10/2021
* **FreeCAD version:** 0.19
* **Category:** Bug
* **Status:** confirmed
* **Tags:** #post-to-forum, IGES, python, STEP

# Original report text

importing the attached .iges file and exporting the model in freecadcmd breaks the model, while using the GUI works fine.

# Additional information

Using the GUI:
- File > Open > f.iges
- selecting the model
- File > Export... < f.step
works fine

# Steps to reproduce

Run freecadcmd.exe
```
import Part
part = Part.read("f.iges")
part.exportStep("f.step")
```

-> generates f.step, but the model is broken

# FreeCAD Info

```
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24291 (Git)
Build type: Release
Branch: releases/FreeCAD-0-19
Hash: 7b5e18a0759de778b74d3a5c17eba9cb815035ac
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: German/Germany (de_DE)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: Windows 10 Home 19042.1288**
* **Platform:** Notebook i7-9750H, 16GB ram
* **Updated:** 11/16/2021

# Discussion from Mantis ticket

### Comment by openBrain 2021-11-16 17:47
knittely : please address your issue by opening a new topic in our forum according our [reporting guidelines](https://forum.freecadweb.org/viewtopic.php?f=3&t=5236)

luzpaz, 2022-04-11:
Forum thread: https://forum.freecadweb.org/viewtopic.php?f=3&t=67920

luzpaz, 2022-04-11:
Bug still exists for 
```
OS: Ubuntu Core 20 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.28671 (Git)
Build type: Release
Branch: master
Hash: aea110ba15897661a59f04ec52c701a00daea430
Python 3.8.10, Qt 5.15.3, Coin 4.0.0, OCC 7.6.1
Locale: English/United States (en_US)
Installed mods: 
  * ToolbarStyle
  * Assembly4 0.11.10
  * stickframe
  * sheetmetal 0.2.49
  * pinger
  * Assembly3 0.11.3
  * Behave-Dark-Colors 0.0.1
  * Curves 0.4.2
  * Ship
  * BIM 2021.12.0
  * A2plus 0.4.56
```

closed 2022-04-11 by wwmayer

luzpaz, 2022-04-11:
Confirmed fix on  
```
OS: Ubuntu Core 20 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.28682 (Git)
Build type: Release
Branch: master
Hash: a13e251ad45c3562875e6bcc8e1c7e84882a4d52
Python 3.8.10, Qt 5.15.3, Coin 4.0.0, OCC 7.6.1
Locale: English/United States (en_US)
```

commit c8c46865 (2022-04-11): Part: fixes #6282: exportStep breaking model