FreeCAD/FreeCAD #6283: Double Constraint symbols not selectable

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4784

* **Reporter:** chrisb
* **Date submitted:** 11/11/2021
* **FreeCAD version:** 0.2
* **Category:** Bug
* **Status:** confirmed
* **Tags:** 

# Original report text

When multiple symbols are associated with the same geometric element like horizontal and the orthogonal constraint on the bottom line in the example attached to the forum, then it is not possible to select either of them.
[Forum discussion](https://forum.freecadweb.org/viewtopic.php?f=3&t=63621&p=546065#p546065).

# FreeCAD Info

```
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.26155 (Git)
Build type: Release
Branch: master
Hash: 0926a4148bcff11249fd4f56bc30256102ffe105
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.3
Locale: C/Default (C)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 11/16/2021

luzpaz, 2022-04-15:
@paddlestroke would you be open to take a stab at this?

PaddleStroke, 2022-04-21:
This bug is indeed annoying.
However I don't know much on that part of the code. Maybe OpenBrain might know how to solve this. As he recently made a commit to disable constraint selectability when using geometry tools.

luzpaz, 2022-12-22:
@0penBrain or @AjinkyaDahale care to weigh-in on this ?

AjinkyaDahale, 2022-12-22:
I am aware of this bug, but I don't work with the drawing aspect of the code myself too much. As paddle said 0penbrain might have a better idea. I'd be willing to take a look if he doesn't have the time.

PLChris, 2023-04-19:
Increasing the font size from 14 to 24 (on a 13" MacBook Air) makes the constraints selectable.

```
OS: macOS High Sierra (10.13)
Word size of FreeCAD: 64-bit
Version: 0.21.0.32532 (Git)
Build type: Release
Branch: master
Hash: 20e44eba50e1365b9e7bb9c5c244afe96a627a28
Python 3.10.10, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * fasteners 0.4.15
  * sheetmetal 0.2.63
  * Curves 0.6.8

```

abdullahtahiriyo, 2023-04-20:
@PLChris 

I am looking into it. There is something that does not add up. Thanks for the indication. It is useful.

abdullahtahiriyo, 2023-04-20:
https://github.com/FreeCAD/FreeCAD/pull/9328

closed 2023-04-20 by abdullahtahiriyo

AjinkyaDahale, 2023-08-24:
@V5Studio please attach version details from Help->About FreeCAD. If issue persists it should be fixed (and perhaps backported).

AjinkyaDahale, 2023-08-24:
> @AjinkyaDahale I tested on the latest version at this time (win). It's very simple to test. Add one constraint, and the constraint icon can be selected. Add one more, and we cannot select it anymore. I can still use the left panel to make individual selection as a workaround
> 

Cannon reproduce with:
```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.33981 (Git) Snap 751
Build type: Release
Branch: master
Hash: 5a153e50ffb6046b6cc4dba498344032f6edb952
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/India (en_IN)
```

V5Studio, 2023-08-24:
@AjinkyaDahale Sorry... maybe it's windows thing? I don't have access to Unix to check... OR could be that this fix will be available only when 0.22 is released?

AjinkyaDahale, 2023-08-25:
> @AjinkyaDahale Sorry... maybe it's windows thing? I don't have access to Unix to check...

If it fails for Windows that's where we should focus. If anything I should apologize for not having it. 

> OR could be that this fix will be available only when 0.22 is released?

That won't be the case. It should be available with 0.21.

commit 11c76c99 (2023-04-20): Sketcher: Fix double icons not selectable

=========================================

The scaling factor used  was the one of the ViewProviderSketch, however ZoomTranslation uses its own scaling factor.

fixes #6283