FreeCAD/FreeCAD #6284: Labels in expressions not updated correctly when Std_MergeProjects is used.

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4787

* **Reporter:** Roy_043
* **Date submitted:** 11/12/2021
* **FreeCAD version:** 0.2
* **Category:** Bug
* **Status:** new
* **Tags:** 

# Original report text

When merging the same project more than once, Labels used in expressions are only updated if they match the Name of the object they refer to.

# Additional information

Related forum pots:
https://forum.freecadweb.org/viewtopic.php?f=3&t=28511
https://forum.freecadweb.org/viewtopic.php?f=3&t=63285

# Steps to reproduce

1. Open a new document.
2. Use Std_MergeProjects to merge the attached file twice.
3. The expression for the Radius of Cylinder001 is `<<Cube>>.Width`, it should be `<<Cube001>>.Width`.
4. The expression for the Height of Cylinder001 has been correctly updated to `<<BoxFromCode001>>.Height`

# FreeCAD Info

```
OS: Windows 8.1 (6.3)
Word size of FreeCAD: 64-bit
Version: 0.20.26306 (Git)
Build type: Release
Python version: 3.8.12
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.3
Locale: Dutch/Netherlands (nl_NL)
```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 11/12/2021

luzpaz, 2022-03-13:
CC @Roy-043 (OP)

Edit: I've added the attached file from the original ticket on mantis in to this issue

Roy-043, 2024-12-09:
Problem still occurs in:
```
OS: Windows 8 build 9600
Architecture: x86_64
Version: 1.1.0dev.39416 (Git) Conda
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

Roy-043, 2025-05-24:
Problem still occurs.
```
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.41736 (Git)
Build date: 2025/05/19 10:15:51
Build type: Release
Branch: main
Hash: 76ce8ccfd5b52e520c052d2193d7a244e83d4cb6
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/86.1283/1
Installed mods:
```

Roy-043, 2025-05-24:
@maxwxyz It's a bit annoying to have to keep this issue alive again after less than 6 months. Maybe the bot's algo needs some fine-tuning?

luzpaz, 2025-05-24:
What's the proposed solution for this  ?

Roy-043, 2025-05-25:
> What's the proposed solution for this ?

See 3. in the original report.

maxwxyz, 2026-11-07:
There seems to be a fix in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/565efa5ce524e11bcc648f1a442986d589657525

Label references in the merged expressions follow the renamed objects, so the second merge does not keep pointing to the first copy.