FreeCAD/FreeCAD #6285: Polar pattern of loft up to arc surface fails

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4788

* **Reporter:** garya
* **Date submitted:** 11/14/2021
* **FreeCAD version:** 0.2
* **Category:** Bug
* **Status:** new
* **Tags:** Loft, Part Design, PolarPattern

# Original report text

If a loft is done from a square pillar resting on a ring, the resulting loft cannot be replicated in a polar array.

Forum thread: https://forum.freecadweb.org/viewtopic.php?f=3&t=63689

Several example files are posted in forum thread; the simplest is PolarPatrn_4.

# Additional information

See the forum thread for other manifestations of this bug and attempts to work around.
It was suggested this is a problem with OpenCascade.

# Steps to reproduce

Load PolarPatrn_4
Select Proto_WindingRib
Click PolarPattern
  Result: Two posts are visible, 180 degrees apart
          The original post has the additive loft; the patterned one does not (ok so far)
Click Add Feature
  Result: The added post is removed (still ok)
Click on the AdditiveLoft
  Result: AdditiveLoft is added to the PolarPattern parameters list in the dialog (correct)
          The original additive loft disappears and no lofts are visible on either post (incorrect)

# FreeCAD Info

```
OS: FreeBSD 11.4-RELEASE-p6 (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.20.master (Built from source 2021-11-01)
Build type: Release
Python version: 3.7.9
Qt version: 5.15.2
Coin version: 4.0.0
OCC version: 7.5.0
Locale: C/Default (C)
```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: FreeBSD 11.4-RELEASE**
* **Platform:** PC
* **Updated:** 11/14/2021

luzpaz, 2022-04-15:
bumped forum thread

Fat-Zer, 2023-03-06:
Is this bug still actual?

I couldn't reproduce it on a recent git build with none of the provided examples. Simple recalculation of the model seams to produce a correct result in each of the provided files.

PLChris, 2023-03-06:
Fixed in 0.21.31917

closed 2023-03-06 by PLChris

PLChris, 2023-03-08:
> Thanks @PLChris. Next time do you mind linking directly to the commit so any devs viewing this in the future can quickly link to the fix? TIA

I didn't check any commit, so I didn't know when or how it was fixed. I would guess though that it was fixed in OCC.