FreeCAD/FreeCAD #6286: Path simulator crash

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4789

* **Reporter:** fcuser123
* **Date submitted:** 11/14/2021
* **FreeCAD version:** 0.2
* **Category:** Bug
* **Status:** new
* **Tags:** 

# Original report text

FreeCAD crashes when trying to run path simulator. The issue is confirmed in 0.20 version.
Forum discussion: https://forum.freecadweb.org/viewtopic.php?f=15&t=63656
The topic contains a project file that was used to reproduce the issue.

# Additional information

FreeCAD 0.19, Libs: 0.19R24291 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2021
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Assembly4 workbench (v0.11.4) loaded
Sheet Metal workbench loaded
Manipulator WB Activated(v 1.4.1)
Enumeration index -1 is out of range, ignore it
Enumeration index -1 is out of range, ignore it
Enumeration index -1 is out of range, ignore it
Enumeration index -1 is out of range, ignore it
Enumeration index -1 is out of range, ignore it
Enumeration index -1 is out of range, ignore it
Enumeration index -1 is out of range, ignore it
Enumeration index -1 is out of range, ignore it
Enumeration index -1 is out of range, ignore it
Enumeration index -1 is out of range, ignore it
Coin info in MeshRenderer(): GL_ARB_vertex_buffer_object extension not supported
Manipulator WB Deactivated()

Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x46210) [0x7f4d2804f210]
#1  0x7f4c6a3ebc83 in cSimTool::GetToolProfileAt(float) from /tmp/.mount_FreeCAqeJXt2/usr/lib/PathSimulator.so+0x43
#2  0x7f4c6a3ec0d6 in cStock::ApplyLinearTool(Point3D&, Point3D&, cSimTool&) from /tmp/.mount_FreeCAqeJXt2/usr/lib/PathSimulator.so+0x436
#3  0x7f4c6a3ea3c5 in PathSimulator::PathSim::ApplyCommand(Base::Placement*, Path::Command*) from /tmp/.mount_FreeCAqeJXt2/usr/lib/PathSimulator.so+0x125
#4  0x7f4c6a3eeb77 in PathSimulator::PathSimPy::ApplyCommand(_object*, _object*) from /tmp/.mount_FreeCAqeJXt2/usr/lib/PathSimulator.so+0xa7
#5  0x7f4c6a3eebf9 in PathSimulator::PathSimPy::staticCallback_ApplyCommand(_object*, _object*, _object*) from /tmp/.mount_FreeCAqeJXt2/usr/lib/PathSimulator.so+0x29
#6  /tmp/.mount_FreeCAqeJXt2/usr/bin/../lib/libpython3.8.so.1.0(PyCFunction_Call+0x66) [0x7f4d2a2ca696]
#7  /tmp/.mount_FreeCAqeJXt2/usr/bin/../lib/libpython3.8.so.1.0(_PyObject_MakeTpCall+0x230) [0x7f4d2a272550]
#8  /tmp/.mount_FreeCAqeJXt2/usr/bin/../lib/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0x4ddd) [0x7f4d2a2dc28d]
#9  /tmp/.mount_FreeCAqeJXt2/usr/bin/../lib/libpython3.8.so.1.0(_PyFunction_Vectorcall+0xfb) [0x7f4d2a2a45bb]
#10  /tmp/.mount_FreeCAqeJXt2/usr/bin/../lib/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0x6e3) [0x7f4d2a2d7b93]
#11  /tmp/.mount_FreeCAqeJXt2/usr/bin/../lib/libpython3.8.so.1.0(_PyFunction_Vectorcall+0xfb) [0x7f4d2a2a45bb]
#12  /tmp/.mount_FreeCAqeJXt2/usr/bin/../lib/libpython3.8.so.1.0(+0x17ad13) [0x7f4d2a2c0d13]
#13  /tmp/.mount_FreeCAqeJXt2/usr/bin/../lib/libpython3.8.so.1.0(PyVectorcall_Call+0x71) [0x7f4d2a2781c1]
#14  0x7f4d27cf8bad in PySide::SignalManager::callPythonMetaMethod(QMetaMethod const&, void**, _object*, bool) from /tmp/.mount_FreeCAqeJXt2/usr/bin/../lib/./libpyside2.cpython-38-x86_64-linux-gnu.so.5.13+0x9d
#15  /tmp/.mount_FreeCAqeJXt2/usr/bin/../lib/./libpyside2.cpython-38-x86_64-linux-gnu.so.5.13(+0x1a081) [0x7f4d27d02081]
#16  0x7f4d2858fa7f in QMetaObject::activate(QObject*, int, int, void**) from /tmp/.mount_FreeCAqeJXt2/usr/bin/../lib/libQt5Core.so.5+0x727
#17  0x7f4d28596912 in QTimer::timeout(QTimer::QPrivateSignal) from /tmp/.mount_FreeCAqeJXt2/usr/bin/../lib/libQt5Core.so.5+0x2a
#18  /tmp/.mount_FreeCAqeJXt2/usr/lib/python3.8/site-packages/PySide2/QtCore.cpython-38-x86_64-linux-gnu.so(+0x2ad94d) [0x7f4d0f65b94d]
#19  0x7f4d2858e533 in QObject::event(QEvent*) from /tmp/.mount_FreeCAqeJXt2/usr/bin/../lib/libQt5Core.so.5+0x5b
#20  /tmp/.mount_FreeCAqeJXt2/usr/lib/python3.8/site-packages/PySide2/QtCore.cpython-38-x86_64-linux-gnu.so(+0x2ad4d5) [0x7f4d0f65b4d5]
#21  0x7f4d28d9f7f2 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAqeJXt2/usr/bin/../lib/libQt5Widgets.so.5+0xdc
#22  0x7f4d28da531d in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAqeJXt2/usr/bin/../lib/libQt5Widgets.so.5+0x1ac9
#23  0x7f4d2a94fdf9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAqeJXt2/usr/bin/../lib/libFreeCADGui.so+0x89
#24  0x7f4d28579606 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAqeJXt2/usr/bin/../lib/libQt5Core.so.5+0x86
#25  0x7f4d285a78a4 in QTimerInfoList::activateTimers() from /tmp/.mount_FreeCAqeJXt2/usr/bin/../lib/libQt5Core.so.5+0x24c
#26  /tmp/.mount_FreeCAqeJXt2/usr/bin/../lib/libQt5Core.so.5(+0x1dbf54) [0x7f4d285a7f54]
#27  /tmp/.mount_FreeCAqeJXt2/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_dispatch+0x27d) [0x7f4d2549623d]
#28  /tmp/.mount_FreeCAqeJXt2/usr/bin/../lib/./libglib-2.0.so.0(+0x554d1) [0x7f4d254964d1]
#29  /tmp/.mount_FreeCAqeJXt2/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x31) [0x7f4d25496571]
#30  0x7f4d285a80e8 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAqeJXt2/usr/bin/../lib/libQt5Core.so.5+0x5e
#31  0x7f4d28576019 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAqeJXt2/usr/bin/../lib/libQt5Core.so.5+0x187
#32  0x7f4d28579e8f in QCoreApplication::exec() from /tmp/.mount_FreeCAqeJXt2/usr/bin/../lib/libQt5Core.so.5+0xfd
#33  0x7f4d2a8d9926 in Gui::Application::runApplication() from /tmp/.mount_FreeCAqeJXt2/usr/bin/../lib/libFreeCADGui.so+0x15f6
#34  /tmp/.mount_FreeCAqeJXt2/usr/bin/freecad(+0x44bf) [0x5562cc5124bf]
#35  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7f4d280300b3]
#36  /tmp/.mount_FreeCAqeJXt2/usr/bin/freecad(+0x47d9) [0x5562cc5127d9]

# Steps to reproduce

1. Start the program
2. Open the project
3. Activate Path workbench
4. Click on "Simulate Path..."
5. Click on "Activate..."
6. FreeCAD crashes

# FreeCAD Info

```
OS: Linux Mint 20.2 (X-Cinnamon/cinnamon)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24291 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.19.2)
Hash: 7b5e18a0759de778b74d3a5c17eba9cb815035ac
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/Canada (en_CA)
```

# Other bug information

* **Priority:** normal
* **Severity:** crash
* **Category:** Bug
* **Updated:** 11/14/2021

sliptonic, 2022-03-30:
As noted in the forum discussion, this crash occurs because of a malformed tool.  Specifically the tool shape is designed so the tool has a hole through it.  There's never a reason to design a tool like this.  It would be nice if the simulator handled the issue without crashing.
The issue is low-priority and I'm tagging it for 0.21.
[image]

evils, 2023-01-21:
a friend of mine ran into this issue with a tool that seems normal to me
[grinder-holder.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/10472610/grinder-holder.FCStd.zip)
(just unzip, open, Path workbench, Path dropdown menu, CAM Simulator, Activate/Resume (play button), crash)

this was on 0.19, i can reproduce it on 0.20.2

commit 59a208e8 (2023-04-26): Fixed segmentation fault in Path workbench

Closes #6286.