FreeCAD/FreeCAD #6288: FreeCAD crashes when trying to create a new file in sway wm

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4793

* **Reporter:** will1
* **Date submitted:** 11/28/2021
* **FreeCAD version:** 0.19
* **Category:** Bug
* **Status:** new
* **Tags:** 

# Original report text

when using [sway](https://github.com/swaywm/sway) window manager in linux I get a crash when I do `File > New` with the output: 
```
FreeCAD 0.19.2, Libs: 0.19.2R2_voidlinux
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2021
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

connect failed: No such file or directory
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3d000) [0x7ff4af357000]
#1  /usr/lib/libX11.so.6(XDefaultScreenOfDisplay+0) [0x7ff4acf552e0]
#2  /usr/lib/libCoin.so.80(+0x458c2e) [0x7ff4aeeb3c2e]
#3  /usr/lib/libCoin.so.80(glxglue_init+0x3c) [0x7ff4aeeb44dc]
#4  /usr/lib/libCoin.so.80(cc_glglue_instance+0x122) [0x7ff4aeead002]
#5  0x7ff4aed3dbe8 in SoGLRenderActionP::isDirectRendering(SoState const*) const from /usr/lib/libCoin.so.80+0x48
#6  0x7ff4aed3ff53 in SoGLRenderActionP::render(SoNode*) from /usr/lib/libCoin.so.80+0xc3
#7  0x7ff4aed395c2 in SoAction::apply(SoNode*) from /usr/lib/libCoin.so.80+0x2d2
#8  0x7ff4b1c1e5f9 in Gui::SoBoxSelectionRenderAction::apply(SoNode*) from /usr/lib/freecad/lib64/libFreeCADGui.so+0x29
#9  0x7ff4b1cc2833 in Gui::View3DInventorViewer::renderScene() from /usr/lib/freecad/lib64/libFreeCADGui.so+0x143
#10  0x7ff4b1c8a415 in SIM::Coin3D::Quarter::QuarterWidget::paintEvent(QPaintEvent*) from /usr/lib/freecad/lib64/libFreeCADGui.so+0x275
#11  0x7ff4b1c90282 in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::paintEvent(QPaintEvent*) from /usr/lib/freecad/lib64/libFreeCADGui.so+0x42
#12  0x7ff4b057c03e in QWidget::event(QEvent*) from /usr/lib/libQt5Widgets.so.5+0x1ce
#13  0x7ff4b06244de in QFrame::event(QEvent*) from /usr/lib/libQt5Widgets.so.5+0x1e
#14  0x7ff4afa11f03 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /usr/lib/libQt5Core.so.5+0x93
#15  0x7ff4b053c16e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt5Widgets.so.5+0x6e
#16  0x7ff4b1a05328 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib64/libFreeCADGui.so+0x88
#17  0x7ff4afa1219a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt5Core.so.5+0x12a
#18  0x7ff4b0574166 in QWidgetPrivate::sendPaintEvent(QRegion const&) from /usr/lib/libQt5Widgets.so.5+0x36
#19  0x7ff4b05749b2 in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) from /usr/lib/libQt5Widgets.so.5+0x7f2
#20  0x7ff4b0575d03 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) from /usr/lib/libQt5Widgets.so.5+0x503
#21  0x7ff4b05746cc in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) from /usr/lib/libQt5Widgets.so.5+0x50c
#22  0x7ff4b0575d03 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) from /usr/lib/libQt5Widgets.so.5+0x503
#23  0x7ff4b05746cc in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) from /usr/lib/libQt5Widgets.so.5+0x50c
#24  0x7ff4b0575d03 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) from /usr/lib/libQt5Widgets.so.5+0x503
#25  0x7ff4b05746cc in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) from /usr/lib/libQt5Widgets.so.5+0x50c
#26  0x7ff4b0575d03 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) from /usr/lib/libQt5Widgets.so.5+0x503
#27  0x7ff4b05746cc in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) from /usr/lib/libQt5Widgets.so.5+0x50c
#28  /usr/lib/libQt5Widgets.so.5(+0x173810) [0x7ff4b054c810]
#29  0x7ff4b057c767 in QWidget::event(QEvent*) from /usr/lib/libQt5Widgets.so.5+0x8f7
#30  0x7ff4b053c17f in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt5Widgets.so.5+0x7f
#31  0x7ff4b1a05328 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib64/libFreeCADGui.so+0x88
#32  0x7ff4afa1219a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt5Core.so.5+0x12a
#33  0x7ff4afa14bd1 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) from /usr/lib/libQt5Core.so.5+0x171
#34  /usr/lib/libQt5Widgets.so.5(+0x475933) [0x7ff4b084e933]
#35  0x7ff4afa3e1c1 in QObject::event(QEvent*) from /usr/lib/libQt5Core.so.5+0x291
#36  0x7ff4b053c17f in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt5Widgets.so.5+0x7f
#37  0x7ff4b1a05328 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib64/libFreeCADGui.so+0x88
#38  0x7ff4afa1219a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt5Core.so.5+0x12a
#39  0x7ff4afa14bd1 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) from /usr/lib/libQt5Core.so.5+0x171
#40  /usr/lib/libQt5Core.so.5(+0x306ef3) [0x7ff4afa69ef3]
#41  /usr/lib/libglib-2.0.so.0(g_main_context_dispatch+0x25b) [0x7ff4ad54ec0b]
#42  /usr/lib/libglib-2.0.so.0(+0x53eb8) [0x7ff4ad54eeb8]
#43  /usr/lib/libglib-2.0.so.0(g_main_context_iteration+0x2f) [0x7ff4ad54ef6f]
#44  0x7ff4afa6957f in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt5Core.so.5+0x5f
#45  0x7ff4afa10b4b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt5Core.so.5+0x12b
#46  0x7ff4afa18dd0 in QCoreApplication::exec() from /usr/lib/libQt5Core.so.5+0x90
#47  0x7ff4b197af68 in Gui::Application::runApplication() from /usr/lib/freecad/lib64/libFreeCADGui.so+0x1978
#48  FreeCAD(+0x4954) [0x55bdd297a954]
#49  /usr/lib/libc.so.6(__libc_start_main+0xea) [0x7ff4af341e0a]
#50  FreeCAD(+0x4c4a) [0x55bdd297ac4a]
```

# Steps to reproduce

Install FreeCAD from the void repository, open it in sway window manager, create a new file

# FreeCAD Info

```
OS: void (Sway)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.2.2_voidlinux
Build type: Release
Python version: 3.10.0
Qt version: 5.15.2
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

```

# Other bug information

* **Priority:** normal
* **Severity:** crash
* **Category:** Bug
* **Updated:** 11/28/2021

yorikvanhavre, 2022-02-14:
Most likely due to bad support of OpenGL. It would be interesting to test:

1. Other OpenGL-based applications, for ex. Blender
2. Other wayland window manager

luzpaz, 2022-02-14:
Closing ticket as per our guidelines to first open a forum thread. Also this must be checked with v0.20dev to see if the issue still persists

ii8, 2023-06-09:
This crash still happens on version `0.20.2`, blender works without problems. It's unlikely to be related to the specific wayland compositor, the crash happens because FreeCAD is trying to use X functions in something called libCoin by the looks of it:
```
(gdb) bt
#0  0x00007ffff2bae220 in XDefaultScreenOfDisplay () from /lib/libX11.so.6
#1  0x00007ffff3c594be in ?? () from /lib/libCoin.so.80
#2  0x00007ffff3c59d6c in glxglue_init () from /lib/libCoin.so.80
#3  0x00007ffff3c528f2 in cc_glglue_instance () from /lib/libCoin.so.80
```
I don't have a forum account and cba to make one lol sry

Jiri-Macha, 2023-08-18:
Yes, I can confirm the crash on 0.20.2 running in Sway (Wayland compositor) on Gentoo Linux. To avoid crash run freecad  this way under XWayland:   env -u WAYLAND_DISPLAY freecad

luzpaz, 2023-08-18:
@Jiri-Macha is this the case also for v0.21 stable ?

Jiri-Macha, 2023-09-08:
Version 0.21 is not yet in Gentoo repository. I will try it later. The
problem is not a problem of missing variable. It is opposite.
The variable WAYLAND_DISPLAY has to be unset. I met one more problem which
is caused by variable QT_QPA_PLATFORM.
I run it now this way:   env -u WAYLAND_DISPLAY -u QT_QPA_PLATFORM
/usr/bin/freecad

This way freecad runs in sway under XWayland (X server) without issues.

luzpaz, 2023-09-09:
@Jiri-Macha Is this something that FreeCAD should run when detecting Sway Window Manager or should we let downstream Sway devs handle this ?

Jiri-Macha, 2023-09-10:
In an ideal case:

FreeCAD developers should make it ready for the new Wayland protocol.
The code base should support compilation flag ensuring correct behavior
of compiled FreeCAD in environment in which X-server is missing and
only Wayland compositor like e.g. Sway present. The same or similar like
Emacs or Firefox already did. Or at least patch the source code base that
way that
it will not crash, but will tell the user what is wrong and how to get
it running.

Linux FreeCAD package maintainers should test FreeCAD running under X server
but also under Wayland compositor and if it is not running well ensure
correct environment to run it at lest under modified environment in XWayland
(X server).

Both presumptions are not valid for released 0.20.2 version.
The command line I have send can help in such situation.

And maybe it was already improved in 0.21 version or the development HEAD -
I do not
know now.

I think it has nothing to do with Sway developers. Wayland ready software
like e.g. above mentioned Firefox or Emacs runs since its specific version
perfectly
there.

ii8, 2023-09-10:
> @Jiri-Macha Is this something that FreeCAD should run when detecting Sway Window Manager or should we let downstream Sway devs handle this ?

There is no problem with sway here so sway devs have nothing to fix, and running under XWayland is only a workaround. For proper wayland support FreeCADs dependencies, in particular Coin3D, need to be fixed to not make calls to X functions when running on a wayland server.

luzpaz, 2023-09-23:
> For proper wayland support FreeCADs dependencies, in particular Coin3D, need to be fixed to not make calls to X functions when running on a wayland server.

@ii8 would you mind reporting this to upstream https://github.com/coin3d/coin and post the ticket back here so we can track it ?

ii8, 2023-09-24:
https://github.com/coin3d/coin/issues/451
https://groups.google.com/g/coin3d-discuss/c/osGzMf4RW80

hasecilu, 2023-10-05:
> Most likely due to bad support of OpenGL. It would be interesting to test:
> 
>     1. Other OpenGL-based applications, for ex. Blender
> 
>     2. Other wayland window manager

I tested on `Hyprland` using FreeCAD 0.21.1, Libs: 0.21.1R33694 (Git):
- I opened Blender but were not any kind of warning/error messages.
- Last month I have been working on several WB, installed external WBs, used Macros, importing *.step files, etc and all seems to work fine.
- On `New file` I get
    ```
        Coin warning in glxglue_isdirect(): Couldn't get current GLX context.
        (qt.qpa.wayland) Wayland does not support QWindow::requestActivate()
        (qt.qpa.wayland) Wayland does not support QWindow::requestActivate()
    ```
     but no problems.
- Apart from frequent `(qt.qpa.wayland) Wayland does not support QWindow::requestActivate()` messages as in #10369 no issues.

luzpaz, 2023-10-15:
bumping for visibility

luzpaz, 2024-09-23:
please :+1: this comment: https://github.com/coin3d/coin/issues/451#issuecomment-2368693041

closed 2025-01-12 by github-actions[bot]