FreeCAD/FreeCAD #6289: Remove CloseShape and ConnectLines

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4794

* **Reporter:** chrisb
* **Date submitted:** 11/29/2021
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

CloseShape and ConnectLines should be removed from Sketcher, see [forum discussion](https://forum.freecadweb.org/viewtopic.php?p=550462#p550462).

# FreeCAD Info

```
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.26375 (Git)
Build type: Release
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.3
Locale: C/Default (C)

```

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 11/29/2021

adrianinsaval, 2022-11-23:
This can be closed as they have been both removed

luzpaz, 2022-11-29:
@adrianinsaval any idea what commit(s) associated with this ticket?

closed 2022-11-29 by luzpaz

adrianinsaval, 2022-11-29:
19230c83ef4d6659bb12140f1d182df4036a7649 I guess