FreeCAD/FreeCAD #6291: Part Design Loft between two shapbinders fails, but Part Loft works with only an out of scope warning.

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4799

* **Reporter:** drmacro
* **Date submitted:** 12/8/2021
* **FreeCAD version:** 0.2
* **Category:** Bug
* **Status:** new
* **Tags:** 

# Original report text

Part Design Loft between two shapbinders fails, but Part Loft works with only an out of scope warning.

This inconsistent behavior; shapebinders can be used with Part Design Pad.

In addition, A Part Sweep of a Part Helix and a shapebinder works, but the Part Design Helix also fails with a Shapebinder.

Unless there is a specific reason the ShapeBinder can't work with Part Design tools, for consistency, it should work with none or all?

Forum thread contains example file and discussion.

# Steps to reproduce

Create two shapbinders and attempt to Loft, Helix etc.

Forum thread contains example file and discussion.

# FreeCAD Info

```
OS: Debian GNU/Linux 10 (buster) (XFCE/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.20.26583 (Git)
Build type: Release
Branch: master
Hash: 4bd77b3b35ea794057ffe6a55858b7b1abee7ed0
Python version: 3.7.3
Qt version: 5.11.3
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 12/8/2021

mfro0, 2023-04-08:
It's not so simple, unfortunately. The issue doesn't have anything to do with Tree view selection in PD (which works fine if you change tabs, btw.).

PartDesign explicitly filters the selection for objects derived from Part2DObject (which SketchObject is but ShapeBinder/SubShapeBinder isn't) at numerous places.
 
I played around and extended all these places for Binders which appears to work somehow. At least I was able to modify an existing PD Loft by manually editing Sections and Profile references from Sketches to ShapeBinders in the Property editor (as I didn't touch the PD Loft UI yet).

My feeling is that this is an extremely ugly approach, however, as it completely bypasses the rather clean PD type hierarchy and one would need to add code to check for the binders to be 2D only at all these places.

I'm wondering if it would be cleaner (and versatile enough) if we only allowed Binders that directly reference Sketches only for selection? Or do we need a specialisation of the binder classes that would only allow referencing 2D objects inheriting from Part2DObject?

mfro0, 2023-04-09:
The original report stated ShapeBinders could be padded.

While this is true, there appears to be no check at all if the binder is  2D (or at least face-shaped only) or solid. In the latter case, the Pad must fail.
So, basically same problem there.

grubertechnic, 2023-07-27:
Through a few test i found some interesting results.

The first thing is, that even when a sketch is not needed, while starting a task and there is no sketch in it, PartDesign comes with an error message,

`No sketch is available in the document.`

This in not only in Loft, this is in all PartDesign operations. When a empty sketch is in the document (note, also an unattached sketch, which is not in the body), it works. This is the first issue i found (i will crate a ticket for that later).

Also i can confirm @adrianinsaval suspects, that have something to do with the tree object or 3D selection. While on sketch objects it does not really matter, if 3D selection or tree selection is used, for the shapebinder there is an difference. It works only while using the face of the shapebinder.

[2023-07-27 17-52.webm](https://github.com/FreeCAD/FreeCAD/assets/109620914/04df9071-15e6-4125-86c3-3b87f5b19658)

[test1.zip](https://github.com/FreeCAD/FreeCAD/files/12185599/test1.zip)

Also her a second test file with differnt shapes, shapebinders and wires.

[2023-07-27 17-55.webm](https://github.com/FreeCAD/FreeCAD/assets/109620914/c90f56eb-06c1-4d71-80c2-00f2865e9841)

[test2.zip](https://github.com/FreeCAD/FreeCAD/files/12185604/test2.zip)

grubertechnic, 2023-07-27:
version forgotten

```
OS: Debian GNU/Linux 11 (bullseye) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33657 (Git)
Build type: Release
Branch: master
Hash: 2a1d4d4994bf3707f7a07aef07e17f5270b4854b
Python 3.9.2, Qt 5.15.2, Coin 4.0.0, Vtk 9.0.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * QuickMeasure 2022.10.28
  * CurvedShapes 1.0.4
  * Curves 0.6.12
```

closed 2025-06-08 by github-actions[bot]