FreeCAD/FreeCAD #6292: customize window: merge SpaceBall Movement and SpaceBall Buttons tabs

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4800

* **Reporter:** roipoussiere
* **Date submitted:** 12/11/2021
* **FreeCAD version:** 0.19
* **Category:** Feature
* **Status:** new
* **Tags:** #post-to-forum

# Original report text

The "SpaceBall Movement" and "SpaceBall buttons" tabs takes a lot of place.

The tab bar could be more clean and concise if those tabs was merged in one "Spaceball" tab.

With default size, the window is even not large enough to disaply all the tabs (at least in French, didn't tried in other languages).

# Other bug information

* **Priority:** normal
* **Severity:** trivial
* **Category:** Feature
* **Updated:** 12/14/2021

# Discussion from Mantis ticket

### Comment by openBrain 2021-12-14 11:05
roipoussiere : please proceed according our [reporting guidelines](https://forum.freecadweb.org/viewtopic.php?f=3&t=5236) by first discussing your request in our forum. Thx

---

### Comment by roipoussiere 2021-12-14 17:11
forum thread: https://forum.freecadweb.org/viewtopic.php?f=34&t=64533

Manuelzwei, 2023-03-06:
***
BUG
***

Hello,

I'm a German user of FreeCAD 0.20.1.

I use a 3DConnexion SpaceMouse Pro. 

But:
In 
Werkzeuge -->> Benutzerdefiniert -->> Spaceball-Knöpfe -->> Referenz anzeigen -->> Drucken

(Tools -->> Custom -->> Spaceball-Buttons -->> View Reference -->> Print) in English?

shows only the first 19 buttons. 

I would like to send you this file:

[image]

The other buttons not printed.

Also, the column width cannot be adjusted when printed or is cut off.

And the next bug:

The great "new" button in the 3D cube (...switch the actual displayed view...) can not selected with the spacaball-buttons.
(...sorry, I do not know the right word from these button...)

Great if that were the case!

Thanks a lot!

M4xTheMan, 2023-06-23:
@Manuelzwei Thanks for your additional comment. I've just upgraded to a SpaceMouse Pro and experienced some bugs regarding the configuration of the buttons. It works fine as long as I only configure button 0-3 (labeled 1-4). I'm currently looking into it. It's the first time I'm working with the FreeCAD source code so please bare with me. I'm on Linux by the way.

Are you able to assign the other buttons without any issues? If not, what's working for me already is to define those buttons in my user.cfg. If you'd like to know more, please let me know. 

I'm not really getting your problem regarding the print out.

Regarding the views: Do you mean the selection of Orthographic view, Perspective view and Isometric? If yes, you can trigger those commands using your SpaceMouse already. You've set the buttons to Std_OrthographicCamera, Std_PerspectiveCamera or Std_ViewIsometric. https://wiki.freecad.org/Std_View_Menu

Manuelzwei, 2023-06-25:
Hello, thank you for your answer!

Yes, I can configure more buttons - see my picture. I can customize all 27 buttons in version 0.21.0

[image]

"I'm not really getting your problem regarding the print out."

Print out: I can't print more than 19 buttons:

[image]

This is a screencopy from my pdf-viewer. You see only the first 19 buttons. The other buttons was'nt printing yet.

Todo: Klick on the "Print Reference"-button on the bottom of the Customize-Screen and make a print...

Maybe, with Linux it can work, but in Windows don't.

"...You've set the buttons to Std_OrthographicCamera, Std_PerspectiveCamera or Std_ViewIsometric."

See Button 13 and 14:
I push the Button "1" and "2" on my SpaceMouse - no function. 
When I press the buttons on the SpaceMous, the relevant entries open on the right side in the  Customize window and the correct icons are displayed. I mean, the buttons 1..4 works on my SpaceMouse as well as they should.

But - on the model no reaction.

"Left", "Right", ect.  Buttons are working OK, Orthographics and the other not 
 
I do not know why, but there are some functions at the right side they can not add to the buttons in the left side.

I don't check it out on Linux, I am only a Windows-user.

By the way, I accidentally deleted my old configuration for the SpaceMouse buttons, so the images are different now.

If I had a printout of the last configuration it wouldn't be a problem. Because... what was button 20..27 again?

And:
Couldn't the selected icons be displayed on the left side as well as in the printout. So I have to guess which button is assigned which icon.
Like this:

[image]

THANKS FOR READING!!!

My version:
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.32110 (Git)
Build type: Release
Branch: master
Hash: a000dcf60af18f489fff48d29d7f2f4b2092483f
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Luxembourg (de_LU)
Installed mods: 
  * kicadStepUpMod 10.17.1
  * offline-documentation 1.0.0-alpha
[/code]