FreeCAD/FreeCAD #6295: use a more explicit file name for backup files

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4805

* **Reporter:** roipoussiere
* **Date submitted:** 12/12/2021
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** #post-to-forum

# Original report text

The presence of some files like `project.FCStd1` that I didn't created really confused me. After some research I discovered that it was a backup file.

I may suggest to rename it to `.project.FCStd1.backup` instead.

First, **.backup** makes the purpose of the file explicit.
Then, the presence of the dot hides the file (so it doesn't bother the user, when there are many FC files in the same folder) and it also avoid to commit it unintentionally.

# FreeCAD Info

```
OS: Ubuntu 20.04.3 LTS (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24291 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.19.2)
Hash: 7b5e18a0759de778b74d3a5c17eba9cb815035ac
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: French/France (fr_FR)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 12/14/2021

# Discussion from Mantis ticket

### Comment by openBrain 2021-12-14 11:06
roipoussiere : please proceed according our [reporting guidelines](https://forum.freecadweb.org/viewtopic.php?f=3&t=5236) by first discussing your request in our forum. Thx

---

### Comment by roipoussiere 2021-12-14 16:52
I just created forum thread for this suggestion: https://forum.freecadweb.org/viewtopic.php?f=34&t=64530

luzpaz, 2022-04-03:
CC @adrianinsaval

adrianinsaval, 2022-04-03:
We have an option to name FCbak, that also mentions date and time of the backup, that should be explicit enough. In any case we could change the default to FCBak. Or if we do a subdirectory the problem is solved.

luzpaz, 2022-04-03:
Thanks for weighing-in. Should I close in favor of #6697 then?

adrianinsaval, 2022-04-03:
I say yes

closed 2022-04-03 by luzpaz