FreeCAD/FreeCAD #6296: Part Design, moving sketch from one body to a new body removes other features from the original body.

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4806

* **Reporter:** garya
* **Date submitted:** 12/14/2021
* **FreeCAD version:** 0.2
* **Category:** Bug
* **Status:** new
* **Tags:** Part Design

# Original report text

Moving a sketch from one body to a new body removes other features from the original body.

# Steps to reproduce

Open the file tmp_StripCut1.FCStd from this topic:
  https://forum.freecadweb.org/viewtopic.php?f=3&t=64463&p=553269#p553269
Create a new body.
Right click on Sk_Cut2_HoldDown and choose "Move object to other body"; use Body001 and click "OK"

# FreeCAD Info

```
OS: FreeBSD 11.4-RELEASE-p6 (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.20.master
Build type: Release
Python version: 3.7.9
Qt version: 5.15.2
Coin version: 4.0.0
OCC version: 7.5.0
Locale: C/Default (C)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: FreeBSD 11.4-RELEASE**
* **Platform:** PC
* **Updated:** 12/14/2021

pedromorgan, 2022-03-30:
This kinda stuff i just experienced.. but a newbie.. am confused.. expected a clone/binder.. sigh!

FreeWilly2022, 2025-05-23:
Tested on version 1.0.1 and there was no problem. None of the parts in the original body were removed, so this appears to have been fixed. The following was in the Report View, which appears correct since there is a lot of external geometry referenced in the sketch:

19:28:59  Sketcher::SketchObject: Link(s) to object(s) 'Sketch002 Sketch004' go out of the allowed scope 'Sketch005'. Instead, the linked object(s) reside within 'N/A'.
19:28:59  PartDesign::Pad: Link(s) to object(s) 'Sketch005' go out of the allowed scope 'Pad'. Instead, the linked object(s) reside within 'Body001'.
19:28:59  PartDesign::Body: Link(s) to object(s) 'Pad' go out of the allowed scope 'Body'. Instead, the linked object(s) reside within 'N/A'.
19:28:59  Sketcher::SketchObject: Link(s) to object(s) 'Sketch005' go out of the allowed scope 'Sketch006'. Instead, the linked object(s) reside within 'Body001'.
19:30:12  Sketcher::SketchObject: Link(s) to object(s) 'Sketch003 Sketch' go out of the allowed scope 'Sketch004'. Instead, the linked object(s) reside within 'N/A'.
19:30:12  Sketcher::SketchObject: Link(s) to object(s) 'Sketch004' go out of the allowed scope 'Sketch005'. Instead, the linked object(s) reside within 'Body001'.
19:30:12  PartDesign::Body: Link(s) to object(s) 'Pad' go out of the allowed scope 'Body'. Instead, the linked object(s) reside within 'N/A'.

OS: macOS 15.5
Architecture: arm64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion