FreeCAD/FreeCAD #6298: Security Vulnerability in PathSanity.py

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4810

* **Reporter:** eldstal
* **Date submitted:** 12/23/2021
* **FreeCAD version:** 0.2
* **Category:** General
* **Status:** resolved
* **Tags:** Path, security

# Original report text

An external command execution by the PathSanity script is vulnerable to OS command injection when applied to a crafted input file.

This allows an attacker to execute arbitrary commands on the victim's system.

# Additional information

***Impact***
Arbitrary Code Execution

***Caveats***
The exploit sequence described above is somewhat unreliable. After working once, the PathSanity script tends to fail entirely until a new path is created. This appears to be a problem with the script itself, not related to the security vulnerability.

***Cause***
The invocation of asciidoctor at [PathSanity.py:467](https://github.com/FreeCAD/FreeCAD/blob/1653d26ffa0b35d57fc3325c18bf5b7bad515703/src/Mod/Path/PathScripts/PathSanity.py#L467) contains unsanitized text taken from the input file. Since the command is invoked using os.system(), there is no separation between command and parameter. Furthermore, the command is passed through the OS shell, which allows for more ways of injecting commands.

***Proposed Mitigation***
Use subprocess.Popen() instead of os.system(). This allows the command to be separated from its parameters, and prevents injection of shell commands using substitution methods (such as the $(galculator) trick demonstrated above.

# Steps to reproduce

***Attacker:***
1. Start with an FCStd document with some content
2. Open the ***Path*** workbench
3. Create a new Path Job object using the toolbar, assign it to any model.
4. Edit the new Path Job, in the [i]Output[i] tab, set [i]Output File[i] to ***./$(galculator)***
5. Save the document

***Victim***
1. Open the evil document
2. Open the ***Path*** workbench
3. Run the [Sanity check](https://wiki.freecadweb.org/Path_Sanity#Usage) on the evil Path Job

Result: galculator is launched by FreeCAD.

# FreeCAD Info

```
OS: Arch Linux (i3/i3)
Word size of FreeCAD: 64-bit
Version: 0.20.26683 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0388fbc)
Hash: 0388fbc98d49d874fb341b9037a743bc691d501f
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.3
Locale: English/United States (en_US)
```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** General
* **Platform:** Linux
* **Updated:** 1/25/2022

# Discussion from Mantis ticket

### Comment by eldstal 2022-01-25 12:58
This vulnerability has been assigned [CVE-2021-45845](https://www.cve.org/CVERecord?id=CVE-2021-45845)

---

### Comment by sliptonic 2022-01-25 15:07
This was corrected with PR #5306.   I think I used the 'fixes' tag wrong.

https://github.com/FreeCAD/FreeCAD/pull/5306

---

### Comment by sliptonic 2022-01-25 15:09
https://github.com/FreeCAD/FreeCAD/pull/5306

---

### Comment by Kunda1 2022-01-25 15:14
eldstal that appimage is now dated btw

sliptonic, 2022-02-07:
Resolved by PR #5306

closed 2022-02-07 by sliptonic

luzpaz, 2022-02-08:
@sliptonic this was backported but hasn't been officially released as a minor release