FreeCAD/FreeCAD #6299: Workbench Texture Arch non viene applicato correttamente

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4811

* **Reporter:** Lucchesi
* **Date submitted:** 12/25/2021
* **FreeCAD version:** 0.19
* **Category:** Bug
* **Status:** new
* **Tags:** 

# Original report text

Ambiente di lavoro :
Freecad 0.19 - Windows 10 64 bit - Processore Intel i5 _Scheda grafica Vidia Gforce 710 
Problema 
Utilizzando la workbench Texture Arch non si trasferisce l'immagine sulla superficie selezionata .
La procedura seguita è la seguente :
Creazione di un cubo con Part
Mostra tutto nella finestra "Vista Combinata "
Cliccando alla voce "Expression" con il tasto destro del mouse aggiungo "Proprieta --> App :: property Link --> Base --> Material 
Cambiamo Workbench da PART--> a -->  Arch Texture 
Selezioniamo "MATERIAL ARCH " --> "WOOD GENERIC " 
A questo Punto dovrei trovarmi alla Voce "CUBO" -->"BASE" -->  Material --> "WOOD GENERIC " 
Quest'ultimo passaggio non viene eseguito e quindi il successivo e cioè Texture Config ( la scelta di una immagine da associare al solido ).
Riporto quanto trovato nel Forum :
Leonardo Chiruzzi
Vezio Lucchesi Ho riprovato con una installazione "particolare", qualcosa è cambiato ma continua a non funzionare[/b]. Mentre con un altra versione:
FreeCAD: 64-bit
Version: 0.19.23964 (Git) AppImage
Python version: 3.8.6
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
sta funzionando.

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 12/25/2021

luzpaz, 2022-02-08:
Looks like this bug should be reported to https://github.com/furti/FreeCAD-ArchTextures ?

yorikvanhavre, 2022-02-11:
Transferred to https://github.com/furti/FreeCAD-ArchTextures/issues/42

closed 2022-02-11 by yorikvanhavre