FreeCAD/FreeCAD #6300: Using custom threads in fasteners ScrewTab generates variable used before assigned exception

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4812

* **Reporter:** Frank_Schruefer
* **Date submitted:** 12/25/2021
* **FreeCAD version:** 0.2
* **Category:** Patch
* **Status:** new
* **Tags:** 

# Original report text

Hy, I have a path for a trivial bug in the fasteners workbench code.
Without it the use of custom thread diameters and pitch won't work.

# Additional information

Patch created via:
stone@merlin:~/.local/share> diff -c FreeCAD/Mod/fasteners/screw_maker.py.orig FreeCAD/Mod/fasteners/screw_maker.py

Diff output:
*** FreeCAD/Mod/fasteners/screw_maker.py.orig   2021-12-22 00:43:13.164645112 +0100
--- FreeCAD/Mod/fasteners/screw_maker.py        2021-12-22 00:44:23.605517538 +0100
***************
*** 3792,3798 ****
      else: # custom pitch and diameter
        P = customPitch
        if self.sm3DPrintMode:
!         dia = self.smNutThrScaleA * dia + self.smNutThrScaleB
        else:
          dia = customDia
      residue, turns = math.modf((l)/P)
--- 3792,3798 ----
      else: # custom pitch and diameter
        P = customPitch
        if self.sm3DPrintMode:
!         dia = self.smNutThrScaleA * customDia + self.smNutThrScaleB
        else:
          dia = customDia
      residue, turns = math.modf((l)/P)
***************
*** 3831,3837 ****
      else: # custom pitch and diameter
        P = customPitch
        if self.sm3DPrintMode:
!          dia = self.smScrewThrScaleA * dia + self.smScrewThrScaleB
        else:
          dia = customDia 
      if self.rThread:
--- 3831,3837 ----
      else: # custom pitch and diameter
        P = customPitch
        if self.sm3DPrintMode:
!         dia = self.smScrewThrScaleA * customDia + self.smScrewThrScaleB
        else:
          dia = customDia 
      if self.rThread:
***************
*** 3870,3876 ****
      else: # custom pitch and diameter
        P = customPitch
        if self.sm3DPrintMode:
!          dia = self.smScrewThrScaleA * dia + self.smScrewThrScaleB
        else:
          dia = customDia
      dia = dia*1.01
--- 3870,3876 ----
      else: # custom pitch and diameter
        P = customPitch
        if self.sm3DPrintMode:
!         dia = self.smScrewThrScaleA * customDia + self.smScrewThrScaleB
        else:
          dia = customDia
      dia = dia*1.01

# Steps to reproduce

-Choose fasteners Workbench
-Create 'Add threaded Rod for tapping holes'
-Change 'diameter' to 'Custom'
-Change 'diameter Custom' and 'pitch Custom'

# FreeCAD Info

```
OS: openSUSE Tumbleweed (KDE//usr/share/xsessions/plasma5)
Word size of FreeCAD: 64-bit
Version: 0.20.26651 (Git) AppImage
Build type: Release
Branch: (HEAD detached at d357d52)
Hash: d357d524d97db884b9bb562415e6b6d782ec02b4
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.3
Locale: German/Germany (de_DE)

```

# Other bug information

* **Priority:** normal
* **Severity:** trivial
* **Category:** Patch
* **Updated:** 12/25/2021

# Discussion from Mantis ticket

### Comment by Frank_Schruefer 2021-12-25 13:14
P.S.: Forgot to mention that in order to reproduce the bug the 'Thread generation' option in the fasteners workbench settings must be set to '3D printer compatible'.

luzpaz, 2022-05-16:
@shaise does this patch make sense to you? (since it's fasteners related)

yorikvanhavre, 2022-05-17:
This should actually be transferred over to https://github.com/shaise/FreeCAD_FastenersWB

luzpaz, 2022-05-17:
It looks like you can't transfer from one org to another. Just in the same org. I have repo access to this FreeCAD master and  fasteners and the option is not available in the transfer dialog. We'll need to close and open there. I was hoping to save some work and figure out if we could just close it here instead, hence I asked @shaise to look at it. (I was being lazy... :blush:)

Frank-Schruefer, 2022-05-19:
Hy, I posted this patch and was asked to 'weigh-in'. Don't know what that means.
Obviously It's the wrong place for the patch, should I post it where yorikvanhavre suggested?

yorikvanhavre, 2022-05-19:
The ideal would be if you could open a pull request on the https://github.com/shaise/FreeCAD_FastenersWB repo. Are you comfortable with git? Otherwise I explain you how to proceed...

Frank-Schruefer, 2022-05-24:
> The ideal would be if you could open a pull request on the https://github.com/shaise/FreeCAD_FastenersWB repo. Are you comfortable with git? Otherwise we explain you how to proceed...

Ok, we have a newborn here so it took me a while to get back to this :-)

Until now I never used git for more than clone / pull / status so I took the chance to learn a bit about it because nowadays there's no more way around it.
Until now I forked the mentionend repo and cloned it to my local machine.
Now I'm at the point where tutorials suggest to create a branch. As a decade long SVN, and formerly CVS, user I don't get the point here as for such a simple change I'd usually never create a new branch.
So my question is: Is creating of a new branch mandatory or not necessary in this case?
And because the next step in the tutorials I found would be to 'git remote add upstream <orig-repo-adress>' a second question arises: Why? I don't understand why to do that because after I've created the fork it is already pointing back to the upstream repo which I can even see at my github fork in the upper left corner (https://github.com/Frank-Schruefer/FreeCAD_FastenersWB).

I hope somebody has a little time to enlighten me a bit otherwise I'll just proceed as the tutorials suggest. The problem is just that I don't like doing things I don't fully understand :-)

Thanks,
   Frank

shaise, 2022-05-24:
1. creating a branch is not necessary.
2. adding the upstream is in case the upstream repository was changed after you forked it and needed to be updated. if the upstream was not change, this is not necessary as well as far as I know.

Frank-Schruefer, 2022-05-26:
Thanks shaise!

I found that in the repo I cloned the patch was either already applied by someone else or they just already fixed it.
So no further action needs to be taken and the case can be closed.

luzpaz, 2022-06-05:
@Frank-Schruefer how is your progress going?

luzpaz, 2022-07-06:
Looks like the code was fixed in Fasteners, right @shaise ?
Closing ticket.

closed 2022-07-06 by luzpaz