FreeCAD/FreeCAD #6301: [Proposal] Run script (bash) for FreeCAD

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4815

* **Reporter:** mcgiwer
* **Date submitted:** 12/28/2021
* **FreeCAD version:** 0.2
* **Category:** Feature
* **Status:** feedback
* **Tags:** #pending-forum

# Original report text

Hello. I had wrote a script in bash that (after placing in the same folder as for example bin) would allow to launch FreeCAD directly from it and with creates a log file.

I hope that it becomes included to the base FreeCAD. I also hope to recieve a upload permission on GitHub to be able to contribute as well.

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **OS: Debian 10+**
* **Platform:** amd64
* **Updated:** 1/23/2022

# Discussion from Mantis ticket

### Comment by openBrain 2022-01-03 10:57
mcgiwer : please address issues/proposals according [our reporting guidelines](https://forum.freecadweb.org/viewtopic.php?f=3&t=5236)by opening a new thread in our forum.
Notice that there is no need to get write access to GH repo to contribute. Just fork, code, then do a pull request. This is how contributions are merged.
Thx

---

### Comment by Kunda1 2022-01-23 00:41
Forum thread: https://forum.freecadweb.org/viewtopic.php?f=4&t=65585

adrianinsaval, 2022-02-08:
rather than an issue OP should make a PR, does it make sense to keep this open?

luzpaz, 2022-02-08:
Closing...lets see what OP does

closed 2022-02-08 by luzpaz