FreeCAD/FreeCAD #6305: Offer full Placement dialog when creating a sketch

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4821

* **Reporter:** chrisb
* **Date submitted:** 1/7/2022
* **FreeCAD version:** 0.2
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

When creating a new sketch it should be possible to see the full attachment dialog. The [current procedure](https://forum.freecadweb.org/viewtopic.php?f=8&t=65098) is laborious.
The corresponding [forum discussion](https://forum.freecadweb.org/viewtopic.php?f=8&t=65098#p558823) discusses als a configurable option
for the currently implemented simpler workflow.

The implementation could be similar to realthunder's branch with the additional goody of showing a local coordinate system, so that the user can see where the sketch is placed and how it is oriented.

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 1/7/2022

maxwxyz, 2023-12-11:
This could be closed as more details are specified here #11133

maxwxyz, 2024-01-14:
@Roy-043 ^

closed 2024-01-14 by Roy-043