FreeCAD/FreeCAD #6309: Add Advanced Section Views

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4830

* **Reporter:** wandererfan
* **Date submitted:** 1/15/2022
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

Add half, broken out, offset, rotated and removed sections.

[Forum discussion.](https://www.forum.freecadweb.org/viewtopic.php?f=35&t=65321)

# FreeCAD Info

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.20.
Build type: Release
Branch: unknown
Hash: cd089a82c741c3725058df4a082d9a27fb857a71
Python version: 3.8.10
Qt version: 5.12.8
Coin version: 4.0.0
OCC version: 7.5.2
Locale: English/United States (en_US)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 1/15/2022

luzpaz, 2022-11-09:
@WandererFan does this have anything to do with Complex Section ?

FEA-eng, 2022-12-07:
> @WandererFan does this have anything to do with Complex Section ?

Yes, the new Complex Section tool fulfills this FR in a large part. Only broken out and rotated/removed sections are still to be implemented. I can create a new request for them or modify this one.

luzpaz, 2022-12-11:
Yes please close this ticket and open new tickets for each new feature?

FEA-eng, 2022-12-21:
> Yes please close this ticket and open new tickets for each new feature?

Unfortunately, I can't edit or close this FR. It was imported from the old bug tracker. And @WandererFan created it (even though it originated from my request on the forum).

luzpaz, 2022-12-21:
I'll close this one. Please open a new one and reference this ticket so folks can follow the progression of these implemented features. TIA

closed 2022-12-21 by luzpaz