FreeCAD/FreeCAD #6311: Freecad hangs when trying to import a step file

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4832

* **Reporter:** chrisdepalma
* **Date submitted:** 1/16/2022
* **FreeCAD version:** 0.2
* **Category:** Bug
* **Status:** assigned
* **Tags:** #pending-forum

# Original report text

I am using latest freecad dev version from 1/16/2022

Freecad hangs when trying to load the step file.  It does this on the .19.3 version also.

Freecad starts to render but when clicking on the UI even after an hour after starting import, the ui is unresponsive.

# Additional information

PC is a I7 5820K with 40 GB of ram.

# Steps to reproduce

Select file import step.

Select step file located here:  https://github.com/ZeroGDesign/Mercury/blob/main/CAD/Zero%20G%20-%20Mercury%20One.step

Freecad will never finish import.  

This step was i believe generated by Fusion 360.

# FreeCAD Info

```
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24267 +99 (Git)
Build type: Release
Branch: Branch_0.19.3
Hash: 6530e364184ce05ccff39501e175cf2237e6ee4b
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)

```

# Other bug information

* **Priority:** normal
* **Severity:** major
* **Category:** Bug
* **Updated:** 2/4/2022

# Discussion from Mantis ticket

### Comment by openBrain 2022-01-17 16:52
chrisdepalma : please report your issues according [our reporting guidelines](https://forum.freecadweb.org/viewtopic.php?f=3&t=5236) by creating a new topic in our forum. Thx

---

### Comment by Kunda1 2022-01-19 16:50
Closing since no forum thread was opened. Re-open if needed. 

---

### Comment by chrisdepalma 2022-01-30 00:12
I added a post on forum as requested.   If anybody cares that step files don't load, then you should try the simple repro steps.

---

### Comment by chrisdepalma 2022-01-30 00:13
post title:  Freecad hangs loading step file.

---

### Comment by Kunda1 2022-01-30 00:51
Y u no add URL ?

---

### Comment by chrisdepalma 2022-01-30 03:49
Kunda1  Url to github is in the issue.

---

### Comment by adrianinsaval 2022-01-31 20:39
url to the forum thread
here it is: https://forum.freecadweb.org/viewtopic.php?f=3&t=65799
OP please read and follow guidelines when doing a bug report

alirezadigi, 2023-04-04:
same here
(but not sure of f360)

superbonaci, 2023-11-21:
CAD Assistant opens it perfectly:

[image]

felipebalbi, 2024-01-12:
I have the same issue with 0.21.2. Just tried importing [this](https://openbeagle.org/beagleplay/beagleplay/-/blob/main/Design/beagleplayv10_a2_board_3D_20221219.stp?ref_type=heads) step file and FreeCAD just hangs.

superbonaci, 2024-01-12:
> I have the same issue with 0.21.2. Just tried importing [this](https://openbeagle.org/beagleplay/beagleplay/-/blob/main/Design/beagleplayv10_a2_board_3D_20221219.stp?ref_type=heads) step file and FreeCAD just hangs.

Same here. The file fully loads and I can do stuff, but after a while of being open the mouse pointer keeps spinning and can't bring the window to front.

If I wait long enough then it un-freezes automatically so kind of works:

[image]

If you can make another test waiting long enough and check if there's any CPU activity.

felipebalbi, 2024-01-12:
yeah, I think the step importer is just pretty slow, perhaps, due to the amount of data within the step file? If wait long enough (really long), then it eventually shows the render.

superbonaci, 2024-01-12:
You can compare the load times with other software like `CAD Assistant`, but remember that FreeCAD is an editor. Software for just viewing usually goes much faster.

PaddleStroke, 2024-03-04:
Have you guys tried to enable 'skip recompute' before importing the step? Option in right click on the document tree item.

superbonaci, 2024-12-08:
do not stale.

superbonaci, 2025-06-08:
do not stale.

superbonaci, 2025-08-08:
I have some updates now that the bot is complaining. FreeCAD 1.0.2 x86_64 on Windows 11.

File of OP https://github.com/ZeroGDesign/Mercury/blob/main/CAD/Zero%20G%20-%20Mercury%20One.step load fine, but takes 10 to 15 minutes:

[image]

superbonaci, 2025-08-08:
@felipebalbi tried today your file https://openbeagle.org/beagleplay/beagleplay/-/blob/main/Design/beagleplayv10_a2_board_3D_20221219.stp?ref_type=heads with FreeCAD 1.0.2 and this is what happens:

It works, but fully loading takes about 30 minutes. The view is loaded earlier but the program is frozen with the mouse pointer circling until fully loads:

[image]