FreeCAD/FreeCAD #6313: Mouse-over Info in Task: Edit welding symbols

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4838

* **Reporter:** bensay
* **Date submitted:** 1/31/2022
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** feedback
* **Tags:** #pending

# Original report text

Detailed information on forum: [Add Weld Tool - thoughts and suggestions](https://www.forum.freecadweb.org/viewtopic.php?f=8&t=48039)

# FreeCAD Info

```
OS: Ubuntu 20.04.3 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.20.
Build type: Release
Branch: unknown
Hash: 3a4100d656182667ac4efd8d4b127ad03f7edf46
Python version: 3.8.10
Qt version: 5.12.8
Coin version: 4.0.0
OCC version: 7.5.2
Locale: English/United States (en_US)

```

# Other bug information

* **Priority:** low
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/2/2022

# Discussion from Mantis ticket

### Comment by Kunda1 2022-02-02 17:11
PR https://github.com/FreeCAD/FreeCAD/pull/5477
Commit https://github.com/FreeCAD/FreeCAD/commit/8506453c933110f124dcb0033ca5f71cfe1b0975

bensay what is left?

luzpaz, 2022-04-03:
bumped [thread](https://forum.freecadweb.org/viewtopic.php?p=585766#p585766)

luzpaz, 2023-03-15:
@WandererFan can this ticket be closed ?

luzpaz, 2023-11-06:
cc @WandererFan

WandererFan, 2023-11-07:
Most of the recommendations in the original forum post have not been implemented, so I would say this should remain active.

WandererFan, 2023-11-07:
From the original forum topic:

- tool tip improvements
-  ISO Standards compliance ("other side", staggered welds)
- UI: add weld finish field
- UI: icons for optional checkboxes

luzpaz, 2023-11-07:
Linking as loosely related: https://github.com/Zolko-123/FreeCAD_WeldingWB/issues/4

luzpaz, 2024-03-09:
Linking to #12812

pierreporte, 2025-04-29:
A **150 € reward** will be given to the contributor that writes a pull request, solving this issue, that gets merged. It will be paid by the FreeCAD Project Association. It is to be considered an incentive. Grant applications are welcome, but the total amount will exclude the reward.