FreeCAD/FreeCAD #6314: Tools: workbench icon error when envoking pure python test module

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4839

* **Reporter:** mdkus
* **Date submitted:** 2/6/2022
* **FreeCAD version:** 0.2
* **Category:** Bug
* **Status:** new
* **Tags:** #pending-PR

# Original report text

The location of the workbench icon for pure python modules should be the recommended resource directory as described under chapter 6.2.2 in the developpers guide, and not the location of users app data. I generated a [PR](https://github.com/FreeCAD/FreeCAD/pull/5507) to solve this.

# Steps to reproduce

1. Generate a pure python module with the fcbt-Tool (CP option)
2. Generate build files with CMake
3. Compile the build files
4. Start FC: The python test modul will be envoked with an error when searching the workbench icon under the users app data

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: Windows 7,11**
* **Platform:** Desktop
* **Updated:** 2/7/2022

# Discussion from Mantis ticket

### Comment by Kunda1 2022-02-07 03:39
PR https://github.com/FreeCAD/FreeCAD/pull/5507

mdkus, 2022-02-15:
This issue can be close. But how?

closed 2022-02-15 by chennes

luzpaz, 2022-02-15:
@mdkus JFYI, you can remotely close a ticket by adding `Fixes #<ticket>` in the commit summary of your PR.