FreeCAD/FreeCAD #6315: parsetab.py not created as part of installation

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4840

* **Reporter:** StefanBruens
* **Date submitted:** 2/6/2022
* **FreeCAD version:** 0.19
* **Category:** Bug
* **Status:** assigned
* **Tags:** 

# Original report text

FreeCAD uses ply.yacc in two places:

```
Mod/Fem/femtools/tokrules.py:yacc.yacc()
Mod/OpenSCAD/importCSG.py:    parser = yacc.yacc(debug=0)
```

According to https://www.skenz.it/compilers/ply
> Since table construction is relatively expensive (especially for large grammars), the resulting parsing table is written to a file called parsetab.py. In addition, a debugging file called parser.out is created.

When FreeCAD has been installed to a write-protected (not writable by regular user) path, the parsetab.py can not be created.

# Additional information

This was already mentioned in:

https://forum.freecadweb.org/viewtopic.php?style=1&p=369267#p369267

# Steps to reproduce

This is e.g. visible when running the "TestFemApp" unit test:

```....WARNING: Couldn't open 'parser.out'. [Errno 13] Permission denied: '/usr/lib64/FreeCAD/Mod/Fem/femtools/parser.out'
WARNING: Token 'COMMENT' defined, but not used
WARNING: There is 1 unused token
Generating LALR tables
WARNING: Couldn't create 'femtools.parsetab'. [Errno 13] Permission denied: '/usr/lib64/FreeCAD/Mod/Fem/femtools/parsetab.py'
```
 

# FreeCAD Info

```
OS: openSUSE Tumbleweed (KDE//usr/share/xsessions/plasma5)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.Unknown
Build type: Release
Python version: 3.8.12
Qt version: 5.15.2
Coin version: 3.1.3
OCC version: 7.5.3
Locale: English/United States (en_US)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 2/6/2022

# Discussion from Mantis ticket

### Comment by StefanBruens 2022-02-06 17:41
As the relevant code is unchanged in master, this also applies to 0.20

---

### Comment by Kunda1 2022-02-06 18:50
StefanBruens thanks. [s]Next time please open a forum thread.  [/s] my bad, there was a forum thread attached
keithsloan52 this is openscad related. Care to weigh-in?

Edit: forum thread exists

chennes, 2022-02-11:
As it appears that this affects OpenSCAD as well, I'll take a look at it.

chennes, 2022-02-12:
I was able to use @wwmayer's suggestion in the forums discussion to add a specific output directory to the `yacc.yacc()` call in the OpenSCAD WB in 2a79ce4bdc3113a03d3c9bb2d30968ace4766fc0. Presumably this approach will also work for FEM, @berndhahnebach .

luzpaz, 2022-02-13:
Oh, so would this potentially fix: (?) 
* [OpenSCAD test errors on v0.20 r27464](https://forum.freecadweb.org/viewtopic.php?f=4&t=66140)
* [Spreadsheet, FEM and Test wb failures on v0.20 r27464](https://forum.freecadweb.org/viewtopic.php?f=4&t=66141)

chennes, 2022-02-13:
No, I don't think so, at least not the OpenSCAD one.

luzpaz, 2022-02-24:
Per @KeithSloan, @chennes fixed this in 2a79ce4bd

StefanBruens, 2022-02-24:
@luzpaz - please reopen. 

Only OpenSCAD is fixed, not FEM.

Though, for FEM fixing it is easier, as the tokrules are self-contained. At build time:
```
cd Mod/Fem
python3 -B femtools/tokrules.py
```

This has the benefit of creating the parsetab exactly once (compared to writing a temporary file each time, which is already discarded on next invocation).

luzpaz, 2022-02-25:
@berndhahnebach has been missing for a while. Anyone know if he's OK?

KeithSloan, 2022-02-25:
Forum info says he last visited/posted 8th January, hope he is okay.

luzpaz, 2022-03-11:
@donovaly or @AjinkyaDahale can you take a look at this since @berndhahnebach is not active (AFAICT?)

StefanBruens, 2022-03-11:
I have rechecked what are the viable options, for both OpenSCAD and FEM:

1. create parsetab.py at build time
2. write parsetab.py at run time to writable directory
3. skip caching

(1.) is easy for FEM, as the parser/lexer rules are selfcontained, but significantly harder for OpenSCAD, as the rules depend on a complete FreeCAD import for some parser callbacks.
(2.) is quite easy (see current OpenSCAD change), but should be done carefully: target path should be user specific and source specific.
(3.) trivial (option in `yacc.yacc()` call)

The benefits of caching are questionable: 
- first time run already happens in milliseconds, the rules are fairly small, and computers have become faster since the 70s.
- Caching is only beneficial on second run (i.e. has to be done after each boot)
- Caching has some overhead (additional filesystem accesses vs some extra CPU cycles)

The implementation of (2.) in OpenSCAD is problematic (@chennes):
- the tempdir is typically `/tmp`, i.e. not user specific. Different users would try to create the same file
- the target path is not source/rule specific. `/tmp/parsetab.py` would clash between e.g. OpenSCAD and FEM.

I would heavily favor option (3.).

AjinkyaDahale, 2022-03-11:
> @donovaly or @AjinkyaDahale can you take a look at this since @berndhahnebach is not active (AFAICT?)

It seems people already are getting to the bottom of this. Am I still needed?

KeithSloan, 2022-03-11:
But option (3) requires that parstab.py is built as part of installation.

For OpenSCAD import parstab.py only changes if there are changes in the ply definitions in importCSG.py, apart from an error message the fact that parstab.py was in a read only part of the os was not a real problem till somebody deleted it.

If going for option 3 then people need to understand that changes in importCSG.py may require it to be rebuilt and that installations are shipped with a copy.

StefanBruens, 2022-03-11:
> But option (3) requires that parstab.py is built as part of installation.

No. ply is able to generate everything just in memory, no reason to write it to disk.

KeithSloan, 2022-03-11:
From https://www.dabeaz.com/ply/ply.html

Since table construction is relatively expensive (especially for large grammars), the resulting parsing table is written to a file called parsetab.py. In addition, a debugging file called parser.out is created. On subsequent executions, yacc will reload the table from parsetab.py unless it has detected a change in the underlying grammar (in which case the tables and parsetab.py file are regenerated). Both of these files are written to the same directory as the module in which the parser is specified. 

I don't see any mention of in memory and anyway performing it every time is like compiling a program every time.

StefanBruens, 2022-03-11:
1. The grammar is not "large" here
2. https://github.com/dabeaz/ply/blob/4378de314b11d2bc75990194f1f3425c473f4e86/CHANGES

> 01/26/20  PLY no longer writes cached table files.  Honestly, the use of
>           the cached files made more sense when I was developing PLY on
>           my 200Mhz PC in 2001. It's not as much as an issue now. For small
>           to medium sized grammars, PLY should be almost instantaneous. 
>           If you're working with a large grammar, you can arrange
>           to pickle the associated grammar instance yourself if need be.
>           The removal of table files eliminated a large number of optional
>           arguments to yacc() concerning the names and packages of these files.

KeithSloan, 2022-03-11:
Fair enough, I originally wrote the importer back in 2012 see copyright notice some eight years before this became an option.

luzpaz, 2022-03-19:
@donovaly note: there is a FEM portion of this ticket that isn't fixed.

donovaly, 2022-03-19:
I am not deep enough to FEM to address this.

KeithSloan, 2022-03-19:
From StefanBruens comments there is a better fix for OpenSCAD just create the Grammar in memory @chennes

chennes, 2022-03-19:
Thanks for the ping, that had fallen off my radar. I'll add it to my to-do list, I should be able to get to it this week.

PR #6611 (2022-03-31): Avoid writing temporary cache files from ply.yacc
Writing the parsetab.py cache file provides no measurable benefit, but causes several problems, so turn it off.

Also omit the generation of any debug output files.

Fixes #6315. 

- [ ]  Your pull request is confined strictly to a single module. That is, all the files changed by your pull request are either in `App`, `Base`, `Gui` or one of the `Mod` subfolders. If you need to make changes in several locations, make several pull requests and wait for the first one to be merged before submitting the next ones
- [x]  In case your pull request do [... cut]