FreeCAD/FreeCAD #6317: Topological Naming

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=922

* **Reporter:** Jriegel
* **Date submitted:** 12/20/2012
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** assigned
* **Tags:** toponaming

# Original report text

Group issue for all issues regarding Topological Naming. All bugs and features will be collected under this issue and targeted by the Naming Project:

http://www.freecadweb.org/wiki/Naming_project

# Other bug information

* **Priority:** high
* **Severity:** major
* **Category:** Feature
* **Updated:** 2/20/2021

# Discussion from Mantis ticket

### Comment by berndhahnebach 2014-04-26 17:32
link on new wiki
http://www.freecadweb.org/wiki/index.php?title=Naming_project

---

### Comment by Kunda1 2017-05-05 12:11
Assigning to jnxd since his project was accepted for [GSOC 2017](https://forum.freecadweb.org/viewtopic.php?f=8&t=22229)

---

### Comment by mihai.dumitresq 2019-03-16 13:22
What's the latest status on topological naming?

---

### Comment by Kunda1 2019-03-16 16:49
mihai.dumitresq currently the best prospects look like in realthunder's link3/Assembly3 brnaches. You can find the discussion on the Forum: https://forum.freecadweb.org/viewtopic.php?f=10&t=27278

yorikvanhavre, 2022-02-11:
I think this can be closed as it is widely referenced elsewhere and partially addressed by #4752

closed 2022-02-11 by yorikvanhavre

luzpaz, 2022-02-11:
@yorikvanhavre That PR will probably be closed per @realthunder's FOSDEM2022 talk where he said he would resubmit TPN patches.

adrianinsaval, 2022-02-11:
shouldn't this be closed once it's merged? But I guess we don't need an issue to be aware of this one