FreeCAD/FreeCAD #6318: Loft tool incorrect result with rotated ellipses (Upstream OCC bug)

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=1858

* **Reporter:** NikB
* **Date submitted:** 12/16/2014
* **FreeCAD version:** 0.14
* **Category:** Bug
* **Status:** acknowledged
* **Tags:** #ready-for-migration, Bountysource Potential, bump, OCC Bug, OCC NotOnRoadmap, upstream

# Original report text

Loft tool gives unexpected result when using identical ellipses rotated. One end edge as expected, but other has scolloping, concave or convex dependent on two or more ellipses.

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: Windows 8**
* **Platform:** x86
* **Updated:** 1/24/2022

# Discussion from Mantis ticket

### Comment by shoogen 2014-12-17 10:14
reproduceable in DRAWEXE of OCCT 6.8.1-dev
reported as http://tracker.dev.opencascade.org/view.php?id=25626

---

### Comment by Kunda1 2017-08-22 11:47
Still reproducible on OCCT 7.1 and upstream isn't on the roadmap. Requires bump.

---

### Comment by Kunda1 2018-10-19 13:36
adbdullah would you be interested in bumping the upstream ticket in the hopes that it may be added to the OCC roadmap ?

---

### Comment by normandc 2018-11-11 06:18
Original attachment is missing, and the attached files in the OCCT bug tracker can't be downloaded.

A loft between three ellipses rotated 90 degrees from each other appear fine in FC 0.18-dev with latest OCCT 7.3.0...

---

### Comment by Kunda1 2018-11-12 00:47
What should be we do about this ticket then?

---

### Comment by normandc 2018-11-12 02:55
Maybe the OCCT bug tracker requires to be registered to download. Since abdullah has an account, we could ask him to upload the files here so we could check with FreeCAD compiled with latest OCCT. From what I understand of the posted DRAWEXE script (in the upstream bug, "Steps to reproduce"), it reads from brep files that are probably included in the zip archive.

---

### Comment by abdullah 2018-11-13 17:51
Files are attached. I will indicate that the FC project is monitoring this ticket and whether it will be assigned to a roadmap.

---

### Comment by abdullah 2018-11-14 19:06
normandc
Kunda1

Bugnote #0081082 added by MSV (developer) 2018-11-14 10:12 

Dear abdullah, OCC has no fonds to solve this bug in the nearest future without external investment or contribution. You are welcome to propose your fix if you can.

---

### Comment by abdullah 2018-11-14 19:08
Unfortunately it is a too big shoe for my feet.

---

### Comment by normandc 2018-11-15 01:53
Thanks, abdullah. I imported the 3 brep files into FreeCAD and created a Part Loft. I'm attaching it for future reference. The issue indeed persists with OCC 7.3.0. Ellipses are rotated only a few degrees, I had tried 90 degrees between each ellipse and that worked.

I also tried a sweep with a straight line from (0,0,0) to (0,0,20) as path; same issue.

That's a bummer on OCC's reply...

luzpaz, 2022-05-31:
Just re-emphasizing that upstream responded with:
> OCC has no fonds to solve this bug in the nearest future without external investment or contribution. You are welcome to propose your fix if you can.

grubertechnic, 2023-07-27:
Shape still invalid.

```
OS: Debian GNU/Linux 11 (bullseye) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33657 (Git)
Build type: Release
Branch: master
Hash: 2a1d4d4994bf3707f7a07aef07e17f5270b4854b
Python 3.9.2, Qt 5.15.2, Coin 4.0.0, Vtk 9.0.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * QuickMeasure 2022.10.28
  * CurvedShapes 1.0.4
  * Curves 0.6.12
```

[image]

A testfile to test it more easier.
[6318.zip](https://github.com/FreeCAD/FreeCAD/files/12185907/6318.zip)

ElementW, 2026-08-24:
No improvement as of OCCT 8.1.0.dev1-3d097a0328. 

I haven't tested this yet, but this though looks like something that could be improved/fixed by #31833, by its surface solver choice and/or (as named currently) its "Check profile compatibility" option.