FreeCAD/FreeCAD #6319: Crash when creating sweep (fix pending on the upstream release of OCC 7.7)

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=1926

* **Reporter:** wvmarle
* **Date submitted:** 1/17/2015
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** acknowledged
* **Tags:** #ready-for-migration, OCC 7.7, OCC Bug, sweep, upstream

# Original report text

Crash when creating a sweep.

# Additional information

OS: Linux Mint 17 Qiana
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.15.4484 (Git)
Branch: master
Hash: f72c802f5daa378298ffbfc3f3a17ec2d35c1816
Python version: 2.7.6
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 6.7.1

# Steps to reproduce

Parts toolbox, select Sweep tool
Add Wire, Wire001, Wire002 for the sweep
Select the ellipse for sweep path
Create the sweep: crash

Probably I'm doing something illegal here by having the ellipse extend a little beyond the Wire and Wire002, but nevertheless it shouldn't crash.

Making it a complete ellipse solves the problem, it only crashes when the ellipse is an arc.

# Other bug information

* **Priority:** normal
* **Severity:** crash
* **Category:** Bug
* **Updated:** 1/24/2022

# Discussion from Mantis ticket

### Comment by cox 2015-01-17 15:25
EDIT -Disregard this comment. I can reproduce

http://forum.freecadweb.org/viewtopic.php?f=3&t=9358#p75957

OS: "Arch Linux"
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.15.4495 (Git)
Branch: master
Hash: 1f44226882071790a86a16f86bfacda0e7a5a880
Python version: 2.7.9
Qt version: 4.8.6
Coin version: 3.1.3
OCC version: 6.7.1

---

### Comment by shoogen 2015-01-17 16:12
I created an example for DRAWEXE from the attached file and uploaded it as sweep-arc2.zip to http://tracker.dev.opencascade.org/view.php?id=24909

---

### Comment by shoogen 2015-04-09 21:10
test http://forum.freecadweb.org/viewtopic.php?f=3&t=10445&p=84239#p84161  before closing the ticket.

---

### Comment by Kunda1 2017-01-11 11:55
Upstream ticket was bumped to OCC 7.2 roadmap.

Edit: tagged as 'upstream' and 'OCC 7.2' to indicate that this issue should be re-examined when 7.2 is released

---

### Comment by Kunda1 2017-08-21 11:04
Upstream ticket was was bumped to OCC 7.2.1 roadmap. Tagged ticket "OCC 7.2.1"

---

### Comment by Kunda1 2018-03-03 10:52
Upstream ticket was was bumped to OCC 7.4 roadmap. Notated in OCC bugtracker list: https://forum.freecadweb.org/viewtopic.php?f=10&t=20264

---

### Comment by Kunda1 2019-09-01 19:35
Upstream ticket was was bumped to OCC 7.5 roadmap. 
Tagged this ticket OCC7.5
Notated in OCC bugtracker list: https://forum.freecadweb.org/viewtopic.php?f=10&t=20264

---

### Comment by Kunda1 2020-10-08 13:47
Upstream ticket was was bumped to OCC 7.6 roadmap.
Tagged this ticket OCC7.6
Notated in OCC bugtracker list: https://forum.freecadweb.org/viewtopic.php?f=10&t=20264

---

### Comment by Kunda1 2022-01-15 12:58
Upstream ticket was was bumped to OCC 7.7 roadmap.
Tagged this ticket OCC7.6
Notated in OCC bugtracker list: https://forum.freecadweb.org/viewtopic.php?f=10&t=20264

luzpaz, 2022-04-18:
Can someone be so kind to Explain Like I'm 5 y.o. (ELI5) what is going on in the [upstream ticket](http://tracker.dev.opencascade.org/view.php?id=24909) ?

grubertechnic, 2023-07-27:
It does not crash here. It does not work, because it is too much twisted without extra steps, but it does not crash.

```
OS: Debian GNU/Linux 11 (bullseye) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33657 (Git)
Build type: Release
Branch: master
Hash: 2a1d4d4994bf3707f7a07aef07e17f5270b4854b
Python 3.9.2, Qt 5.15.2, Coin 4.0.0, Vtk 9.0.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * QuickMeasure 2022.10.28
  * CurvedShapes 1.0.4
  * Curves 0.6.12
```

Sweeping only two profiles work, but the result can be weird. But since the ellipse path and the turning of the profile is damn steep, i do not know if there is any valid result to be expect. Note, the results in Part and Part Deign are similar/same.

Attached the file, if someone wants to test it.

[sweep crash.zip](https://github.com/FreeCAD/FreeCAD/files/12184729/sweep.crash.zip)

ElementW, 2026-08-24:
No crash but same unexpected (though valid) geometry on both `26.3.0dev.20260819` with OCC 7.8.1 and faf5feae9d2d7b0d9bb81ee9a9fda266ed38ae2c and OCC 8.1.0.dev1-3d097a0328

[image]

@maxwxyz If the issue is strictly about the crash, then it can be closed.

luzpaz, 2026-08-25:
I just renamed it so it still addresses the unexpected geometry and we don't need to open a new ticket for that.