FreeCAD/FreeCAD #6320: Allow optical+physical material properties in addition to RGB color specification

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=1942

* **Reporter:** arust
* **Date submitted:** 1/31/2015
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** assigned
* **Tags:** colors, IFC

# Original report text

Advanced spectral renderers (like LuxRender and to some extend Povray) do not use RGB internally, but physical spectral reflection, refraction, emission properties, given for example as color temperature, spectrum texture or look-up table[1], lamp-vendor-provided IES file[2], and physical refraction and reflection indices.[3]

In addition to specify plain RGB color, shininess and translucency which were fine in the time of "classical" fixed-function OpenGL not using actual shaders and programmable shaders implementing the traditional Phong lighting model, it is desirable to specify physical optical properties, which are supported by LuxRender, Povray and other advanced rendering engines.

[1] http://www.luxrender.net/wiki/LuxRender_Textures_Spectrum
[2] http://www.luxrender.net/wiki/LuxRender_Lighting#IES_data
[3] http://www.luxrender.net/wiki/Scene_file_format_dev#Material

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by clel 2017-01-10 15:31
I support the request, makes sense and is used in common CAD systems.

---

### Comment by eivindkvedalen 2017-01-10 18:48
Work in progress: http://forum.freecadweb.org/viewtopic.php?f=10&t=16714

---

### Comment by Kunda1 2017-01-11 01:33
Should this ticket be assigned to you then, eivindkvedalen ?

---

### Comment by Kunda1 2017-02-17 11:22
Test branch is at https://github.com/eivindkv/FreeCAD2/tree/Material

davesrocketshop, 2023-10-02:
The web site quoted in the original text has moved, so unless you want custom running shoes go to the new site [https://wiki.luxcorerender.org/](https://wiki.luxcorerender.org/) 

Many of the original wiki topics are there but not at the same path. You'll have to search for the topic you want.

All of this should be possible by defining an appropriate properties YAML file. No application editor is currently planned for this so use your editor of choice.

sliptonic, 2023-10-03:
@davesrocketshop  Does that mean this issue can be closed?

davesrocketshop, 2023-10-03:
I would say yes

howetuft, 2023-10-22:
Please note Render Workbench already extends standard materials to support physically-based rendering shaders (at least the most usual: diffuse, glass, disney etc.). It would be convenient to converge, IMHO.